bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #1: I can not compile Marlin 2

sn0wrider, 2019-07-04:
Hello to all,
 I can not compile Marlin 2 for BIGTREETECH-SKR-PRO-V1.1:

In file included from Marlin\src\module/stepper.h:221:0,
from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper_indirection.h:59:12: fatal error: TMCStepper.h: No such file or directory
 
********************************************************************
* Looking for TMCStepper.h dependency? Check our library registry!

*
* CLI  > platformio lib search "header:TMCStepper.h"
* Web  > https://platformio.org/lib/search?query=header:TMCStepper.h
*
********************************************************************
 
#include <TMCStepper.h>
^~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\black_stm32f407ve\src\src\Marlin.cpp.o] Error 1
In file included from Marlin\src\core\utility.cpp:25:0:
Marlin\src\core\../Marlin.h:363:29: error: 'pin_t' does not name a type; did you mean 'pid_t'?
bool pin_is_protected(const pin_t pin);
^~~~~
pid_t
*** [.pioenvs\black_stm32f407ve\src\src\core\utility.cpp.o] Error 1
 [ERROR] Took 31.72 seconds
 

but if you do a search in platformi:

platformio lib search "header:TMCStepper.h"
>>
Found 1 libraries:

TMCStepper
==========
#ID: 5513
Arduino library for configuring Trinamic stepper drivers.

Keywords: tmc, trinamic, stepper, driver, spi, uart, tmc2130, tmc2160, tmc2208, tmc2224, tmc2660, tmc5130, tmc5160, tmc5161
Compatible frameworks: Arduino
Compatible platforms: Atmel AVR, Atmel SAM, Espressif 32, Espressif 8266, Infineon XMC, Intel ARC32, Kendryte K210, Microchip PIC32, Nordic nRF51, Nordic nRF52, ST STM3
2, ST STM8, Teensy, TI MSP430
Authors: teemuatlut

what could be the problem????
Best rgs
Alex

gyutacs0804, 2019-07-04:
I have the same issue.

gyutacs0804, 2019-07-04:
That is what I get when try to compile:

^~~~~
pid_t
*** [.pioenvs\BIGTREE_SKR_PRO\src\src\core\utility.cpp.o] Error 1
Marlin\src\Marlin.cpp:249:29: error: 'pin_t' does not name a type; did you mean 'pid_t'?
bool pin_is_protected(const pin_t pin) {
^~~~~
pid_t

Marlin\src\Marlin.cpp: In function 'bool pin_is_protected(int)':
Marlin\src\Marlin.cpp:250:16: error: 'pin_t' does not name a type; did you mean 'pid_t'?
static const pin_t sensitive_pins[] PROGMEM = SENSITIVE_PINS;
^~~~~
pid_t
In file included from Marlin\src\inc/MarlinConfigPre.h:32:0,
from Marlin\src\inc/MarlinConfig.h:28,
from Marlin\src\Marlin.h:24,
from Marlin\src\Marlin.cpp:31:
Marlin\src\Marlin.cpp:251:33: error: 'sensitive_pins' was not declared in this scope
for (uint8_t i = 0; i < COUNT(sensitive_pins); i++) {
^
Marlin\src\inc/../core/macros.h:193:37: note: in definition of macro 'COUNT'
#define COUNT(a)            (sizeof(a)/sizeof(*a))
^
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\BIGTREE_SKR_PRO\src\src\Marlin.cpp.o] Error 1

yangwenxiong, 2019-07-05:
> I have the same issue.
> That is what I get when I try to compile:
> In file included from Marlin\src\core\utility.cpp:25:0:
> Marlin\src\core../Marlin.h:363:29: error: 'pin_t' does not name a type; did you mean 'pid_t'?
> bool pin_is_protected(const pin_t pin);
> ^~~~~
> pid_t
> In file included from Marlin\src\Marlin.cpp:31:0:
> Marlin\src\Marlin.h:363:29: error: 'pin_t' does not name a type; did you mean 'pid_t'?
> bool pin_is_protected(const pin_t pin);
> ^~~~~
> pid_t
> In file included from Marlin\src\lcd/../sd/SdVolume.h:41:0,
> from Marlin\src\lcd/../sd/SdBaseFile.h:37,
> from Marlin\src\lcd/../sd/SdFile.h:36,
> from Marlin\src\lcd/../sd/cardreader.h:34,
> from Marlin\src\lcd/ultralcd.h:82,
> from Marlin\src\Marlin.cpp:34:
> Marlin\src\lcd/../sd/Sd2Card.h:143:59: error: 'pin_t' does not name a type; did you mean 'pid_t'?
> bool init(const uint8_t sckRateID=SPI_FULL_SPEED, const pin_t chipSelectPin=SD_CHIP_SELECT_PIN);
> ^~~~~
> pid_t
> *** [.pioenvs\BIGTREE_SKR_PRO\src\src\core\utility.cpp.o] Error 1
> Marlin\src\Marlin.cpp:249:29: error: 'pin_t' does not name a type; did you mean 'pid_t'?
> bool pin_is_protected(const pin_t pin) {
> ^~~~~
> pid_t
> Marlin\src\Marlin.cpp: In function 'bool pin_is_protected(int)':
> Marlin\src\Marlin.cpp:250:16: error: 'pin_t' does not name a type; did you mean 'pid_t'?
> static const pin_t sensitive_pins[] PROGMEM = SENSITIVE_PINS;
> ^~~~~
> pid_t
> In file included from Marlin\src\inc/MarlinConfigPre.h:32:0,
> from Marlin\src\inc/MarlinConfig.h:28,
> from Marlin\src\Marlin.h:24,
> from Marlin\src\Marlin.cpp:31:
> Marlin\src\Marlin.cpp:251:33: error: 'sensitive_pins' was not declared in this scope
> for (uint8_t i = 0; i < COUNT(sensitive_pins); i++) {
> ^
> Marlin\src\inc/../core/macros.h:193:37: note: in definition of macro 'COUNT'
> #define COUNT(a) (sizeof(a)/sizeof(*a))
> ^
> compilation terminated due to -fmax-errors=5.
> *** [.pioenvs\BIGTREE_SKR_PRO\src\src\Marlin.cpp.o] Error 1

[](url)

yangwenxiong, 2019-07-05:
@gyutacs0804  [
[image]
](url)

yangwenxiong, 2019-07-05:
@gyutacs0804   In fact, this place should not be changed here. When using tmc2208uart mode, this place needs to be shielded.Since the original marlin 2.0 firmware does not support tmc2208 uart, it is now implemented in Pro. However, a new 2208 SoftwareSerial port driver file needs to be added. The files to be added are already in the SoftwareSerial folder.C:\Users\BiQu_\. PlatformIO \packages\framework-arduinoststm32\cores\arduino[
[image]
](url)

yangwenxiong, 2019-07-05:
> Hello to all,
> I can not compile Marlin 2 for BIGTREETECH-SKR-PRO-V1.1:
>
> In file included from Marlin\src\module/stepper.h:221:0,
> from Marlin\src\Marlin.cpp:37: [...]

> Hello to all,
> I can not compile Marlin 2 for BIGTREETECH-SKR-PRO-V1.1:
>
> In file included from Marlin\src\module/stepper.h:221:0,
> from Marlin\src\Marlin.cpp:37: [...]
@sn0wrider Hi, I have encountered this compilation error on your side. It is likely that there are other bugs in marlin 2.0. This may need to be tested by marlin officials, but I later found that the compilation passed by adding an electric hangar file.The files that need to be added are all uploaded to the firmware folder, and the add path is as follows [
[image]
](url)

Msq001, 2019-07-05:
sorry, it be updated， and it should be ok now, please download the Latest version, and Pay attention to the Readme.md

gyutacs0804, 2019-07-05:
There is no Marlin in the Firmware folder now.

sn0wrider, 2019-07-05:
Hi i'm dowwnload latest Marlin from BIGTREETECH and fully compiled

closed 2019-07-05 by sn0wrider

Msq001, 2019-07-05:
Thank you for your patience. This weekend we will create a PR to merge SKR Pro into the main branch of marlin 2.0.x. and we will also create a PR to TMC library file to support the UART mode of tmc2208. Before this, you need to manually copy the files in the SoftwareSerial folder to the platformio library directory, see [readme.md](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/firmware/Readme.md)