bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #3: Compiling

apohl23, 2019-07-06:
Hello,

I have some problems with compiling the firmware. I followed the instructions as they described. However, there are also errors (letters exchanged) within some files which I could correct to some extent.
I still don't get the board compiled because there is always a Collect2.exe error. Also in the line 166:40 in the tmc_util.h where it probably goes over the TX and RX channels...

Exact description:

Marlin\src\module\.../feature/tmc_util.h:166:40: error: no matching function for call to 'TMC2208Stepper: :TMC2208Stepper(const uint16_t&, const uint16_t&, const float&, const bool&)'
TMC2208Stepper(RX, TX, RS, has_rx)

Msq001, 2019-07-06:
hello.  are you use tmc2208 uart mode?  have you copy the 3 softwareserial files to platformio install path and comment out pin_t in hal.h file according the readme.md.

Msq001, 2019-07-06:
oh. maybe the sw_capability_platform need be updated too.  i will creat a pr for tmc220x uart in tmc library code. and it will solve this issue.   this pr will creat a softwareserial in tmc library. and it will be a generic code for most of 32 bit motherboard.

apohl23, 2019-07-07:
Hello and thank you for your reply.

I am using the Bigtreetech TMC2208 v3 UART driver
please help me thank you

I have reinstalled all libraries and PlatformIO...
i have followed all instructions as described in their description
I keep getting these mistakes 

In file included from C:\users\apohl\.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp:47:0:
C:\users\apohl\.platformio\packages\framework-arduinoststm32\cores\arduino/SoftwareSerial.h:52:3: error: 'pin_t' does not name a type; did you mean 'pid_t'?
pin_t _receivePin;
^~~~~
pid_t
C:\users\apohl\.platformio\packages\framework-arduinoststm32\cores\arduino/SoftwareSerial.h:53:3: error: 'pin_t' does not name a type; did you mean 'pid_t'?
pin_t _transmitPin;
^~~~~
pid_t
C:\users\apohl\.platformio\packages\framework-arduinoststm32\cores\arduino/SoftwareSerial.h:91:24: error: expected ')' before 'receivePin'
SoftwareSerial(pin_t receivePin, pin_t transmitPin, bool inverse_logic = false);
^~~~~~~~~~
C:\users\apohl\.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp: In member function 'bool SoftwareSerial::listen()':
C:\users\apohl\.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp:100:7: error: '_receivePin' was not declared in this scope
if (_receivePin >= 0) {
Compiling .Firmware\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_flash.c.o^~~~~~~~~~~

C:\users\apohl\.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp:100:7: note: suggested alternative: 'active_in'
if (_receivePin >= 0) {
^~~~~~~~~~~
active_in
C:\users\apohl\.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp: In member function 'void SoftwareSerial::setTX()':
C:\users\apohl\.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp:143:18: error: '_transmitPin' was not declared in this scope
digitalWrite( _transmitPin,   _inverse_logic ? LOW : HIGH);
^~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
Compiling .Firmware\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_flash_ex.c.o
*** [.Firmware\BIGTREE_SKR_PRO\FrameworkArduino\SoftwareSerial.cpp.o] Error 1

davidbilly, 2019-07-07:
> oh. maybe the sw_capability_platform need be updated too. i will creat a pr for tmc220x uart in tmc library code. and it will solve this issue. this pr will creat a softwareserial in tmc library. and it will be a generic code for most of 32 bit motherboard.

@Msq001 what are the value for TMC 2009

#define SW_CAPABLE_PLATFORM defined(__AVR__) || defined(TARGET_LPC1768) || defined(ARDUINO_ARCH_ESP32) || defined(WHAT IS THE VALUE FOR THIS??)

It's ridiculous release product such way, I have to debug 1 by 1 and fixing ur bug.

davidbilly, 2019-07-07:
> Hello and thank you for your reply.
> 
> I am using the Bigtreetech TMC2208 v3 UART driver
> please help me thank you
> 
> I have reinstalled all libraries and PlatformIO...
> i have followed all instructions as described in their description
> I keep getting these mistakes
> 
> In file included from C:\users\apohl.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp:47:0:
> C:\users\apohl.platformio\packages\framework-arduinoststm32\cores\arduino/SoftwareSerial.h:52:3: error: 'pin_t' does not name a type; did you mean 'pid_t'?
> pin_t _receivePin;
> ^~~~~
> pid_t
> C:\users\apohl.platformio\packages\framework-arduinoststm32\cores\arduino/SoftwareSerial.h:53:3: error: 'pin_t' does not name a type; did you mean 'pid_t'?
> pin_t _transmitPin;
> ^~~~~
> pid_t
> C:\users\apohl.platformio\packages\framework-arduinoststm32\cores\arduino/SoftwareSerial.h:91:24: error: expected ')' before 'receivePin'
> SoftwareSerial(pin_t receivePin, pin_t transmitPin, bool inverse_logic = false);
> ^~~~~~~~~~
> C:\users\apohl.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp: In member function 'bool SoftwareSerial::listen()':
> C:\users\apohl.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp💯7: error: '_receivePin' was not declared in this scope
> if (_receivePin >= 0) {
> Compiling .Firmware\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_flash.c.o^~~~~~~~~~~
> 
> C:\users\apohl.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp💯7: note: suggested alternative: 'active_in'
> if (_receivePin >= 0) {
> ^~~~~~~~~~~
> active_in
> C:\users\apohl.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp: In member function 'void SoftwareSerial::setTX()':
> C:\users\apohl.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp:143:18: error: '_transmitPin' was not declared in this scope
> digitalWrite( _transmitPin, _inverse_logic ? LOW : HIGH);
> ^~~~~~~~~~~~
> compilation terminated due to -fmax-errors=5.
> Compiling .Firmware\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_flash_ex.c.o
> *** [.Firmware\BIGTREE_SKR_PRO\FrameworkArduino\SoftwareSerial.cpp.o] Error 1

@apohl23 for this you have to uncomment typedef int16_t pin_t; at line 44 on SoftwareSerial.h

and go HAL_STM32/HAL.h line 135 change typedef int8_t pin_t;  to typedef int16_t pin_t;

apohl23, 2019-07-07:
@davidbilly I followed the instructions.

[image]

[image]

[image]

davidbilly, 2019-07-07:
> @davidbilly I followed the instructions.
> 
> [image]
> 
> [image]
> 
> [image]

Its remove errors for me, did you replace to C:\Users\xxxxx\.platformio\packages\framework-arduinoststm32\cores\arduino ?

Msq001, 2019-07-09:
have update firmware, tmc2208 now is tested ok, and tmc2009 Compiling success,  But the actual work seems abnormal.

apohl23, 2019-07-10:
Hi I'm still getting bugs for the TMC2208 UART drivers.

I'm already despairing if it works at all...

i also tested it with the DRV8825 because everything went through without problems, but it is not shown on the display after flashing. it only lights up, but no text

here is the complete error code

> Executing task: C:\Users\apohl\.platformio\penv\Scripts\platformio.exe run <

Processing BIGTREE_SKR_PRO (platform: ststm32; board: BigTree_SKR_Pro; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/BigTree_SKR_Pro.html
PLATFORM: ST STM32 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz 192KB RAM (1MB Flash)
DEBUG: CURRENT(cmsis-dap) EXTERNAL(cmsis-dap, jlink, stlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 20 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.4.5 #7fda12f
|   |-- <SPI> 1.0
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SPI> 1.0
|-- <EEPROM> 2.0.1
|-- <IWatchdog> 1.0.0
|-- <Servo> 1.1.4
|-- <Wire> 1.0
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduinoVariant\variant.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\IPAddress.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\Print.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\RingBuffer.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\Stream.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\Tone.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\USBSerial.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\WInterrupts.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\WMath.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\WSerial.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\WString.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\abi.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\avr\dtostrf.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\board.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\hooks.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\itoa.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\main.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\new.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\pins_arduino.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_adc.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_adc_ex.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_can.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cec.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_comp.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_comp_ex.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cortex.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_crc.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_crc_ex.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cryp.c.o
[... 783 lines cut ...]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1768                     [SKIP]
Environment LPC1769                     [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment fysetc_STM32F1              [SKIP]
Environment BIGTREE_SKR_MINI            [SKIP]
Environment STM32F4                     [SKIP]
Environment ARMED                       [SKIP]
Environment mks_robin                   [SKIP]
Environment mks_robin_mini              [SKIP]
Environment mks_robin_nano              [SKIP]
Environment JGAURORA_A5S_A1             [SKIP]
Environment black_stm32f407ve           [SKIP]
Environment BIGTREE_SKR_PRO             [ERROR]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
Environment fysetc_f6_13                [SKIP]
Environment linux_native                [SKIP]

closed 2019-07-10 by apohl23