bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #4: Compile failed

davidbilly, 2019-07-06:
Processing BIGTREE_SKR_PRO (platform: ststm32; board: STM32F407ZGT6; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/STM32F407ZGT6.html
PLATFORM: ST STM32 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz 128KB RAM (502.23KB Flash)
DEBUG: CURRENT(cmsis-dap) EXTERNAL(cmsis-dap, jlink, stlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.4.5
|   |-- <SPI> 1.0
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0.1
|-- <SPI> 1.0
|-- <IWatchdog> 1.0.0
|-- <Servo> 1.1.2
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\SoftwareSerial.cpp.o
In file included from C:\users\davidbilly\.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp:47:0:
C:\users\davidbilly\.platformio\packages\framework-arduinoststm32\cores\arduino/SoftwareSerial.h:52:3: error: 'pin_t' does not name a type; did you mean 'pid_t'?
pin_t _receivePin;
^~~~~
pid_t
C:\users\davidbilly\.platformio\packages\framework-arduinoststm32\cores\arduino/SoftwareSerial.h:53:3: error: 'pin_t' does not name a type; did you mean 'pid_t'?
pin_t _transmitPin;
^~~~~
pid_t
C:\users\davidbilly\.platformio\packages\framework-arduinoststm32\cores\arduino/SoftwareSerial.h:91:24: error: expected ')' before 'receivePin'
SoftwareSerial(pin_t receivePin, pin_t transmitPin, bool inverse_logic = false);
^~~~~~~~~~
C:\users\davidbilly\.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp: In member function 'bool SoftwareSerial::listen()':
C:\users\davidbilly\.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp:100:7: error: '_receivePin' was not declared in this scope
if (_receivePin >= 0) {
^~~~~~~~~~~
C:\users\davidbilly\.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp:100:7: note: suggested alternative: 'active_in'
if (_receivePin >= 0) {
^~~~~~~~~~~
active_in
C:\users\davidbilly\.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp: In member function 'void SoftwareSerial::setTX()':
C:\users\davidbilly\.platformio\packages\framework-arduinoststm32\cores\arduino\SoftwareSerial.cpp:143:18: error: '_transmitPin' was not declared in this scope
digitalWrite( _transmitPin,   _inverse_logic ? LOW : HIGH);
^~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
Compiling .pioenvs\BIGTREE_SKR_PRO\src\src\gcode\feature\trinamic\M122.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\src\src\gcode\feature\trinamic\M569.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\src\src\gcode\feature\trinamic\M906.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\src\src\gcode\feature\trinamic\M911-M914.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\src\src\gcode\geometry\G92.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\src\src\gcode\host\M114.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\src\src\gcode\lcd\M0_M1.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\src\src\gcode\motion\G0_G1.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\src\src\gcode\motion\G4.cpp.o
In file included from Marlin\src\gcode\feature\trinamic\../../../feature/../lcd/ultralcd.h:40:0,
from Marlin\src\gcode\feature\trinamic\../../../feature/tmc_util.h:25,
from Marlin\src\gcode\feature\trinamic\M569.cpp:28:
Marlin\src\gcode\feature\trinamic\../../../feature/../lcd/../Marlin.h:363:29: error: 'pin_t' does not name a type; did you mean 'pid_t'?
bool pin_is_protected(const pin_t pin);
^~~~~
pid_t
[... 631 lines cut ...]
  //#define E4_HARDWARE_SERIAL Serial1

  //
  // Software serial
  //   
  #define X_SERIAL_TX_PIN  PC13
  #define X_SERIAL_RX_PIN  PE4

  #define Y_SERIAL_TX_PIN  PE3
  #define Y_SERIAL_RX_PIN  PE2

  #define Z_SERIAL_TX_PIN  PE0
  #define Z_SERIAL_RX_PIN  PE1

  #define E0_SERIAL_TX_PIN PD4
  #define E0_SERIAL_RX_PIN PD2

  #define E1_SERIAL_TX_PIN PD0
  #define E1_SERIAL_RX_PIN PD1

  #define Z2_SERIAL_TX_PIN PD6
  #define Z2_SERIAL_RX_PIN PD5
#endif

on pins_BIGTREE_SKR_PRO_V1.1.h

davidbilly, 2019-07-06:
hI, 

I'm gonna use TMC 2209 on Ender 3 Pro. please assist on compile issue.

if use UART mode also need change config or just wiring on the board?

#if HAS_DRIVER(TMC2209)
/**

TMC2208 stepper drivers
Hardware serial communication ports.
If undefined software serial is used according to the pins below
*/
//#define X_HARDWARE_SERIAL Serial
//#define X2_HARDWARE_SERIAL Serial1
//#define Y_HARDWARE_SERIAL Serial1
//#define Y2_HARDWARE_SERIAL Serial1
//#define Z_HARDWARE_SERIAL Serial1
//#define Z2_HARDWARE_SERIAL Serial1
//#define E0_HARDWARE_SERIAL Serial1
//#define E1_HARDWARE_SERIAL Serial1
//#define E2_HARDWARE_SERIAL Serial1
//#define E3_HARDWARE_SERIAL Serial1
//#define E4_HARDWARE_SERIAL Serial1
//
// Software serial
//
#define X_SERIAL_TX_PIN PC13
#define X_SERIAL_RX_PIN PE4

#define Y_SERIAL_TX_PIN PE3
#define Y_SERIAL_RX_PIN PE2

#define Z_SERIAL_TX_PIN PE0
#define Z_SERIAL_RX_PIN PE1

#define E0_SERIAL_TX_PIN PD4
#define E0_SERIAL_RX_PIN PD2

#define E1_SERIAL_TX_PIN PD0
#define E1_SERIAL_RX_PIN PD1

#define Z2_SERIAL_TX_PIN PD6
#define Z2_SERIAL_RX_PIN PD5
#endif

on pins_BIGTREE_SKR_PRO_V1.1.h

apohl23, 2019-07-07:
Das gleiche Problem habe ich auch und meine Anfrage wurde geschlossen...

davidbilly, 2019-07-07:
Now after tried to fixes some error. 

Processing BIGTREE_SKR_PRO (platform: ststm32; board: STM32F407ZGT6; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/STM32F407ZGT6.html
PLATFORM: ST STM32 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz 128KB RAM (502.23KB Flash)
DEBUG: CURRENT(cmsis-dap) EXTERNAL(cmsis-dap, jlink, stlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.4.5
|   |-- <SPI> 1.0
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0.1
|-- <SPI> 1.0
|-- <IWatchdog> 1.0.0
|-- <Servo> 1.1.2
Linking .pioenvs\BIGTREE_SKR_PRO\firmware.elf
.pioenvs/BIGTREE_SKR_PRO/FrameworkArduino/stm32/timer.c.o: In function `TIM4_IRQHandler':
C:\users\davidbilly\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/timer.c:1141: multiple definition of `TIM4_IRQHandler'
.pioenvs/BIGTREE_SKR_PRO/FrameworkArduino/SoftwareSerial.cpp.o:C:\users\davidbilly\.platformio\packages\framework-arduinoststm32\cores\arduino/SoftwareSerial.cpp:265: first defined here
.pioenvs/BIGTREE_SKR_PRO/src/src/Marlin.cpp.o: In function `manage_inactivity(bool)':
e:\Development\Projects\git\BIGTREETECH-SKR-PRO-V1.1\firmware\Marlin-SKR-Pro/Marlin\src/Marlin.cpp:477: undefined reference to `Serial2'
.pioenvs/BIGTREE_SKR_PRO/src/src/Marlin.cpp.o: In function `setup':
e:\Development\Projects\git\BIGTREETECH-SKR-PRO-V1.1\firmware\Marlin-SKR-Pro/Marlin\src/Marlin.cpp:1119: undefined reference to `Serial2'
.pioenvs/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o: In function `serialprintPGM(char const*)':
e:\Development\Projects\git\BIGTREETECH-SKR-PRO-V1.1\firmware\Marlin-SKR-Pro/Marlin\src\core/serial.cpp:38: undefined reference to `Serial2'
.pioenvs/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o: In function `serial_echopair_PGM(char const*, char const*)':
e:\Development\Projects\git\BIGTREETECH-SKR-PRO-V1.1\firmware\Marlin-SKR-Pro/Marlin\src\core/serial.cpp:42: undefined reference to `Serial2'
.pioenvs/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o: In function `serial_echopair_PGM(char const*, int)':
e:\Development\Projects\git\BIGTREETECH-SKR-PRO-V1.1\firmware\Marlin-SKR-Pro/Marlin\src\core/serial.cpp:44: undefined reference to `Serial2'
.pioenvs/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o:e:\Development\Projects\git\BIGTREETECH-SKR-PRO-V1.1\firmware\Marlin-SKR-Pro/Marlin\src\core/serial.cpp:45: more undefined references to `Serial2' follow
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\BIGTREE_SKR_PRO\firmware.elf] Error 1
============================================================================================== [ERROR] Took 23.13 seconds ==============================================================================================

======================================================================================================= [SUMMARY] =======================================================================================================
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment at90usb1286_cdc             [SKIP]
Environment at90usb1286_dfu             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1768                     [SKIP]
Environment LPC1769                     [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment fysetc_STM32F1              [SKIP]
Environment BIGTREE_SKR_MINI            [SKIP]
Environment STM32F4                     [SKIP]
Environment ARMED                       [SKIP]
Environment mks_robin                   [SKIP]
Environment mks_robin_mini              [SKIP]
Environment mks_robin_nano              [SKIP]
Environment JGAURORA_A5S_A1             [SKIP]
Environment black_stm32f407ve           [SKIP]
Environment BIGTREE_SKR_PRO             [ERROR]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
Environment fysetc_f6_13                [SKIP]
Environment linux_native                [SKIP]
============================================================================================== [ERROR] Took 23.20 seconds ==============================================================================================
The terminal process terminated with exit code: 1

@bigtreetech  I'm stuck here. I have no idea about those warning log.

davidbilly, 2019-07-08:
Now I using marlin base code, compile success.

closed 2019-07-08 by davidbilly