bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #8: SKR PRO Ender 3 Display

ilakuli, 2019-07-11:
I fitted my SKR Pro to my Ender 3 with LV8729 Steppers. I connected the Ender 3 Display (1 Ribbon cable) to EXP 1 on the Board and used #define CR10STOCKDISPLAY in the Firmware. The Display lights up and shows something but it is full of graphical faults. Is this a Marlin bug?
[image]

Msq001, 2019-07-11:
this is because of delay of spi sequence
STM32F407 I have tested it [delay](https://github.com/MarlinFirmware/Marlin/blob/3218c4b33f5322e385eafb536bbcf3b557f2b04e/Marlin/src/pins/stm32/pins_BIGTREE_SKR_PRO_V1.1.h#L231)

>       #define ST7920_DELAY_1 DELAY_NS(96)
>       #define ST7920_DELAY_2 DELAY_NS(48)
>       #define ST7920_DELAY_3 DELAY_NS(600)

is worked ok.
but if defined `#define CR10STOCKDISPLAY`, this will set all delay in [`Conditionals_LCD.h`](https://github.com/MarlinFirmware/Marlin/blob/948c3f580b16c53f1f75f480fe7d9c028f94dd23/Marlin/src/inc/Conditionals_LCD.h#L113) to DELAY_NS(125),  it will not work
if you want to work normal, Please modify [here](https://github.com/MarlinFirmware/Marlin/blob/3218c4b33f5322e385eafb536bbcf3b557f2b04e/Marlin/src/pins/stm32/pins_BIGTREE_SKR_PRO_V1.1.h#L231)

>        #if HAS_GRAPHICAL_LCD
>            #ifndef ST7920_DELAY_1
>              #define ST7920_DELAY_1 DELAY_NS(96)
>            #endif
>            #ifndef ST7920_DELAY_2
>              #define ST7920_DELAY_2 DELAY_NS(48)
>            #endif
>            #ifndef ST7920_DELAY_3
>              #define ST7920_DELAY_3 DELAY_NS(600)
>            #endif
>        #endif
to
>     #if HAS_GRAPHICAL_LCD
>         #undef ST7920_DELAY_1
>         #define ST7920_DELAY_1 DELAY_NS(96)
>         #undef ST7920_DELAY_2
>         #define ST7920_DELAY_2 DELAY_NS(48)
>         #undef ST7920_DELAY_3
>         #define ST7920_DELAY_3 DELAY_NS(600)
>     #endif
and it will be ok,  I can't test this now, because I was at home and I don't have SKR Pro right now.  I will test this tomorrow. and create a PR to Marlin to fix this issue

ilakuli, 2019-07-11:
I will test tonight and report back. Thanks

ilakuli, 2019-07-11:
> this is because of delay of spi sequence
> STM32F407 I have tested it [...]

This change has worked perfectly. Stock Ender 3 Display is working fine now.

Msq001, 2019-07-13:
thanks for your test and feedback， I have created a PR in Marlin2.0 to fix this issue,  and update source code in this repository

ilakuli, 2019-07-14:
Perfect. Will close this. Thanks for your help :)

closed 2019-07-14 by ilakuli