bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #10: if enable LASER_FEATURE compiling fail!!!

sn0wrider, 2019-07-12:
Hi if if enable LASER_FEATURE. Where is PIN PWM 5v?
Im try #define SPINDLE_LASER_PWM_PIN SERVO0_PIN
not working !!!!

.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.h:223:51: error: missing
binary operator before token "("
#define digitalPinHasPWM(p) (pin_in_pinmap(digitalPinToPinName(p), PinMap_PWM))
^
Marlin\src\HAL\HAL_STM32../../inc/../HAL/./HAL_STM32/fastio_STM32.h:83:33: note: in expansion of macro 'digitalPinHasPWM'
#define PWM_PIN(P) digitalPinHasPWM(P)
^~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_STM32../../inc/SanityCheck.h:2332:12: note: in expansion of macro 'PWM_PIN'
#elif !PWM_PIN(SPINDLE_LASER_PWM_PIN)
^~~~~~~
In file included from C:\users\administrator.home.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:48:0,
from Marlin\src\HAL\HAL_STM32../../inc/../HAL/./HAL_STM32/../shared/Marduino.h:34,
from Marlin\src\HAL\HAL_STM32../../inc/../HAL/./HAL_STM32/HAL.h:28,
from Marlin\src\HAL\HAL_STM32../../inc/../HAL/HAL.h:52,
from Marlin\src\HAL\HAL_STM32../../inc/MarlinConfig.h:30,
from Marlin\src\HAL\HAL_STM32\fastio_STM32.cpp:25:

Msq001, 2019-07-13:
[image]
hello， comment out here， Can compile normally
this issue is because of here
[image]

> digitalPinHasPWM(p)

It's a function, and the return value is a variable, not a const. So there are errors in precompiling.
You can also modifiy 

> # define PWM_PIN (P) true

to fix this issue，  But not all pins of STM32 have PWM capability，So you need to use it carefully to make sure that the pins you use are PWM.
You can do this way before we find a better way to do it.

sn0wrider, 2019-07-13:
Thanks, PA1 or PA2 is a PWM pins?

Msq001, 2019-07-13:
both PA1 and PA2 is PWM pins

sn0wrider, 2019-07-13:
write the transcript of all the pins, there will be less stupid questions like mine

sn0wrider, 2019-07-13:
Thank for all, pls close the issue

closed 2019-07-15 by bigtreetech

Haschtl, 2019-10-17:
@sn0wrider 
You can find all PWM-pins (and all other special pins) in this file:
buildroot\share\PlatformIO\variants\BIGTREE_GENERIC_STM32F407_5X\PeripheralPins.c

MinSoeVertex, 2019-12-30:
or you can set Fan speed to 255 and it will works fine