bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #9: Can't establish connection between SKR-PRO and ESP01S

rafaljot, 2019-07-12:
I have 
- skr-pro 1.1
- last Marlin 2.0
- LCD 12864 - (works with last Marlin 2.0) 
- ESP01S with last ESP3d web interface works, baudrate=250000

configuration.h 
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
#define BAUDRATE 250000
#define MOTHERBOARD BOARD_BIGTREE_SKR_PRO_V1_1
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

I can't establish connection between mainboard and ESP3d

rafaljot, 2019-07-12:
I found solution:
Modify file
Marlin-bugfix-2.0.x/buildroot/share/PlatformIO/variants/BIGTREE_GENERIC_STM32F407_5X/variant.h
#define SERIAL_UART_INSTANCE    1 //ex: 2 for Serial2 (USART2)
to
#define SERIAL_UART_INSTANCE    6 //ex: 2 for Serial2 (USART2)

and configuration.h
from 
#define SERIAL_PORT_2 1
to
#define SERIAL_PORT_2 6

Msq001, 2019-07-13:
hello see this picture [image]

> serial port 
>-1 is for USB CDC
> 1 is for TFT
> 6 is for ESP-01S

and you can also enable Serial 6 in this way
[image]

Namorence, 2019-07-20:
Does that mean that you cannot use TFT (like the new BIGTREETECH TFT24 V1.1) together with ESP01S?

Namorence, 2019-07-21:
I managed to compile successfully, but how do you connect the module to your network, can find a good guide... :/

dant220, 2019-07-27:
Any luck on this? I am trying too.  I found this video. But I wonder what has already been done here? 
https://www.youtube.com/watch?v=pJGBRriNc9I

@Msq001
Can you tell me if this is correct? What do I have to build here? Do I have to compile everything with an arduino? Or is it already done if I ordered the one with my Bigtree MKS pro board?

Namorence, 2019-07-27:
@dant220 Yes, I manage to make it work. You can basically flash new FW to the ESP01S module from the SD Card directly, you only have to name it "esp3d.bin" and once you reboot the board it will search for that bin and flash it. Other than that, the FW BTT provided in their GitHub page didn't work for me at all and a friend gave me his .bin which he compiled (https://github.com/luc-github/ESP3D) and I'd suggest you join the SKR Pro group on facebook: https://www.facebook.com/groups/2264108593625228/ people are helping each other there a lot!

dant220, 2019-07-27:
Hi Namorance. Thank you. I do not have a facebook account and am very dubious of facebook. :( 

I will try that though. Thanks!!

Namorence, 2019-07-27:
@dant220 create a fake one ;b

dant220, 2019-07-27:
I guess. I hate to do it though.

defalkner, 2019-09-14:
> I managed to compile successfully, but how do you connect the module to your network, can find a good guide... :/

When its powered on it acts like another router (access point), you need to connect to it directly with laptop/cell phone to configure it to connect to your actual router and your done. It will disconnect and reconnect to your router.

kojderek, 2020-04-25:
> @dant220 Yes, I manage to make it work. You can basically flash new FW to the ESP01S module from the SD Card directly, you only have to name it "esp3d.bin" and once you reboot the board it will [...]

I have flashed my ESP01S with several various methods also from SD CARD but when I plug It to SKR PRO 1.1 no wifi connection appears :(

oldman4U, 2020-06-08:
Hi.

Is this till an issue @kojderek? If so, could you please start a new ticket. Thank you

@rafaljot Hi. Could you please close this ticket in case you do not need it any more. Thank you