bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #12: Compile Error when using E0_AUTO_FAN_PIN.

SnowMB, 2019-07-14:
Hello,

I tried to use the 2nd fan pin as hotend cooling fan with the AUTO_FAN_PIN Feature. However this results in a compilation error:

Configuration_adv.h:
```
#define E0_AUTO_FAN_PIN PE5
```

Error:
```
...

In file included from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:41:0,
                 from Marlin/src/HAL/HAL_STM32/HAL_timers_STM32.h:29,
                 from Marlin/src/HAL/HAL_STM32/HAL.h:103,
                 from Marlin/src/HAL/HAL_STM32/HAL.cpp:26:
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1825:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from /home/marc/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:48:0,
                 from Marlin/src/HAL/HAL_STM32/../shared/Marduino.h:33,
                 from Marlin/src/HAL/HAL_STM32/HAL.h:32,
                 from Marlin/src/HAL/HAL_STM32/HAL.cpp:26:
/home/marc/.platformio/packages/framework-arduinoststm32/cores/arduino/pins_arduino.h:223:51: error: call to non-constexpr function 'bool pin_in_pinmap(PinName, const PinMap*)'
 #define digitalPinHasPWM(p)         (pin_in_pinmap(digitalPinToPinName(p), PinMap_PWM))
                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/fastio_STM32.h:83:33: note: in expansion of macro 'digitalPinHasPWM'
 #define PWM_PIN(P)              digitalPinHasPWM(P)
                                 ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1825:19: note: in expansion of macro 'PWM_PIN'
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~
In file included from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:41:0,
                 from Marlin/src/HAL/HAL_STM32/HAL_Servo_STM32.cpp:25:
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1825:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from /home/marc/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:48:0,
                 from Marlin/src/HAL/HAL_STM32/../../inc/../HAL/HAL_STM32/../shared/Marduino.h:33,
                 from Marlin/src/HAL/HAL_STM32/../../inc/../HAL/HAL_STM32/HAL.h:32,
                 from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/HAL_STM32/HAL_Servo_STM32.cpp:25:
/home/marc/.platformio/packages/framework-arduinoststm32/cores/arduino/pins_arduino.h:223:51: error: call to non-constexpr function 'bool pin_in_pinmap(PinName, const PinMap*)'
 #define digitalPinHasPWM(p)         (pin_in_pinmap(digitalPinToPinName(p), PinMap_PWM))
                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/../../inc/../HAL/HAL_STM32/fastio_STM32.h:83:33: note: in expansion of macro 'digitalPinHasPWM'
 #define PWM_PIN(P)              digitalPinHasPWM(P)
                                 ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1825:19: note: in expansion of macro 'PWM_PIN'
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~
*** [.pioenvs/BIGTREE_SKR_PRO/src/src/HAL/HAL_STM32/HAL.cpp.o] Error 1
*** [.pioenvs/BIGTREE_SKR_PRO/src/src/HAL/HAL_STM32/HAL_Servo_STM32.cpp.o] Error 1
In file included from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:41:0,
                 from Marlin/src/HAL/HAL_STM32/HAL_spi_STM32.cpp:30:
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1825:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from /home/marc/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:48:0,
                 from Marlin/src/HAL/HAL_STM32/../../inc/../HAL/HAL_STM32/../shared/Marduino.h:33,
                 from Marlin/src/HAL/HAL_STM32/../../inc/../HAL/HAL_STM32/HAL.h:32,
                 from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/HAL_STM32/HAL_spi_STM32.cpp:30:
/home/marc/.platformio/packages/framework-arduinoststm32/cores/arduino/pins_arduino.h:223:51: error: call to non-constexpr function 'bool pin_in_pinmap(PinName, const PinMap*)'
 #define digitalPinHasPWM(p)         (pin_in_pinmap(digitalPinToPinName(p), PinMap_PWM))
                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/../../inc/../HAL/HAL_STM32/fastio_STM32.h:83:33: note: in expansion of macro 'digitalPinHasPWM'
 #define PWM_PIN(P)              digitalPinHasPWM(P)
                                 ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1825:19: note: in expansion of macro 'PWM_PIN'
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~
In file included from Marlin/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:41:0,
                 from Marlin/src/HAL/HAL_STM32/HAL_timers_STM32.h:29,
                 from Marlin/src/HAL/HAL_STM32/HAL.h:103,
                 from Marlin/src/HAL/HAL_STM32/HAL_timers_STM32.cpp:29:
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1825:5: error: non-constant condition for static assertion
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
     ^~~~~~~~~~~~~
In file included from /home/marc/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:48:0,
                 from Marlin/src/HAL/HAL_STM32/../shared/Marduino.h:33,
                 from Marlin/src/HAL/HAL_STM32/HAL.h:32,
                 from Marlin/src/HAL/HAL_STM32/HAL_timers_STM32.cpp:29:
/home/marc/.platformio/packages/framework-arduinoststm32/cores/arduino/pins_arduino.h:223:51: error: call to non-constexpr function 'bool pin_in_pinmap(PinName, const PinMap*)'
 #define digitalPinHasPWM(p)         (pin_in_pinmap(digitalPinToPinName(p), PinMap_PWM))
                                     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/fastio_STM32.h:83:33: note: in expansion of macro 'digitalPinHasPWM'
 #define PWM_PIN(P)              digitalPinHasPWM(P)
                                 ^~~~~~~~~~~~~~~~
Marlin/src/HAL/HAL_STM32/../../inc/SanityCheck.h:1825:19: note: in expansion of macro 'PWM_PIN'
     static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);
                   ^~~~~~~
*** [.pioenvs/BIGTREE_SKR_PRO/src/src/HAL/HAL_STM32/HAL_spi_STM32.cpp.o] Error 1
*** [.pioenvs/BIGTREE_SKR_PRO/src/src/HAL/HAL_STM32/HAL_timers_STM32.cpp.o] Error 1
========================================== [ERROR] Took 10.12 seconds ==========================================

================================================== [SUMMARY] ==================================================
Environment megaatmega2560              [IGNORED]
Environment megaatmega1280              [IGNORED]
Environment at90usb1286_cdc             [IGNORED]
Environment at90usb1286_dfu             [IGNORED]
Environment DUE                         [IGNORED]
Environment DUE_USB                     [IGNORED]
Environment DUE_debug                   [IGNORED]
Environment LPC1768                     [IGNORED]
Environment LPC1769                     [IGNORED]
Environment melzi                       [IGNORED]
Environment melzi_optiboot              [IGNORED]
Environment rambo                       [IGNORED]
Environment sanguino_atmega644p         [IGNORED]
Environment sanguino_atmega1284p        [IGNORED]
Environment STM32F1                     [IGNORED]
Environment fysetc_STM32F1              [IGNORED]
Environment BIGTREE_SKR_MINI            [IGNORED]
Environment STM32F4                     [IGNORED]
Environment ARMED                       [IGNORED]
Environment mks_robin                   [IGNORED]
Environment mks_robin_mini              [IGNORED]
Environment mks_robin_nano              [IGNORED]
Environment JGAURORA_A5S_A1             [IGNORED]
Environment black_stm32f407ve           [IGNORED]
Environment BIGTREE_SKR_PRO             [FAILED]
Environment teensy35                    [IGNORED]
Environment malyanm200                  [IGNORED]
Environment esp32                       [IGNORED]
Environment fysetc_f6_13                [IGNORED]
Environment linux_native                [IGNORED]
==================================== 1 failed, 0 succeeded in 10.13 seconds ====================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

```

Msq001, 2019-07-15:
please according to here, https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/10
This is because of PWM_PIN

ledfreak3d, 2019-07-30:
works just fine for me

MinSoeVertex, 2019-12-30:
Don't change fan speed from 255 and it will work fine.

oldman4U, 2020-06-08:
Hello snow.

@SnowMB Please close this ticket in case you do not need it any more.

Thank you

SnowMB, 2020-06-08:
sure. this is resolved :D

closed 2020-06-08 by SnowMB