bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #13: SKR Pro v1.1

SkuterGaming, 2019-07-16:
### Help,
**how to fix this problem ?**

GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
swd
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
cortex_m reset_config sysresetreq
### **Error: unable to find CMSIS-DAP device**
### **Error: No Valid JTAG Interface Configured.**
### ***** [upload] Error -1**

Evg33, 2019-07-16:
> how to fix this problem ?

1. Build (only build. w/o upload)
2. Copy \firmware\Marlin-SKR-Pro\.pioenvs\BIGTREE_SKR_PRO\firmware.bin  to SD-Card.
3. Insert SD-Card into board
4. reboot board

golfromeo-fr, 2019-07-31:
please close this issue

jeffkarney, 2019-08-03:
This is by no means a solution to the problem.... do not close this until the problem is solved.

farmstar72, 2019-08-05:
I just got my SKR Pro and don't see the SD Card like I do with my SKR V1.3 (It shows up as drive D:). However, I can upload to the SKR Pro using Upload_Protocol = dfu . But that doesn't put the firmware on the SD - I guess it's going to the board's memory? I also copied the firmware directly to the SD card with a card reader then put it into the SKR Pro and that seemed to work.
I still feel like I'm missing something because I don't see the SD card as drive D: when I plug the board into my computer with the USB cable.
 - Is that by design or am I missing a step?

Scott

farmstar72, 2019-08-05:
Kingprint answered my question - It's by design. Unlike other Arm based control boards, you cannot access the SD Card via the boards USB

rflulling, 2019-10-13:
Going to kick up a bit of dust by saying that I have two SD cards. The internal is seen as a FLASH, and is set to behave in this way by pins_BIGTREE_SKR_PRO.h 
```
// Use one of these or SDCard-based Emulation will be used
//#define SRAM_EEPROM_EMULATION   // Use BackSRAM-based EEPROM emulation
#define FLASH_EEPROM_EMULATION  // Use Flash-based EEPROM emulation
```

Neither show up on the desktop. This might also be due to the fact I have not configured SD communication I think are needed to talk to the storage device. In this case I honestly expect ONLY that SD card 1 will show if at all. 

Honestly I care not if it shows, only that the software we use can see it to upload files. I care not if we can see SD card 0, but that PlatformIO or ArduinoIDE can program the card when we generate new firmware. 

The reason we cannot flash the firmware is a factory installed bootloader that manages the Controller. I believe the presence of the bootloader is obvious. I believe the way to resolve this is to ask BIGTREETECH aka BIQU to fix the bootloader.

jgwinner, 2019-12-11:
I realize the instructions say to copy to the SD Card and put it in the Pro, but that is a PITA. It also doesn't allow developers to do on-device debugging or programming.

BIQU/BIGTREETECH, is there any way to make the device remote upload work? This is a much needed feature. 

@farmstar72 I tried the dfu upload, but got the error "No DFU capable USB device available" Doing more research ...