bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #14: parts cooling fan control?

fenyx13, 2019-07-16:
I have hottend cooling on cnc fan 1 but when I have fans attached to cnc fan 2 or 3 they don't start and I can't control them.
thanks

closed 2019-07-21 by fenyx13

rlvanhuis, 2019-09-14:
@fenyx13, What were the steps used to solve.  I am at the same point.

Thanks

fenyx13, 2019-09-15:
@rlvanhuis
I ended up using one of the sets of heater pins for the parts cooling fan and reassigning the pins in the firmware. Worked for me.

rlvanhuis, 2019-09-15:
Not fully understanding how to trace the input to the output yet.  I hope
this is what you did to solve your issue.
I will remap the pins located in the file "pins_BIGRTEE_SKR_RRO_V1.1.h".  I
will swap PE5 for PD14.  If I follow correctly the output will move to the
heater 1 location?

Thanks

fenyx13, 2019-09-16:
Yup, that’s what I did. Good luck! 🙂