bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #15: BLTOUCH

p1234321, 2019-07-17:
Hello,

I always get an error when using bltouch.

> Error:STOP called because of BLTouch error - restart with M999
[ERROR] Error:STOP called because of BLTouch error - restart with M999 <

please help

Tarot67, 2019-07-17:
There some explaination on this error:

http://marlinfw.org/docs/gcode/M999.html

p1234321, 2019-07-18:
thanks, but there is not really useful information. i still dont get it to work. I dont get an error, but nothing is happening when homing z axis. i can sied all bltouch commands and they work (M280 S0 P120....).
Anybody who has this setup working? SKR Pro and bltouch?
thanks
paul

Tarot67, 2019-07-18:
Ok! you say with M280 S0 P120 the pin of BLTouch move!. 
Your BL-Touch work on SET mode the servo work
but there are another mode "GET" retrive information of your BL-Touch it's difficulte to test this feature. Perharps it's existe a sketch for testing BL-Touch retrive information on arduino uno for example. 

There are a lot of issue with BL-Touch! even it's brand new!

You are tested another BL-Touch ?

I see that

This is a arduino sketch to test BL-Touch

#include <Servo.h> 

Servo servo;

void setup(){
  Serial.begin(115200);
  servo.attach(D2);
}

void loop() {
  servo.write(120);
  Serial.println(servo.read());
  delay(5000);
  servo.write(121);
  Serial.println(servo.read());
  delay(5000);
}

The output result is like the following:

119
120
119
120
119
120
119
120

p1234321, 2019-07-18:
I tried two different BLtouch, wihich worked perfectly under marlin 1.1.4 and an mks gen board.
so i really dont know whats the issue. maybe its got something to do with marlin 2.0?

thanks

paul

dant220, 2019-07-18:
This seems an excellent Tutorial here:
https://www.youtube.com/watch?v=5cSzFCv7K4Q

On an SKR 1.3. covers settings and hardware config. (it will be a bit different on the pro board). But worth a watch.

albrk2902, 2019-07-21:
I do have a working solution. Just ignore the picture in the manual and connect the black wire to GND on the ZMin connector ( middle_Pin) and the with wire to ZMin Signal   and it will work, at least for me it does

p1234321, 2019-08-05:
Thanks! that also works for me!

closed 2019-08-05 by p1234321