bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #16: SKR PRO - ENV_DEFAULT = LCP1768 or BIGTREE_SKR_PRO?

ruttenfam, 2019-07-20:
Hi,

The manual states that in platfrom.io the default_env is set as LCP1768 but that the processor of the SKR 1.3 not the PRO. It now shows the BIGTREE_SKR_PRO as an environment so do I use this?
thanks

dant220, 2019-07-26:
See compile issues thread
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/17

Black6spdZ, 2019-08-16:
can't this be closed as its pretty well documented this uses an STM micro and not a LCP1768 like older SKR boars

omahena, 2019-08-20:
I vote that the issue remains opened until the documentation is fixed. The manual for SKR Pro 1.1 points to a dropbox file for Marlin building instructions:
https://www.dropbox.com/s/ppjfflhf3j5yzh2/MarlinV2.0%20SKRV1.1%20instruction.docx?dl=0

The document clearly states that LPC1768 should be used. It also suggests to use the obsolete `env_default` variable and not the `default_envs`. Documentation is pretty poor and increasingly obsolete.

Haschtl, 2019-08-20:
The referred building manual is for the SKR 1.1 Not the Pro. Its almost the same for the Pro. Maybe they will Update the building manual, even if Its almost identically

omahena, 2019-08-21:
I hear you and agree with you. But... The doc is referenced directly in the SKR Pro 1.1 manual and points to SKR 1.1 (Pro... one could assume) Marlin building instructions. 

Even with 20+ years of (non arduino / platformio) development experience they had me staring at the LPC1768 value for a few seconds before I decided that it is likely for the SKR 1.1. Watching a few videos from Teaching Tech on YouTube on the subject also helped gather some hints.

So having a manual like this may be better than nothing... But when the manual is misleading in more than one way... The manual benefit becomes arguable. The manual is also increasingly obsolete and lacking. The English is fine as it is because my Chinese is definitely worst than the English in the manual - so thank you BTT for writing the manual in a language hopefully... More people can understand.

tphan26363, 2020-02-03:
I don't believe  that is the right chip.  defined here is STM32F407ZG
#
# BIGTREE_SKR_PRO (STM32F407ZGT6 ARM Cortex-M4)
#
[env:BIGTREE_SKR_PRO]
platform = ststm32@5.4.3
framework = arduino
board = BigTree_SKR_Pro
extra_scripts = pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
build_flags = ${common.build_flags}
  -DUSBCON -DUSBD_USE_CDC -DUSBD_VID=0x0483 -DUSB_PRODUCT=\"STM32F407ZG\"
  -DTARGET_STM32F4 -DSTM32F407_5ZX -DVECT_TAB_OFFSET=0x8000 -DHAVE_HWSERIAL6
lib_deps = ${common.lib_deps}
lib_ignore = Adafruit NeoPixel, c1921b4, TMC26XStepper, SailfishLCD, SailfishRGB_LED, SlowSoftI2CMaster
src_filter = ${common.default_src_filter} +<src/HAL/HAL_STM32> +<src/HAL/HAL_STM32_F4_F7> -<src/HAL/HAL_STM32_F4_F7/*> +<src/HAL/HAL_STM32_F4_F7/STM32F4>
monitor_speed = 250000