bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #17: Compile Error for environment BIGTREE_SKR_PRO

ruttenfam, 2019-07-20:
This is probably related to my previous issue. 
Documentation doesn't seem to be quite up to date with the files here.

I used SKR PRO firmware files, and it uses; env_default  = BIGTREE_SKR_PRO

However, on compiling the file I get an error: Environment BIGTREE_SKR_PRO             [FAILED]
==================== 1 failed, 0 succeeded in 1.22 seconds ====================
See screenshot 
[image]

Thanks

golfromeo-fr, 2019-07-21:
same here PinNames.h 
error: 'PortA' (and others) undeclared here

Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\core_callback.c.o
In file included from C:\users\romagnyg\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/pinmap.h:22:0,
                 from C:\users\romagnyg\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PeripheralPins.h:34,
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\dwt.c.o
                 from C:\users\romagnyg\.platformio\packages\framework-arduinoststm32\variants\BIGTREE_GENERIC_STM32F407_5X/variant.h:37,
                 from C:\users\romagnyg\.platformio\packages\framework-arduinoststm32\system\STM32F4xx/stm32f4xx_hal_conf.h:4,
                 from C:\users\romagnyg\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_hal.h:30,
                 from C:\users\romagnyg\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include/stm32f4xx.h:250,
                 from C:\users\romagnyg\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\stm32_def.h:35,
                 from C:\users\romagnyg\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\PortNames.h:33,
                 from C:\users\romagnyg\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\PortNames.c:30:
C:\users\romagnyg\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:16:12: error: 'PortA' undeclared here (not in a function)

ruttenfam, 2019-07-21:
UPDATE: Reinstalled GIT (soooo slow. not sure whats wrong with that software!) Finally managed to make ATOM the default reader and the code now ran a bit further. Still Not completing. Now the errors start with the Port packages. 

Any support from Bigtreetech?

[image]

ruttenfam, 2019-07-21:
BIGTREETECH: pls update your manual for the SKR PRO inline with where your Marlin merge is up to. Also, should we use ATOM or Microsoft as the reader? Does it matter?

Thanks

Evg33, 2019-07-22:
platformio.ini

```
[env:BIGTREE_SKR_PRO]
#platform = ststm32
platform = ststm32@5.4.3

```

NickMilton, 2019-07-22:
> This is probably related to my previous issue.
> Documentation doesn't seem to be quite up to date with the files here.
>
> I used SKR PRO firmware files, and it uses; env_default = BIGTREE_SKR_PRO [...]

I am having identical issue - very frustrating. Need help or the board goes in the bin!

ruttenfam, 2019-07-22:
Bigtreetech please respond with a full answer to allow customers to compile the firmware successfully. 
Without proper customer support, the industry will turn their backs to your products and company pretty quickly. Already pretty pissed off customers in this thread!

golfromeo-fr, 2019-07-22:
I guess we can stop using this repository for now

I have managed to install vscode + plateformio and MarlinFirmware/Marlin
on bugfix-2.0.x

using 
env_default  = BIGTREE_SKR_PRO
I can compile with 
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
but my MKS version of the screen doesn't light up
I am trying to buy the 
LCD 12864 LCD Control from BigTreeTech if case it works better

But anyway I can send Gcode command from the USB connection with pronterface
so there are positive things happening. I need a working screen

note : vscode and platformio can become very messy as all the configuration and stored package are much hidden. I have deleted .plateformio local folder

NickMilton, 2019-07-23:
> Bigtreetech please respond with a full answer to allow customers to compile the firmware successfully.
> Without proper customer support, the industry will turn their backs to your products and company pretty quickly. Already pretty pissed off customers in this thread!

To be fair - I am a first time user of VCS/Platformio so I might be making a basic mistake? However as I followed the Bigtreetech guides and used their version of Marlin 2.0 I would have expected it at leat to compile... I don't know enough about this error to be able to proceed -

NickMilton, 2019-07-23:
> I guess we can stop using this repository for now. I guess the repository is trash now
> 
> I have managed to install vscode + plateformio and MarlinFirmware/Marlin from bugfix-2.0.x
> using
> env_default = BIGTREE_SKR_PRO
> 
> I can compile with
> #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> but my MKS version of the screen doesn't light up
> => I am trying to buy the LCD 12864 LCD Control from BigTreeTech in case it works better (my hope)
> 
> So now I can send Gcode commands to the USB connection with pronterface so I can see the nema motors moving
> => so there are positive things happening but I need a working screen especially the LCD 12864
> 
> note : vscode and platformio can become very messy as all the configuration and stored packages are much hidden to the user. I have deleted .plateformio local folder and others things to make the official Marlin compile properly

Ahh - I have a Bigtreetech LCD 12864 so I can test....once I can get the f/w to compile LOL

NickMilton, 2019-07-23:
> I guess we can stop using this repository for now. I guess the repository is trash now
>
> I have managed to install vscode + plateformio and MarlinFirmware/Marlin from bugfix-2.0.x
> using
> env_default = [...]

Great news - I replicated what you did. I am getting Environment BIGTREE_SKR_PRO  [FAILED]. Can you tell me how you deleted the local folder and other things? I'll see if I can replciate on my end..

cheers

bigtreetech, 2019-07-23:
Please according to here https://github.com/MarlinFirmware/Marlin/issues/14672 and here https://github.com/MarlinFirmware/Marlin/pull/14674
this if because of the STM32 platform dependency libraries have be updated，But it's not sure whether the latest version is incompatible or have bugs.  so will Limited Version to 5.4.3， This will solve this issue.

bigtreetech, 2019-07-23:
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/commit/21356040541a08aedf731c7e7be9c43f9250b678 have be updated

NickMilton, 2019-07-24:
> [2135604](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/commit/21356040541a08aedf731c7e7be9c43f9250b678) have be updated

Thank you! This works now
Kind regards

dant220, 2019-07-26:
I am stuck...I am getting:
<command-line>:0:0: note: this is the location of the previous definition
*** [.pioenvs/BIGTREE_SKR_PRO/src/src/HAL/HAL_STM32/HAL_Servo_STM32.cpp.o] Error 1
*** [.pioenvs/BIGTREE_SKR_PRO/src/src/HAL/HAL_STM32/HAL_timers_STM32.cpp.o] Error 1
*** [.pioenvs/BIGTREE_SKR_PRO/src/src/HAL/HAL_STM32/HAL.cpp.o] Error 1
*** [.pioenvs/BIGTREE_SKR_PRO/src/src/HAL/HAL_STM32/HAL_spi_STM32.cpp.o] Error 1

[Configurations.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3433649/Configurations.zip)

Any clues?

golfromeo-fr, 2019-07-26:
> 
> 
> > [2135604](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/commit/21356040541a08aedf731c7e7be9c43f9250b678) have be updated
> 
> Thank you! This works now
> Kind regards

does your bigtreetech LCD 12864 work please?

NickMilton, 2019-07-26:
Yes it works fine and the SD card slot works too. What I found are 2 things:

1 The LCD lights up on power up but does not display boot screen until I plug the USB cable into  computer

2 The LCD will not move off the boot screen until I connect the host (In this case Repetier Host) running on the computer.

The board is configured for external power. Looking through forums to see if this problem is known 

Ta

DavidThijs, 2019-07-29:
@NickMilton This is indeed my observation too. I'm wondering why it only starts working when the USB connector is plugged in. Tought about a ground loop, but +5v and GND are there on EXP1, but when using only EXP1 and EXP2, the display is dim. Only when adding the TFT connector, then it is bright. The 12864 emulation only seems to work when USB plugged in. Strange indeed.

chasinreno, 2019-07-31:
I downloaded the current BIGTREE_PRO firmware with the changes above but still get the BITGREE_SKR_PRO environment error.
ARGGG!  bigtreetech SKR v1.3 usb-driver won't install on my win8.1 so that board is wasted and now SKR PRO won't compile.

dant220, 2019-07-31:
Use the marlin latest release instead.

ruttenfam, 2019-08-01:
@Bigtreetech can you please ensure that the files available here are a combination of marlin 2.0 and the board specific changes are all reflected in the firmware. If the base file should still be the latest marlin 2.0 code from marlin and not the one you have posted here, then the files should be deleted from GitHub and only the required changes to the stock code should be shown. 

Also please answer the USB/LCD question/above. The reason user post here on the @bigtreetech GitHub page and not a general Forum, is because we expect answers from the manufacturer to support their product all these users paid for. If other users can help that is great, but the answers should first come from @bigtreetech in the first place. Thank you.

chasinreno, 2019-08-01:
My issue was solved by deleting git in control panel.  I wonder if it was a conflict between git desktop and git client.  I can't explain it but bugfix 2.0 now compiles without error.
Thanks to everyone who helped including Bit Tree Tech support. ;)

rlvanhuis, 2019-08-02:
Unable to compile.  Keep getting numerous errors.  
Windows 10 Home Edition
VSCode with Platformio
Approximately 20 minutes to have a fail.
I have tried just compiling the code as given from the Bigtreetech GitHub SKR V1.1 Pro location
Epic Fail.
Support from Bigtreetech is telling me the code is good and I am modifying it.  
No to sure but unzip does not modify the code.
I have installed Git, VSCode, Platformio.  Am I missing something?
Pretty new to troubleshooting code problems.  Passable at entering parameter changes and pressing buttons.
Any help would be appreciated.  Thanks

RudiBlubb, 2019-08-02:
Dear All,
i just compile it minutes before with the the same Platformio and the results are 1 Warning (i am not sure if relevant) and 4 compile PROBLEMS (but all 4 the same thing)

I seems some one has use an outdated notation "boolean".
I corrected it to "bool" for times. 
See attached pdf !

Bigtreetech guys, please check and change for all others the relevant file !

Happy programming !

wesk-ms, 2019-08-02:
Hard to help without knowing the errors or your config.  :)

Please zip your platformio.ini, configuration.h, configuration_adv.h, and the complete output from your compile attempt.

rlvanhuis, 2019-08-02:
I am very new at this, sorry for the unconventional approach.  Something
has changed from this morning.  I am now getting this error set.  I do not
even know where to start.  The project seemed simple at first, then nothing
wanted to compile.  So I tried to just compile the image from their site
...no changes.  still fails

Warning! `env_default` configuration option in section [platformio] is
deprecated and will be removed in the next release! Please use
`default_envs` instead
Processing BIGTREE_SKR_PRO (platform: ststm32@5.4.3; board:
BigTree_SKR_Pro; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION:
https://docs.platformio.org/page/boards/ststm32/BigTree_SKR_Pro.html
PLATFORM: ST STM32 5.4.3 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: framework-arduinoststm32 3.10500.190327 (1.5.0),
toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing c1921b4
Using cache:
C:\Users\rlvan\.platformio\.cache\8c\f244bfa38b7bfff5e9042f5d9f9f5c8c
WindowsError: [Error 206] The filename or extension is too long:
'c:\\Users\\rlvan\\Desktop\\New folder
(2)\\BIGTREETECH-SKR-PRO-V1.1\\firmware\\Marlin-SKR-Pro\\.piolibdeps\\BIGTREE_SKR_PRO\\_tmp_installing-cr6pke-package\\TMC26XStepper-c1921b4856a5ef4b659b661b8cb5dd56f3075e57\\examples\\TMC26XMotorTester\\processing\\TMC26XMotorTest':
  File
"C:\Users\rlvan\.platformio\penv\lib\site-packages\platformio\builder\main.py",
line 126:
    env.SConscript("$BUILD_SCRIPT")
  File
"C:\Users\rlvan\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py",
line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File
"C:\Users\rlvan\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py",
line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\rlvan\.platformio\platforms\ststm32@5.4.3\builder\main.py",
line 91:
    target_elf = env.BuildProgram()
  File
"C:\Users\rlvan\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py",
line 224:
    return self.method(*nargs, **kwargs)
  File
"C:\Users\rlvan\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py",
line 124:
    _build_project_deps(env)
  File
"C:\Users\rlvan\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py",
line 49:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File
"C:\Users\rlvan\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py",
line 224:
    return self.method(*nargs, **kwargs)
  File
"C:\Users\rlvan\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py",
line 1043:
    project.install_dependencies()
  File
"C:\Users\rlvan\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py",
line 878:
    lm.install(uri)
  File
"c:\users\rlvan\.platformio\penv\lib\site-packages\platformio\managers\lib.py",
[... 46 lines cut ...]
Environment sanguino_atmega1284p        [IGNORED]
Environment STM32F1                     [IGNORED]
Environment fysetc_STM32F1              [IGNORED]
Environment BIGTREE_SKR_MINI            [IGNORED]
Environment STM32F4                     [IGNORED]
Environment STM32F7                     [IGNORED]
Environment ARMED                       [IGNORED]
Environment alfawise_U20                [IGNORED]
Environment mks_robin                   [IGNORED]
Environment mks_robin_mini              [IGNORED]
Environment mks_robin_nano              [IGNORED]
Environment JGAURORA_A5S_A1             [IGNORED]
Environment black_stm32f407ve           [IGNORED]
Environment BIGTREE_SKR_PRO             [FAILED]
Environment teensy35                    [IGNORED]
Environment malyanm200                  [IGNORED]
Environment esp32                       [IGNORED]
Environment fysetc_f6_13                [IGNORED]
Environment linux_native                [IGNORED]
=========================================================== 1 failed, 0
succeeded in 1.50 seconds
===========================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

rlvanhuis, 2019-08-02:
Here are the other files you were looking for.

I really appreciate you looking at this for me.
I am using VScode Platformio.
Thank You!

wesk-ms, 2019-08-03:
@rlvanhuis I've tried two different browsers and I'm not seeing any attached files in your last response.  Are you sure they are getting attached to your message?
If you're attaching them in an email response, it doesn't appear to be working.  You might try attaching the files through the Github website.

What stepper drivers do you have plugged into the SKR pro board?  Platformio is trying to process anything and everything related to a long list of dependencies bigtreetech has specified in the platformio.ini file.  I doubt you actually need all that excess baggage.  The problem causing the compile failure is that platformio is trying to process the TMC26XStepper library and the folder structure is so nested that it's hitting a path limitation.

'c:\\Users\\rlvan\\Desktop\\New folder
(2)\\BIGTREETECH-SKR-PRO-V1.1\\firmware\\Marlin-SKR-Pro\\.piolibdeps\\BIGTREE_SKR_PRO\\_tmp_installing-cr6pke-package\\TMC26XStepper-c1921b4856a5ef4b659b661b8cb5dd56f3075e57\\examples\\TMC26XMotorTester\\processing\\TMC26XMotorTest':

The fix is probably real easy, we just need to know what stepper drivers you're using.

@bigtreetech , your most recent commit that changed the TMCStepper library to your branch is causing this folder to be created on the local disk:  **TMC26XStepper-c1921b4856a5ef4b659b661b8cb5dd56f3075e57**

jeffkarney, 2019-08-03:
Whats sad is this is the worst thread I have ever seen. Absolutely no information from anyone as to what their problem is.
"Failed"
is not a description of a problem. A screenshot of your config helps none without the real error.

Unfortunately I think this is the closest thing I can find to a possible problem I can not find anything on.

Trying to compile Marlin on OSX for this board...
```
AVAILABLE: dfu, jlink, stlink
CURRENT: upload_protocol = stlink
Uploading .pioenvs/BIGTREE_SKR_PRO/firmware.elf
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-23-00:01)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
embedded:startup.tcl:56: Error: Can't find scripts/interface/stlink.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 56
*** [upload] Error 1
```

Have no idea what the issue is.

rlvanhuis, 2019-08-03:
I was using the files from the Bigtreetech SKR Pro V1.1 GitHub location.  No changes were made.  I wanted to have confirmation of a good compile before wasting my time changing all the values.  So far I have not been able to perform a successful compile in its virgin state.  The list of issues dropped dramatically yesterday from when I tried on 8/1.  No changes were made to my computer.

I believe the WindowsError [Error:206] was caused by the file system not liking the number of levels from root.  I moved the Marling-SKR-Pro folder to my root directory.  No longer getting the 206 error.   
My drivers are TMC5160.  I will be filling all six locations with the same.  I will be setting it up for SPI. 
 I will also be using an ESP-01S and a TFT35 controller.
I believe in order to program the board I will need a properly compiled file to place on the SD card followed by a power cycle.
Again, thank you for taking your time to give me some valued advice.

darknode, 2019-08-03:
@jeffkarney You have compiled/builded firmware, then you tried to upload ("CURRENT: upload_protocol = stlink").

"Uploading .pioenvs/BIGTREE_SKR_PRO/firmware.elf" - according to this, you could copy firmware file .pioenvs/BIGTREE_SKR_PRO/firmware.bin on sdcard and update your board.

wesk-ms, 2019-08-03:
@rlvanhuis, Yes, the 206 error is caused by a nested folder structure.  Bigtreetech warns about this here:  https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware

> _Note: if you want to design your own version naming rules, folder names must not exceed 64 characters in length, marlin 2.0 project files in other folders, folder nesting depth can not exceed 3, otherwise, when opening the project compilation project, there will be unexpected compilation errors! It is recommended that the file name should not be named too long, the shorter the better. File nesting depth should not be too much, the less nesting, the better._

Unfortunately, you still haven't attached any configuration files or compiler output, so there's really no way for anyone to help you.  If you are now getting a completely different compile error we have no way of knowing what it is unless you tell us.

If you're going to be using TMC5160 drivers, then you need to edit the configuration.h and specify TMC5160 as the drivers you're using.

Also, I'd recommend removing this line from the platformio.ini since you aren't using TMC 26xx drivers.

`https://github.com/trinamic/TMC26XStepper/archive/c1921b4.zip`

rlvanhuis, 2019-08-03:
Sorry for the slow reply.  I am at work.

jeffkarney, 2019-08-03:
> @jeffkarney You have compiled/builded firmware, then you tried to upload ("CURRENT: upload_protocol = stlink").
> 
> "Uploading .pioenvs/BIGTREE_SKR_PRO/firmware.elf" - according to this, you could copy firmware file .pioenvs/BIGTREE_SKR_PRO/firmware.bin on sdcard and update your board.

Thank you for pointing this out! Even though I knew this, I was unable to find this file originally. But you inspired me to re-read the instructions, it does clearly state to just "Build" not "Build/Program". Which will leave the file there.

Black6spdZ, 2019-08-05:
proper board "#define BOARD_BIGTREE_SKR_PRO_V1_1 " and proper env set " default_envs = BIGTREE_SKR_PRO " but still cannot compile with latest 2.x, why unknown motherboard error ?!

Marlin\src\HAL\HAL_STM32\../../inc/../pins/pins.h:537:4: error: #error "Unknown MOTHERBOARD value set in Configuration.h"
   #error "Unknown MOTHERBOARD value set in Configuration.h"

budi67, 2019-08-11:
Hello all
I have bought a SKR Pro for my CoreXY, before i receipt it, i have to try to compile the firmware.
It's the firmware from Bigtreetech Github

I have this error, any idea ?
(sorry for my english)

A big thanks
[image]

Formation3D, 2019-08-16:
I need some help with this issue as well. I tried to download Atom at first. But it is stuck on Atom Dependencies, and wont install PlatformIO. So I tried to download VSC, and it cant find the board in the environment. I downloaded the files from GitHub, and I can open Marlin up in VSC, but it cant find the environment when I try to build.
[image]
[image]

Romano26, 2019-08-20:
hi 
Someone would have done a tutorial?
Thx

defalkner, 2019-09-15:
> @NickMilton This is indeed my observation too. I'm wondering why it only starts working when the USB connector is plugged in. Tought about a ground loop, but +5v and GND are there on EXP1, but when [...]

Did you move the jumper in the middle of the board so it uses the PSU's power and not the USB? The board gives you the option for either - this jumper dictates where the power comes from for the CPU & minor devices.