bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #18: SKR PRO Serial Ports

ilakuli, 2019-07-23:
Hello

when I set the Serial Ports for the SKR Pro like in the Firmware provided here on GitHub, my board would not boot up until a USB Connection is established. It would wait for USB Connection to boot up.

Fixed this issue by changing the Serial Ports in Configuration.h [here](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/979d0b338d3f15887463eb8dcdb6ca13e3919f4a/firmware/Marlin-SKR-Pro/Marlin/Configuration.h#L107) as follows:

#define SERIAL_PORT -1
#define SERIAL_PORT_2 1

omahena, 2019-08-20:
This works. for me as well. But some hints are strewn in the issue #9 .

It is stated that it is better to use:
```
#define SERIAL_PORT 1   // Port for TFT
#define SERIAL_PORT_2 6  // Port 6 for ESP-01s
```

On the other hand the comment for `SERIAL_PORT` says:
`Select the serial port on the board to use for communication with the host.`

And for `SERIAL_PORT_2`:
`Select a secondary serial port on the board to use for communication with the host.`

Serial port:
-1 is for USB CDC
1 is for TFT
6 is for ESP-01S

So.. If one reads between the lines... I guess it is no surprise that with `SERIAL_PORT` defined as 1 and `SERIAL_PORT_2` as 6... This will connect Marlin's command interface to these two ports and accept commands from the TFT screen and the ESP3D WiFi module... But this will render the USB serial port useless so you cannot use Octoprint for example.

So...
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
```
Seems to make sense until SERIAL_PORT_3 is added and I can throw my BTT WiFi module in the waste BCPs box it seems for future use. The LCD seems more useful after all if you already have OctoPrint.

jimhiggs, 2020-04-18:
Glad I found this comment I was pulling my hair out trying to find the right combination to get both the usb and esp01 Wifi working together. So its one or the other onboard. Previously I had the esp01 plugged into the TFT and that gave me both . Problem I have been having is that running pronterface over USB it cant see files on the onboard sd and running ESP3D on wifi it cant read the SD card on the TFT.... I just want to be able to stop having to copy files by loading them onto an sd card on my PC and plugging it into the TFT.

TechnicalWorld2016, 2020-11-04:
I had uploaded marlin 2.0 firmware on my Bigtree tech SKR PRO V1.1 board successfully and connected with pronterface and checked the motor orientations. Next day I am about to do the PID tuning, but the board was not connecting to my PC, no COM port is detected my PC, I have tried to connect with other laptops in my office, but there is no result. The board is getting power on and taking input supply. When I try to press ENDSTOP manually, the respective led near stepper driver is blinking, resulting in normal functionality. But I am unable communicate with the board neither with Pronterface nor with TFT 2.8. I have tried changing the serial_port, Serial_port 2 in marlin and tried to upload the firmware, but no change in the result, finally I am unable to cummunicate with the Board. 

If anyone have a solution, please do reply.

gafillll, 2021-04-04:
[image]
check the image