bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #22: MKS MINI 12864 LCD

degogarn, 2019-07-30:
try to compile, but keeps failing with error 
Marlin\src\lcd\dogm\ultralcd_DOGM.h:116:21: error: 'DOGLCD_CS' was not declared in this scope
   #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                              // 8 stripes
                     ^
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.h:116:21: note: suggested alternative: 'DOGLCD_SCK'
   #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                              // 8 stripes
                     ^
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
and
Marlin\src\lcd\dogm\ultralcd_DOGM.h:116:32: error: 'DOGLCD_A0' was not declared in this scope
   #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                              // 8 stripes
                                ^
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.h:116:32: note: suggested alternative: 'DOGLCD_SCK'
   #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                              // 8 stripes
                                ^
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
*** [.pioenvs\BIGTREE_SKR_PRO\src\src\lcd\dogm\ultralcd_DOGM.cpp.o] Error 1
_ have had a look in the pins folder and it seems that the MKS Mini 12864 LCD is not supported or I am missing something_

golfromeo-fr, 2019-07-31:
I have the same compilation problem with Marlin 2.0. I guess you should also report to Marlin

degogarn, 2019-07-31:
> I have the same compilation problem with Marlin 2.0. I guess you should also report to Marlin

I have added this to the pins file
// Alter timing for graphical display
#ifndef ST7920_DELAY_1
  #define ST7920_DELAY_1 DELAY_NS(96)
#endif
#ifndef ST7920_DELAY_2
  #define ST7920_DELAY_2 DELAY_NS(48)
#endif
#ifndef ST7920_DELAY_3
  #define ST7920_DELAY_3 DELAY_NS(715)
#endif
/**
  * Special pins
  *   P1_30  (37) (NOT 5V tolerant)
  *   P1_31  (49) (NOT 5V tolerant)
  *   P0_27  (57) (Open collector)
  *   P0_28  (58) (Open collector)
  */
  // MKS MINI12864 with graphic controller and SD support
  // http://reprap.org/wiki/MKS_MINI_12864
  //
  #if ENABLED(MKS_MINI_12864)
  #if (MOTHERBOARD == BOARD_BIGTREE_SKR_PRO_V1_1)
  #define DOGLCD_CS PG3
  #define DOGLCD_A0 PG6
  #endif
  #endif
but now getting this error
{
	"resource": "/c:/Users/derek/OneDrive/Documents/Arduino/firmware/Marlin-SKR-Pro/Marlin/src/lcd/dogm/ultralcd_DOGM.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'BOOTSCREEN_TIMEOUT' was not declared in this scope",
	"startLineNumber": 176,
	"startColumn": 16,
	"endLineNumber": 176,
	"endColumn": 16
}

golfromeo-fr, 2019-08-05:
maybe it is hopeless to use MKS MINI 12864 LCD or other MKS LCD because (read the Marlin files about MKS SBASE board) MKS reverse the pin layout of the LCD compared to REPRAP standard
if you manage to compile, you will have to deal with the pins !!!!
:(

degogarn, 2019-08-05:
MKS MINI 12864 works with the BIGTREETECH SKR V1.3 board so I don´t see why it shouldn´t work with the pro

degogarn, 2019-08-06:
> maybe it is "hopeless" to use MKS MINI 12864 LCD or other MKS LCD because (read the Marlin files about MKS SBASE board) MKS reverse the pin layout of the LCD compared to REPRAP standard
> if you manage to compile, you will have to deal with the pins !!!!
> :(
> I have received my BigtreeTech LCD 12864 in 7 days

can you try this and let me know if it works
// Alter timing for graphical display
  #if HAS_GRAPHICAL_LCD
    #undef ST7920_DELAY_1
    #define ST7920_DELAY_1 DELAY_NS(96)
    #undef ST7920_DELAY_2
    #define ST7920_DELAY_2 DELAY_NS(48)
    #undef ST7920_DELAY_3
    #define ST7920_DELAY_3 DELAY_NS(600)
  #endif

  // MKS MINI12864 with graphic controller and SD support
  // http://reprap.org/wiki/MKS_MINI_12864
  //
  #if ENABLED(MKS_MINI_12864)
  #if (MOTHERBOARD == BOARD_BIGTREE_SKR_PRO_V1_1)
  #define DOGLCD_CS PG3
  #define DOGLCD_A0 PG6
  #endif
  #endif

Lastnovik, 2019-09-27:
Working display and  SD card if add in pins_BIGTREE_SKR_PRO_V1.1.h this lines:
#if ENABLED(MKS_MINI_12864)
#if (MOTHERBOARD == BOARD_BIGTREE_SKR_PRO_V1_1)
#define DOGLCD_CS PG3
#define DOGLCD_A0 PG6
#define LCD_PINS_RS    PD10
#define BTN_EN1        PG10
#define BTN_EN2        PF11
#define SD_DETECT_PIN  PF12
#define LCD_SDSS       PB12
#define LCD_PINS_ENABLE PD11
#define LCD_PINS_D4    PG2
#define LCD_CONTRAST_MIN 0
#define LCD_CONTRAST_MAX 255
#define DEFAULT_LCD_CONTRAST 200
#endif
#endif

tphan26363, 2019-10-06:
Have any one able to use advance pause with the 12864 LCD? it seem that the M108 did not work for SKR-pro  so the OK button can not clear the "wait_for_user" global variable

Relyat, 2019-10-19:
I'm having a problem which may be connected to the Mini 12864 LCD board.  The printer's motherboard is a Makerbase  MKS Gen L V1.0.  From the beginning - the LCD screen was clicking on and off twice a second - and swapping between the 'info' and 'Main' menu screens. Thinking that it needed to be re-flashed -I recompiled and uploaded the Marlin  firmware.  It made no difference to the problem.  I ordered a new motherboard (same model) - thinking that a fault on it may be the problem.  I've just installed the new motherboard - and switched on the printer. Unfortunately - the same problem is occurring.

So I'm now thinking that there is a problem with this LCD screen.

Does anyone have any idea what might be going on here?