bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #23: E3D Thermocoule E1 Thermal Runaway

hopeinformer, 2019-08-04:
I'm not sure if this is a bug or operator error.
I have tried many combinations of setting up the thermocouple for the E3D V6 hotend on the Bigtreeteck MKR Pro board. I am using the thermocouple amplifier board and reassigned the pin for Temp Sensor 0 in the PINS.h to use one of the Expansion pins (PF8). I also tried to use the original sensor 0 pin (PF4) but none of these work. 
I followed the E3D instructions found here: https://reprap.org/wiki/ExtThermoCouple_1.0

AnHardt, 2019-08-04:
Neither nor.
This part of Marlin was never ported to 3.3V processors.

This module needs 5V to work - 3.3V is not enough.
It also will produce up to 5V at its output.
5V is to much for the 3.3 processors and will destroy the processors ADC.

**Using this Modules with 3.3V processors is not a good idea!**

Even if we ignore that problem and never exceed 330°C and for that stay below 3.3V we do need an other table because now not 5V is 500°C == 1023ADC, but 3.3V is 330°C == 1023ADC.

Haschtl, 2019-08-04:
I have the same Problem with my E3d Pt100 amplifier Board.
I will supply this Board with 5V and add a voltage divider to the output, that the maximum output voltage is 3,3V. As I know, we wouldnt need to modify the LUT... But maybe i'm wrong.
Anyway I will calibrate my thermistors manually and adjust the lookuptable, so thats Not a Real Problem for me

hopeinformer, 2019-08-06:
Thank you for your comments AnHardt and Haschtl.
Based on that feedback, I'll stick with the traditional thermistor and forget about the higher temperatures until this gets worked out.

evandene, 2019-10-06:
> I have the same Problem with my E3d Pt100 amplifier Board.
> I will supply this Board with 5V and add a voltage divider to the output, that the maximum output voltage is 3,3V. As I know, we wouldnt [...]

Dear Haschtl,
I have the same problem but because I'm quite a hardware dummy, could you please explain how you installed the E3D PT100 sensors and the amplifier board to the SKR Pro v1.1?
What pins are you using for hooking up the amplifier(s)
I have two extruders and two hot-ends

tphan26363, 2019-10-06:
I did have the same problem with PT100 amp.  How ever this happen to all other hardware as well since the rail voltages of all chips are not the same.  For all 32 bit some of them max out and 3.2 V some of them is 3.25V ect. So the PT100 board can not really adapt to all of them.  It just an analog amplification so it is not accurate to scale with an AD converter.  I just use the 3.3V as the VCC for the PT100( it's ok the chip could works at much lower voltages) and compensate for the different of the AD voltage range(3.0 for some chip mine is 3.2V) by multiply the temperature table to a 3.2/3.3V ratio( you must look at the data sheet of your chip to find out max Voltage for you AD converter).  Or you can use 100 ohm resistor reference to recalculate the table.  Practically it is easier to use a buck converter to lower the 5V to 3.3V to PT100 board and adjust until you read correctly ADC values

hopeinformer, 2019-10-06:
Thank you tphan26363. That's the most helpful information on these PT100 and amp board. I will use the buck converter. Which pins are best used for this on the Pro v1.1?

tphan26363, 2019-10-06:
Just use the existing pin for e0.  it has a pull up resistor but should not
affecting the PT100  much since the board should be able to dive much more
current than that

evandene, 2019-10-08:
For your information: I use MARLIN 2.0 from BIGTREE-TECH because of some pin and stepper corrections. https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1

It works, ... BUT do not connect the amplifier output signal with the thermistor input E1, E2, etc .. my board froze up.

I use E3D PT100 sensor and E3D amplifier.
Will change this later by a Adafruit amplifier because of the 5V input and the 3.3V output signal of the board.
T0 is responding on my E3D amplifier with PT100 sensor
I use PF8 from EXTENSION1 as signal pin. 
I use GND and 3.3V from a free limit switch connector
Do not forget to change pin settings in the:
[image]

[image]

tphan26363, 2019-10-08:
That does not sound right.  Froze up when connect the amp to the T0 input?
I think it can't froze up when connect the amp board to the T0 input.
Unless your amp board is freaking weird.  But I think it more like a
firmware reaction to shutdown the board when it received weird temperature
reading from the amp board.  I have no problem with mine what so ever.  The
Adafruit look good to me because it does have a part number Max31865 and
clear label to the board.  That can help .  But for me I think this board
is  totally different from the amp board mine you.
This is actually is a A-D converter not an amp board. The T0 and T1
actually is a analog input so I don't know how you can configure it.  But
it probably that you have to change it to a digital input instead.  Good
luck.

evandene, 2019-10-08:
T0 is NOT responding on my E3D amplifier with PT100 sensor
I use PF8 from EXTENSION1 as signal pin.
I use GND and 3.3V from a free limit switch connector.

settings in configuration.h
#define TEMP_SENSOR_0 20
#define TEMP_SENSOR_1 20

ALL Works fine now. I installed also my second extruder and hot-end.
I also have my WiFi working
Good luck

emilhse, 2019-11-11:
> T0 is NOT responding on my E3D amplifier with PT100 sensor
> I use PF8 from EXTENSION1 as signal pin.
> I use GND and 3.3V from a free limit switch connector.
> In pins_BIGTREE_SKR_PRO_V1.1.h re-asign the pin number(s) PF8
> 
> settings in configuration.h
> #define TEMP_SENSOR_0 20
> #define TEMP_SENSOR_1 20
> 
> ALL Works fine now. I installed also my second extruder and hot-end.
> I also have my WiFi working
> Good luck

Can you please explain what changes you made in pins_BIGTREE_SKR_PRO_V1.1.h?

emilhse, 2019-11-29:
> That does not sound right. Froze up when connect the amp to the T0 input? I think it can't froze up when connect the amp board to the T0 input. Unless your amp board is freaking weird. But I think [...]

Can you please share, what amplifier board you used, and which pins you used?

AnHardt, 2019-11-29:
The Tx connectors are made for comparing a thermistors **resistance** to a on-board 4K7 pull-up resistor.

The E3d Pt100 amplifier Boards have an proportional to the temperature **voltage** at their output. You can connect them to the T connectors, but the measured temperature will be wrong because the pull up resistor changes the measured voltage. Better connect them to a ADC-pin with no pull-up resistor.

The MAX31865 are completely different and connected to a SPI-bus. They have an ADC on board.

evandene, 2019-11-29:
The Adafruit PT100 board do work fine for me as well as the E3D Pt100 board

emilhse, 2019-11-29:
Thank you very much guys. Did anyone tried connecting max31865? If yes how? I'm quite dummy in all of this .And which board is more accurate? The reason I'm asking because I have no choice, I have built pellet extruder which needs accurate and stable temperature. I can of course use external temperature regulator but I want to control directly from the board

emilhse, 2019-11-29:
> The Adafruit PT100 board do work fine for me as well as the E3D Pt100 board

What type of stepper drivers do you use , do you use spi for them at the same time?

evandene, 2019-11-29:
> > The Adafruit PT100 board do work fine for me as well as the E3D Pt100 board
> 
> What type of stepper drivers do you use , do you use spi for them at the same time?

Okay,
Please find my configuration file, my configuration_adv and my pins_BIGTREE_SKR_PRO_V1,1 file.
I us TMC5160 drivers.
The pins file need to be renamed as ........ V1.1

[configuration.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3906049/configuration.txt)
[configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3906050/configuration_adv.txt)
[pins_BIGTREE_SKR_PRO_V11.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3906053/pins_BIGTREE_SKR_PRO_V11.txt)

emilhse, 2019-11-29:
thank you very much sir.

emilhse, 2019-12-01:
do i need to remove pull up resistor?

evandene, 2019-12-01:
PF9 will be okay

AnHardt, 2019-12-01:
> do i need to remove pull up resistor?

You could - but using any of these ADC-pins is much easier.

[image]

evandene, 2019-12-01:
[image]

emilhse, 2019-12-04:
thank you very much guys for great support

johnny49r, 2020-03-17:
Hi, You can use the E3D PT100 board without modification on the BigTreeTech boards but should work for other 3.3V / 32 bit controllers.
First you need to run the E3D board at 3.3V since the BTT analog inputs are not 5V tolerant. The E3D PT100 [INA826] amp will run fine down to 3V.
Second you will need to compile the Marlin FW with the attached thermistor_22.h file. This has correct translation for 3.3V operation and also offset caused by the <standard> 4.7K pullup on the BTT thermistor inputs.
Limitations: Due to the limited voltage swing of the amplifier at 3.3V the max temperature it can report is approx 360 C but I think that is useful for most folks.
Please see the attached file for more details. You will need to change the file extension from .txt to .h.
[thermistor_22.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4343517/thermistor_22.txt)

TheMiguelBi, 2020-04-13:
> Hi, You can use the E3D PT100 board without modification on the BigTreeTech boards but should work for other 3.3V / 32 bit controllers.
> First you need to run the E3D board at 3.3V since the BTT analog inputs are not 5V tolerant. The E3D PT100 [INA826] amp will run fine down to 3V.
> Second you will need to compile the Marlin FW with the attached thermistor_22.h file. This has correct translation for 3.3V operation and also offset caused by the 4.7K pullup on the BTT thermistor inputs.
> Limitations: Due to the limited voltage swing of the amplifier at 3.3V the max temperature it can report is approx 360 C but I think that is useful for most folks.
> Please see the attached file for more details. You will need to change the file extension from .txt to .h.
> [thermistor_22.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4343517/thermistor_22.txt)

Thanks for this info **johnny49r** !
I am trying to operate my PT100 amplifier. But I use another card, **BTT GTR v1.0**
The no-load measurement is 31°C, when it should measure 20-21°C.
I am feeding the electronics with a powerbank via USB. The power supply of the amplifier is not 3.3v but slightly lower (3.25v).

_Something escapes me ....._

johnny49r, 2020-04-14:
Hi,

Please verify that you are using this PT100 amplifier:
https://wiki.e3d-online.com/E3D_PT100_Amplifier_Documentation#Schematic
This works fine with the BTT V1.3/V1.4 boards.

I couldn't find a schematic for the BTT GTR board but the thermistor table
calculation assumes that the thermistor analog input has a 4.7K ohm pullup
to VCC (in your case 3.25V). If the pullup value is different this would
account for the discrepancy.

I have attached the THERMISTOR_22.h file and it shows the calculations for
the Marlin lookup table using a PT100 temperature vs resistance chart as
described here:
http://www.manoraz.com/_Uploads/dbsAttachedFiles/PT-100_Table.pdf

The Marlin lookup table has data points for every 10 degrees C associating
raw ADC values for each point. Finer resolution is derived by
interpolation.
I hope this helps. Please let me know if you need further info.

John

johnny49r, 2020-04-14:
Hi,

Sorry, one small detail I might have forgotten to mention: You need to add
the following text to the thermistors.h file located in the
Marlin/src/module/thermistor directory:

#if ANY_THERMISTOR_IS(22) // Pt100 with E3D amp @ 3.3v w/4.7K pullup
(BigTreeTech, etc.). Abbycus 2020
  #include "thermistor_22.h"
#endif

Also make sure the thermistor_22.h file is installed in that same directory.

Then recompile with the latest Marlin 2.0.x bugfix.

Good luck!

Symbyos83, 2020-04-14:
> Hi, Sorry, one small detail I might have forgotten to mention: You need to add the following text to the thermistors.h file located in the Marlin/src/module/thermistor directory: #if [...]

Hi johnny49r, I have found perhaps the problem that says TheMiguelBi, in the GTR wiring diagram there is a 22 ohm resistor in series, perhaps it is what creates temperature differences ...
[image]

johnny49r, 2020-04-14:
Hi symbyos83,

The 22 ohm shouldn't make any difference since it is connected to the ADC
input which is high impedance. The 22 ohm resistor and the two diodes are
there to protect the analog input from incorrect voltages that could damage
the chip but normally there is almost no current flow through the resistor.
Maybe TheMiguelBi might be using a different PT100 amplifier or perhaps he
forgot to compile the firmware using my lookup table (thermistor_22.h).
If the lookup table is wrong I can make another if I know what the PT100
amp circuit looks like.

Thanks, John

tiros76, 2020-05-15:
@johnny49r  

Man, a huge thanks to you for sharing this!  I was initially having the same issue with my SKR1.4 Turbo as @TheMiguelBi  was with my E3D pt100 board and your thermistor_22.h file and thermistor.h edit. I was consistently showing 7 degrees over ambient with the PT100 powered from the 3.3v rail.  So I decided to use an adjustable buck  converter connected to Vin to tune the supply voltage until I was getting correct temp. the PT100 board seems to be incredibly sensitive to variations in the 3.3v supply voltage. 

I pulled Vin (in my case 12v) from one of the extra always-on fan headers on the SKR1.4. Connected the 12v supply to the input on the buck converter and output to the PT100 board.  You will need to tune the converter before you wire it up to avoid torching the thermistor inputs. I my case I just attached the converter to a benchtop power supply (you can use Vin)  and adjusted until I had exactly 3.3v on the output.  From there I connected it up to the printer and tweaked in real time so that it matched ambient on the LCD. All it took was around 0.03v  to bring it into spec. 

Confirmed accuracy with a K-type thermocouple stuffed under the silicone sock. Compared and verified the temps from 80-220C to an identical hotend setup measured in the exact same fashion. The readings were spot on. 

I got the adjustable buck converter from amazon. just search Mini MP1584EN DC-DC Buck Converter

Not sure if this is the safest setup considering the source of the 3.3 volts, although it seemed very steady on my meter. Perhaps someone else has a better idea for a cleaner power source for the PT100 board since the 3.3v from the SKR1.4 was not working. 

Hopefully this provides something helpful. I will update when I get around to clearing out the PLA and testing to a higher temp with this setup.

panecki, 2020-05-18:
Hi, does anyone have a working version 2.0.5.3(skr pro 1.1) with the MAX31865 amplifier? I've been working on the launch for several days and I'm still getting Err MAXTEMP: E1. I've also tested MAX31855 and MAX6675 I get the same error everywhere.

GadgetAngel, 2020-07-28:
> Hi, does anyone have a working version 2.0.5.3(skr pro 1.1) with the MAX31865 amplifier? I've been working on the launch for several days and I'm still getting Err MAXTEMP: E1. I've also tested MAX31855 and MAX6675 I get the same error everywhere.

I would like this information also

GadgetAngel, 2020-07-28:
When using 21 as the setting for TEMP_SENSOR_0 in Configuration.h for Marlin 2.0.x, is the 4.7K ohm resistor on the T1 (E0) port taken into account when using this table at 3.3VDC (with PT100 and PT100 amplifier board)?

johnny49r, 2020-07-28:
Hi,

Please check here: https://github.com/johnny49r/BigTreeTech

The calculated table is for the E3D PT100 amp using the INA826 which has a
diff gain of 10. You can modify the calculations to create your own
thermistor table. Please let me know if you need help.
John H

GadgetAngel, 2020-07-29:
> Hi, Please check here: https://github.com/johnny49r/BigTreeTech The calculated table is for the E3D PT100 amp using the INA826 which has a diff gain of 10. You can modify the calculations to create your own thermistor table. Please let me know if you need help. John H
> […](#)

@johnny49r In the newest version of Marlin 2.0.6 they added new temperature sensor tables to the firmware.  You use table number 22 when setting up your table.  Why not change your table number to 222?  

Marlin version 2.0.6's  definitions for PT100:                                                             
 *    -5 : PT100 / PT1000 with MAX31865 (only for sensors 0-1)                                                                                                             
 *    20 : Pt100 with circuit in the Ultimainboard V2.x with 5v excitation (AVR)
 *    21 : Pt100 with circuit in the Ultimainboard V2.x with 3.3v excitation (STM32 \ LPC176x....)                                                            
 *   201 : Pt100 with circuit in Overlord, similar to Ultimainboard V2.x                                                                                                 
 *   147 : Pt100 with 4k7 pullup
 *   110 : Pt100 with 1k pullup (non standard)   

Newest edition to temperature sensor tables (but these do not affect PT100 sensor they just use number 22 and 23 and are for 100k Thermistor sensors): 
*    22 : 100k (hotend) with 4.7k pullup to 3.3V and 220R to analog input (as in GTM32 Pro vB)
*    23 : 100k (bed) with 4.7k pullup to 3.3v and 220R to analog input (as in GTM32 Pro vB)

## BUT, I am confused, what are the differences between table 20, table 21, table 201,  table 147 and table 110 beyond the obvious pull-up resistors? 
## What is table 201 used for??  What does "with circuit in Overlord" mean?

#  This is my current understanding:                                                                                                                                                                               

  * I should use table -5 if I am using a PT100 sensor attached to the MAX31865 board and the SPI bus.                                        
  * I should use table 20 when I use a PT100 sensor attached to a PT100 amplifier board that use 5VDC as its input voltage and is wired up to an 8-bit MCU's analog input PIN which is 5VDC tolerant and has 0 Ohm pull-up resistor in the MCU's connected port.
  * I should use table 21 when I use a PT100 sensor attached to a PT100 amplifier board that uses 3.3VDC as its input voltage and is wired up to a 32-bit MCU's analog input PIN which is 3.3VDC tolerant and has 0 Ohm pull-up resistor in the MCU's connected port.                                                                                                                                                                                                                                       
  * I use table 147 when I use a PT100 sensor attached to a PT100 amplifier board that uses 5VDC as its input voltage and is wired up to an 8-bit MCU's analog input PIN which is 5VDC tolerant and has a 4.7k pull-up resistor in the MCU's connected port.                          
  * I use table 110 when I use a PT100 sensor attached to a PT100 amplifier board that use 5VDC as it input voltage and is wired up to an 8-bit MCU analog input PIN which is 5VDC tolerant and has a 1k Ohm pull-up resistor in the MCU's connected port.                         
  * I use a table I created, say number 222 (@johnny49r  suggestion), if I use a PT100 sensor attached to a PT100 amplifier board that uses 3.3 VDC as it input voltage and is wired up to a 32-bit MCU analog input PIN which is 3.3VDC tolerant and has a 4.7K pull-up resistor in the MCU's connected port.      

 ####  Is my understanding correct?  If so, why doesn't Marlin make another table number that is permanently installed in Marlin for those of us that have moved over to 32-bit MCUs who's spare analog input pins are ONLY 3.3VDC tolerant and have 4.7k pull up resistor on the thermistor ports?   To create your own table and install it in Marlin is somewhat advanced for most Marlin users.          

## For 32-bit MCU's I have see four option in the current version of Marlin for attaching PT100 sensor to the 32-bit MCU: 
  * Option number 1:    Use table 21 and attached the PT100 amplifier board to a spare analog pin, but wire up  3.3VDC referenced to the MCU's digital ground (GND) to power the PT100 amplifier board and use GND to ground the PT100 amplifier board.   
  * Option number 2:     Use table -5 an attached the PT100 sensor to the MAX31865 board which uses the SPI bus.  But this type of connection tends to be noisier compared to the analog connection via the ADC input.  
  * Option number 3:    Use table 21 and attach the PT100 amplifier board to the hacked thermistor port.  The hack would mean I removed the thermistor port's resistor and capacitor and tapped into the 3.3VDC side of the thermistor's port resistor to gain access to the 3.3VDC that is reference to the 32-bit MCU's analog ground (Analog_GND).  Now I use the 3.3VDC referenced to Analog_GND to power my PT100 amplifier and wire the thermistor port's Analog_GND PIN to the PT100 amplifier board.  Now I do not have a ground loop problem and I have noise immunity.
[... cut ...]
## Does this make sense? or am I totally wrong on this information?  If I am wrong please someone point out why I am wrong.
### Please assume I know that I have to change the TEMP_SENSOR_0 marlin variable and the TEMP_0_PIN in the pins file of the MCU. 

### Also please assume that I know I need to add the following to the pins file if I chose to use the MAX31865 amplifier board: 
```
#ifndef MAX31865_CS_PIN
  #define MAX31865_CS_PIN TEMP_0_PIN
  #define MAX6675_SS_PIN MAX31865_CS_PIN   
  #define MAX31865_MOSI_PIN TMC_SW_MOSI
  #define MAX31865_MISO_PIN TMC_SW_MISO
  #define MAX31865_SCK_PIN TMC_SW_SCK
#endif
```

johnny49r, 2020-07-30:
Hi GadgetAngel,

The project I put on Github describes using the E3D PT100 amplifier with
the BTT board. This is a bit different than using the MAX31865 board.
The E3D PT100 amplifier has an analog output and is read by the ADC on the
BTT board whereas the MAX31865 provides temperature information digitally
through the SPI serial bus.
It seems that Marlin firmware has a 'thermistor' option for the MAX31865
board which I believe is -5.
I'm not sure how to wire the MAX31865 to the BTT board but there is an SPI
bus available on the EXP2 connector.
I hope this helps. John

GadgetAngel, 2020-07-31:
@johnny49r Thanks for the input.  
## What I am trying to figure out now is why would I use the hardware SPI over the Software SPI on a 3D printer? 
  * I have searched with google about MAX31855 (thermocouple amplifier via SPI) and MAX31865 (PT100 amplifier via SPI) and a lot of users are using the hardware SPI option to connect up their amplifier boards.  Why?
### I tell myself this: "The rise in temperature on a 3D printer is not faster than the 32-bit processor so SPI protocol implemented by software (bit-banging) is good enough".  
   * With the hardware SPI option you MUST find the SPI MISO, SPI MOSI, and SPI CLK lines for the SPI bus (to tap into).  Like you said you could use the EXP2 port. But, if you are using the EXP2 port for a display you will now have to create a custom cable for the EXP2 port so you can still get access to the SPI CLK line and SPI MOSI and SPI MISO lines.  

   * But if you use Software SPI you can use any free digital I/O pins and assign them to be the SPI CLK, SPI MOSI, SPI MISO lines and the CS line.  Software SPI will not run as fast as the hardware SPI but the change in temperature does not occur in less than a second. On a 32-bit processor slow is really not that slow anymore.

johnny49r, 2020-07-31:
Hi GadgetAngel,

I had a quick look at the BTT SKR 1.4 schematic and it looks like EXP2 is
the only choice for connecting any external SPI devices.
If you are using a display you would need to make a custom cable that would
break out the SPI signals from the EXP2 connector. The SPI bus on the EXP2
is one of two hardware SPI buses from the CPU and hopefully Marlin firmware
can resolve which is which. Also, software bit-banging isn't needed since
the SPI bus on EXP2 is a hardware SPI bus.

Another option you might consider is to use a standard thermistor but if
you need temperatures greater than about 270C they won't work for you.
Yet another choice would be to use an analog PT100 amplifier (like the E3D
board) and deal with making an analog thermistor lookup table. This was the
option I chose, mainly because I already had the E3D board.
Keep in mind that absolute accuracy of temperature measurement isn't all
that important to 3D printing. What is important is consistent and stable
measurement.

BTT documentation is poor (understatement?) and things like PT100 support
isn't on their radar.

Regards, John

GadgetAngel, 2020-07-31:
## Marlin version 2.0.6's  definitions for PT100:                                                             
  *    -5 : PT100 / PT1000 with MAX31865 (only for sensors 0-1)                                                                                                             
 *    20 : Pt100 with circuit in the Ultimainboard V2.x with 5v excitation (AVR)
  *    21 : Pt100 with circuit in the Ultimainboard V2.x with 3.3v excitation (STM32 \ LPC176x....)                                                            
  *   201 : Pt100 with circuit in Overlord, similar to Ultimainboard V2.x                                                                                                 
  *   147 : Pt100 with 4k7 pullup
  *   110 : Pt100 with 1k pullup (non standard)   

# What is table 201 used for?  What does "with circuit in Overlord" mean?

johnny49r, 2020-07-31:
Hi,

Maybe it's a typo? They might mean "circuit in overload" which doesn't make
a lot of sense either. Sorry but I can't be much help there.

John

GadgetAngel, 2020-07-31:
> Hi, Maybe it's a typo? They might mean "circuit in overload" which doesn't make a lot of sense either. Sorry but I can't be much help there. John
> […](#)

I found out ---- table 201 you use it when I have a PT100 sensor attached to a PT100 amplifier board that uses LMV324 amp (used by the 3D printer Overlord V1.1)

GadgetAngel, 2020-08-01:
> Hi GadgetAngel, I had a quick look at the BTT SKR 1.4 schematic and it looks like EXP2 is the only choice for connecting any external SPI devices. If you are using a display you would need to make a [...]

@johnny49r I tried your method last night.  I followed your instructions and your thermistor_22.h table will not compile in the latest version of Marlin (Marlin-2.0.6).  

The array "const short" is no longer supported its needs to be "const temp_entry_t" array  and the two dimensional array will not work.  The array has to be one dimensional. 

```
const short temptable_22[][2] PROGMEM = {
changes to 
const temp_entry_t temptable_22[] PROGMEM = {

```
Plus the

```
#define REVERSE_TEMP_SENSOR_RANGE flag needs to be changed to 
#define REVERSE_TEMP_SENSOR_RANGE_22 1
```

If you adjust your table or create another table thermistor_22.h for Marlin-2.0.6 with the above adjustments your instruction will work.

johnny49r, 2020-08-01:
Hi,

I appreciate the heads up. I haven't installed 2.0.6 yet but I will do that
soon.

Thanks, John

codyrickman, 2020-08-09:
Hello All, 

I am using the MAX31855 which is very similar to the MAX31865 with the BigTreeTech V1.4 Turbo. 
This issue seems to be the only one I can find regarding how to wire an SPI thermopile amp to the Turbo, and im not sure if my configs are correct. 

As soon as I set #define TEMP_SENSOR_0 -3 and define my pin P0_17, the board just hangs and does not show as a serial port or SD Card drive. I have no LCD on the printer yet so I don't know if it is stuck in a loop or what. 

**Current Config:**
# Configuration.h
#define TEMP_SENSOR_0 -3

#pins_BTT_SKR_V1_4_TURBO.h
#ifndef MAX31865_CS_PIN
  #define TEMP_0_PIN P0_17
  #define MAX31865_CS_PIN TEMP_0_PIN
  #define MAX6675_SS_PIN MAX31865_CS_PIN
#endif

**This is my current wiring:**
**MAX31855** -> **SKR 1.4 Turbo**
Vin              -> ZMAX +5v
GRND         -> ZMAX GRND
DO              -> P0_17 (EXP 2)(0.17)
CS              -> P0_16 (EXP 2)(SD_SS)(0.16)
CLK            -> P0_15 (EXP 2)(MISO)(SCK)(0.15)

johnny49r, 2020-08-10:
When Marlin firmware detects overtemp it goes into shutdown and goes
offline without any warning.
I think you can disable this by commenting out the "#define
THERMAL_PROTECTION_HOTENDS" in configure.h.
This should allow you to troubleshoot your hot end temp sensor but don't
forget to restore this after you have resolved the issue.
John

codyrickman, 2020-08-10:
Thank yo @johnny49r 

I commented out the line  "#define THERMAL_PROTECTION_HOTENDS"
But unfortunately the board still wont boot/connect to serial, even if I disconnect the MAX31855. The only way I can get it up and running again is setting the -3 back to 1 for the thermistor type. 

Is there any documentation out there that shows how to use the MAX31855 to the SPI bus and configure it properly as a hotend thermistor?

code505, 2020-08-16:
> @johnny49r Thanks for the input.
> 
> ## EDIT: Since SPI is not working with analog ADC inputs (it is completely digital) whether or not the I/O line has a resistor in series with the I/O line is not a concern. I write these things out to keep them straight in my own mind.
> You could use it on the thermistor port. You would then use the thermistor port pin name as TEMP_0_PIN. Set TEMP_SENSOR_0 to -5 and set in configuration_adv.h enable TMC_USE_SW_SPI line. Then include the following in your pins_file for your processor (this way uses Software SPI):
> 
> ```
>  #ifndef MAX31865_CS_PIN
>     #define MAX31865_CS_PIN TEMP_0_PIN
>     #define MAX6675_SS_PIN MAX31865_CS_PIN
>     #define MAX31865_MOSI_PIN TMC_SW_MOSI
>     #define MAX31865_MISO_PIN TMC_SW_MISO
>     #define MAX31865_SCK_PIN TMC_SW_SCK
> #endif
> ```
> 
> If you want Hardware SPI you would have to use the true SPI CLK pin, SPI MOSI pin, SPI MISO pin (on EXP2 connector) and set the thermistor port pin name as TEMP_0_PIN. Set TEMP_SENSOR_0 to -5 and then include the following in your pins_file for your processor (this way uses Hardware SPI):
> 
> ```
>  #ifndef MAX31865_CS_PIN
>   #define MAX31865_CS_PIN TEMP_0_PIN
>   #define MAX6675_SS_PIN MAX31865_CS_PIN
> #endif
> ```
> 
> ## What I am trying to figure out now is why would I use the hardware SPI over the Software SPI on a 3D printer?
> * I have searched with google about MAX31855 (thermocouple amplifier via SPI) and MAX31865 (PT100 amplifier via SPI) and a lot of users are using the hardware SPI option to connect up their amplifier boards.  Why?
> 
> ### I tell myself this: "It takes more than 1 second for a 3D printer to change temperature on the extruder, therefore the 32-bit processor can keep up with the change in temperature without any issues (process the interrupts) so SPI protocol implemented by software (bit-banging) is good enough".
> * With the hardware SPI option you MUST find the SPI MISO, SPI MOSI, and SPI CLK lines for the SPI bus (to tap into).  Like you said you could use the EXP2 port. But, if you are using the EXP2 port for a display you will now have to create a custom cable for the EXP2 port so you can still get access to the SPI CLK line and SPI MOSI and SPI MISO lines.
> * But if you use Software SPI you can use any free digital I/O pins and assign them to be the SPI CLK, SPI MOSI, SPI MISO lines and the CS line.  Software SPI will not run as fast as the hardware SPI but the change in temperature does not occur in less than a second. On a 32-bit processor slow is really not that slow anymore.
> 
> ## @johnny49r, Yes, I see your point about the SKR V1.4 board. You would be hard pressed to find 4 free I/O lines on the SKR V1.4 board to use for Software SPI. So on the SKR V1.3/SKR V1.4/SKR V1.4 Turbo making a custom cable to the EXP2 port would be your only reasonable option. But for the SKR PRO V1.1 board you have EXTENSION-1 and EXTENSION-2 headers which give you more than enough free I/O pins so setting up Software SPI would be the easier option.

Hello,
I'm complete newbie here :)
I have SKR pro v1.1, pt100 and MAX31865 Board
I would like to connect MAX31865 using hardware SPI, I'm using tmc5160, so I'm I'll be using EXP2 to connect MAX31865, I figured out how to connect the board to MISO, MOSI and CLK to EXP2, and what I understand from your post, is that I should connect CS of the board to thermistor pin in my board (pin PF4) AND copy this to pins_BTT_SKR_PRO_V1_1.h

 ```
  #ifndef MAX31865_CS_PIN
  #define MAX31865_CS_PIN TEMP_0_PIN
   #define MAX6675_SS_PIN MAX31865_CS_PIN
 #endif
 ```

and that's it, please correct me If I'm wrong?? 
Thanks in advance

GadgetAngel, 2020-08-17:
> Hello All,
> 
> I am using the MAX31855 which is very similar to the MAX31865 with the BigTreeTech V1.4 Turbo.
> This issue seems to be the only one I can find regarding how to wire an SPI thermopile amp to the Turbo, and im not sure if my configs are correct.
> 
> As soon as I set #define TEMP_SENSOR_0 -3 and define my pin P0_17, the board just hangs and does not show as a serial port or SD Card drive. I have no LCD on the printer yet so I don't know if it is stuck in a loop or what.
> 
> **Current Config:**
> #Configuration.h
> #define TEMP_SENSOR_0 -3
> 
> From @GadgetAngel s suggestions above
> #pins_BTT_SKR_V1_4_TURBO.h
> #ifndef MAX31865_CS_PIN
> #define TEMP_0_PIN P0_17
> #define MAX31865_CS_PIN TEMP_0_PIN
> #define MAX6675_SS_PIN MAX31865_CS_PIN
> #endif
> 
> **This is my current wiring:**
> **MAX31855** -> **SKR 1.4 Turbo**
> Vin -> ZMIN +5v
> GRND -> ZMIN GRND
> DO -> P0_17 (EXP 2)(0.17)
> CS -> P0_16 (EXP 2)(SD_SS)(0.16)
> CLK -> P0_15 (EXP 2)(MISO)(SCK)(0.15)

Try this:

[image]

GadgetAngel, 2020-08-17:
> Hello,
> I'm complete newbie here :)
> I have SKR pro v1.1, pt100 and MAX31865 Board
> I would like to connect MAX31865 using hardware SPI, I'm using tmc5160, so I'm I'll be using EXP2 to connect [...]

@code505 , 

### Why make this harder on yourself.  You do not have to splice into the ribbon cable of the EXP2 port on the SKR PRO V1.1 board.  If you **must** use HARDWARE SPI instead of SOFTWARE SPI use the SPI header on the SKR PRO V1.1 board.  Since you will be using Digital I/O to get your temperature sensor data you DO NOT want to use the thermistor port.  You want to use a spare  Digital I/O pin and it so happens that the **Extension-1 or Extension-2 headers** have a lot of free ones. 

# Here is a PT100 Guide for SKR PRO V1.1 via the SPI bus:  https://drive.google.com/file/d/1zOky07F_R_DO_-XxZx6C3xULqhuxYMOi/view?usp=sharing

### May be this guide will help you.  You can download the PDF and let me know how it goes.  I get you to review  my document you get help with your problem.  We both win.

code505, 2020-08-18:
Thank you so much bro for this detailed workup, you are the best MAN.

However now I'm facing a problems, once I changed TEMP_SENSOR_0 to -5, My BTT TFT3.5 is showing no printer attached.

I have tried every thing I could find with no luck, once I return TEMP_SENSOR_0 to 1, it work flawlessly, compiling is showing no error 

regarding the MAX31865, I have decided to go the easier route with software SPI, as I'm using TMC5160, and I think I can't  use the SPI expansion while using TMC driver in SPI mode

GadgetAngel, 2020-08-18:
@code505 I found this on the web at (https://www.reddit.com/r/BIGTREETECH/comments/ew35tg/tft35_v2_with_skr14_no_printer_attached/)  A person writes the following:

> If anyone has tried all the solutions offered here, and still “no printer attached” is on the screen. Try testing the tx and rx pins for continuity between gnd. Biqu are sending me a new screen. They advised me on all the same solution posted here and when none worked they told me to test these pins on the motherboard and tft. Mine had continuity on the screen. Tx rx and gnd were shorting on all the pins

Good Luck but it beats me that setting the TEMP_SENSOR_0 to -5 would effect your TFT screen,  that make no logical sense.  I wish you could draw me up a block diagram showing me how you have hooked up your MAX31865 and how your TFT screen is hooked up to your printer.  That is just really weird.  I hope you figure out what is wrong.  When you do find out, please let me know.

code505, 2020-08-24:
I really lost hope in this thing,

I think it's not possible to run TMC5160 , MAX31865 and TFT on this board.
I wire it exactly as technique one using software SPI,  in your manual

code505, 2020-08-25:
I posted an issue here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1020

I got this answer:

> In Marlin -5 means INDEX_NONE = -5 (probe is using MAX31865 lib files) I would think you may need marlin mode to use the PT100
I searched BTT touch code for MAX31865 is it did showed MAX31865 any where so may not be supported yet.

>

GadgetAngel, 2020-08-29:
> I posted an issue here:
> [bigtreetech/BIGTREETECH-TouchScreenFirmware#1020](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1020)
> 
> I got this answer:
> 
> > In Marlin -5 means INDEX_NONE = -5 (probe is using MAX31865 lib files) I would think you may need marlin mode to use the PT100
> > I searched BTT touch code for MAX31865 is it did showed MAX31865 any where so may not be supported yet.
> 
> >

## In Platformio.ini is your default_envs=BIGTREE_SKR_PRO?  From what you sent in this forum (https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/163#issuecomment-682769998) your default_envs was set to mega2560.

code505, 2020-08-30:
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/163#issuecomment-683332881

GadgetAngel, 2020-09-28:
If anyone is having trouble with MAX31865 or PT100 working please refer to here: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/163#issuecomment-699483500

GadgetAngel, 2020-09-29:
> I really lost hope in this thing,
> 
> I think it's not possible to run TMC5160 , MAX31865 and TFT on this board.
> I wire it exactly as technique one using software SPI, in your manual

Well we got it work! see https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/163#issuecomment-698239668 on how we finally got a PT100 working with MAX31865 on an SKR PRO V1.1

oldman4U, 2020-10-10:
@GadgetAngel you and all the others are Heros!!!

👍🏻

Please help the community and close the ticket once you do not need it anymore. Thank you

GadgetAngel, 2020-11-16:
I want to add to this on how to hook up the MAX31865 boards to the SKR V1.3/ SKR V1.4 and SKR V1.4 Turbo boards. I am still testing them out on the bench but I can make them work with two Adafruit MAX31865 boards in both Hardware and Software SPI mode.