bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #24: EEPROM settings not working

p1234321, 2019-08-05:
Hello

I just tried to enable EEPROM settings and cant compile any more:

`<command-line>:0:0: note: this is the location of the previous definition
Marlin/src/HAL/HAL_STM32/persistent_store_sdcard.cpp:77:4: error: #error "Please define SPI_EEPROM (in Configuration.h) or disable EEPROM_SETTINGS."
   #error "Please define SPI_EEPROM (in Configuration.h) or disable EEPROM_SETTINGS."
    ^~~~~
*** [.pioenvs/BIGTREE_SKR_PRO/src/src/HAL/HAL_STM32/persistent_store_sdcard.cpp.o] Error 1`

please help!

Haschtl, 2019-08-05:
I had the same problem. 
I think i was able to compile by enabling 'SDSUPPORT'

golfromeo-fr, 2019-08-05:
> I had the same problem.
> I think i was able to compile by enabling 'SDSUPPORT'

maybe SDSUPPORT activate the emulation of EEPROM in a file on the SDcard (I read something related on MarlinFirmware/Marlin

p1234321, 2019-08-05:
Thanks! awesome!
now it works

closed 2019-08-05 by p1234321