bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #25: Bootloader

AnHardt, 2019-08-07:
I'm looking for the bootloader - ideally for its sources - even if not *Arduino*.

golfromeo-fr, 2019-08-12:
afaik you can only update the firmware from the sdcard as of now.
I am not sure you need a bootloader

AnHardt, 2019-08-12:
I also can upload software via ST-Link. But then the risk of overwriting the bootloader is high.
Would be nice if there would be a way to repair the board in that case. Especially for those who have not saved the bootloader before.

> I am not sure you need a bootloader

You do. The custom bootloader is the piece of software reading from SD and burning into flash. The build into all ST-processors bootloader can't do that.

----------------

The custom bootloader seems to be an interesting piece of software. According to the contained strings it's seemingly able to update the WiFi board.

golfromeo-fr, 2019-08-13:
>  WiFi board.

ok now you made the point, that's needed! no wifi firmware loaded, no wifi at all!

omahena, 2019-08-20:
I can confirm that the bootloader can update the ESD3D (or rather the ESP01S WiFi module) by placing the esp3d.bin file on the SD card. It still won't make the WiFi module work, but it results in a pretty light show.

I was actually contemplating on getting an ST-Link module to allow remote updates of firmware. Can you give any insights into ST-Link parameters / procedure? Thank you in advance.

I can't seem to convince the board to accept a firmware upload through the USB connection- not sure the BOOT0 pin and reset pins are connected to the USB serial link... Some hints here: https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/README.md#special-note-for-the-skr-pro-v11-stm32-board-1

But I also cannot get the USB drive to appear when I connect it to my PC / RaspberryPi. That actually never worked for me - not even out of the box. I always have to remove the SD card and plug it into my computer, copy the firmware.bin and back to the printer it goes... reset... wait... try... retry... It gets old fast.

Any hints would be very welcome. I am using the latest version of Marlin.

redone00, 2019-08-27:
trying to compile and send firmware to sd card get error
ploading .pio\build\BIGTREE_SKR_PRO\firmware.elf
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
embedded:startup.tcl:56: Error: Can't find scripts/interface/stlink.cfg

in procedure 'script' 
at file "embedded:startup.tcl", line 56
*** [upload] Error 1

redone00, 2019-08-27:
cannot get it to load to the sdcard seen to be not looking for the card  works fine with other firmware just not the  ran debug only a watch dog error

redone00, 2019-08-27:
I can read the usb card in a adapter but refuses to put the firmware on it get the error above

redone00, 2019-08-27:
looks like it flashed it does it need full power for the lcd to work

AnHardt, 2019-08-27:
Right. 
But seems to be a PlatfornIO setup problem. Try "start debugging" - that worked for me. Else use the card method. That works out of the box. 
To *work* with the bootloader better use ST-Link directly. At first save 0x0000 to 0x8000 somewhere - that's the bootloader. Than write protect these two blocks.

AnHardt, 2019-08-27:
> But I also cannot get the USB drive to appear when I connect it to my PC ...

For the used core not even a example composite USB CDC/MSC device seems to exist. No wonder BTT can't do that now on this board.

AnHardt, 2019-08-27:
Don't click "run" or "upload but "verify" to avoid the error. Than copy the "firmware.bin" file by hand per explorer. The file resides in ...\Marlin\Marlin\.pio\build\BIGTREE_SKR_PRO\firmware.bin now. (I'm talking MarlinFirmware/Marlin always.)

redone00, 2019-08-27:
trying it with riprap screens still waiting on my steppers

AnHardt, 2019-08-27:
For me it works with RRDFGSC or RRDSC plus ESP01S but without stepsticks powered over USB only. TFT35 3.0 is still over the ocean.

nvip12041994, 2019-09-27:
Dear, my Micro processor get issues, I need to replace new one, can I use ST link to save bootloader and upload to other processor and replace broken processor on board?

rflulling, 2019-10-13:
Hello all. I be the newest sucker to invest in this board. So it seem that they have a very real and active bootloader that is factory installed. It handles of course the firmware.bin from SD card, or rather it stores the location needed for the hardware to look for the data. Basically, it's two sets of firmware. Theirs, then ours. 

The problem I think we all see is that there is of course missing information that we all need to communicate with the Controller Card. This is at every level, effect every software we know. 

But the less discussed is that this may be less a matter of missing data and more a mater of an intentionally locked door. I cannot even talk to the chip through STM32CubeProgrammer which should be able to at least read the chip, but it cannot. Attempts to open a link are rejected or lockup. 

I think the quest we are on as users is to pursued BIGTREETECH aka BIQU to adjust the bootloader so that we are able to flash/upload firmware files, even if it is just a copy to the onboard SD card through drag and drop to the device from desktop. 

I think we can mostly agree that we do not want to pull the card every time we update firmware. I update often as Marlin is being developed. I know from experience that constant removal of any device including SD cards increases the odds of ESD, damage to the card, or the slot. I also have a SD card on the LCD that works just fine, and is far easier to access. 

From this point forward, I am going to push for this fix. I hope all of you are with me.

Franchelli, 2019-10-29:
@bigtreetech 
I hope bigtreetech does something very soon, otherwise I will have to change the card with another company's product

rflulling, 2019-10-30:
I cannot say what BTT is doing. They have ignored all attempts to contact. However, I can say that gears are turning and there are more than a few coders working on this from various angles. For the moment there is I think no more pushing I can do. I can only ask we all be patient for now. Hour by hour, they are working on the fixes. 

However, apart from whatever unknown code lurks in the BTT bootloader that we cannot know... I have noticed conversation that may indicate that BTT has left off resistors needed for communication of serial lanes. This could be to support the jumpers that switch between SD card and USB card. I don't know. With luck a IEEE gets a look at the hardware and determine if the resisters need to be added (MOD) to enable desired functions. 

Anyway. Help is coming. We just need to hold out.

sjasonsmith, 2019-10-30:
> BTT has left off resistors needed for communication of serial lanes.

If you are referring to my comments, it was regarding the TX lines documented as going to the stepper drivers. You just have to use the RX pin in half-duplex mode. TMC drivers use half-duplex serial anyway.

robustini, 2019-12-22:
@bigtreetech it would be useful if you could release the bootloader sources, thanks.

rflulling, 2019-12-22:
You may find more on this in the Marlin/Marlin git

Bob-the-Kuhn, 2020-04-01:
Thanks for the bootloader info.  

I tanked my board by using ST-Link to upload Marlin.  I have a second board so I was able to download the bootloader and move it to the tanked board.  After inserting a disk with firmware.bin on it the board is now fully functional.

FYI - here's the bootloader:
[bootloader_0x_0000_0000.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4411937/bootloader_0x_0000_0000.zip)

npmomchev, 2020-04-03:
Is possible send bootloader.bin.
Thank You

skypuppy, 2020-04-03:
What if you use the 32k offset to store your functions?  Read how klipper does it.

pecherskiy-v, 2020-04-05:
> Спасибо за информацию о загрузчике.
> 
> Я загрузил свою доску, используя ST-Link для загрузки Marlin. У меня есть вторая плата, так что я смог загрузить загрузчик и перенести его на заправленную плату. После установки диска с firmware.bin на нем плата теперь полностью работоспособна.
> 
> К вашему сведению - вот загрузчик:
> [bootloader_0x_0000_0000.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4411937/bootloader_0x_0000_0000.zip)

I tried to flash your bootloader, but got an error: "Programming error @: 0x00000000."
maybe I'm doing something wrong.
[image]

pecherskiy-v, 2020-04-05:
https://www.youtube.com/watch?v=cUn_A5N9mMc - working dump.

microhell, 2020-04-10:
> https://www.youtube.com/watch?v=cUn_A5N9mMc - рабочий дамп.
> https://www.dropbox.com/s/dd4r8n2oo4v00qn/SKR_PRO.zip?dl=0

осенило меня после просмотра твоего видео. и заливки твоего дампа(я понял что плата жива ну или по крайней мере  com port).
у меня gtr плата. в момент прошивки марлина пропало питание. и начались приключения.
при подсовывании на sd карте прошивки марлина - boot диод мигает, файл на карте переименовывается, диод остается гореть зеленым а не тухнет. плата не видит дисплей, компьютер не видит плату.
 я логично предположил что испортился bootloader. пришлось купить st-link.  но оригинальный bootloader не найдет, китайцы мило улыбаются на просьбы выслать. 
хотел уже заказывать новую плату.
не буду описывать весь процесс метода научного тыка скажу лишь выводы.
1. что бы заработал марлин, нужно привести, перед передачей управления марлину, плату в какое то определенное состояние.
2. этим занимается bootloader. иначе марлин не работает.
3. если bootloader во flash процессора есть. то он передает управление по определенному адресу и там должен быть марлин. тогда все работает даже если прописать марлин программатором. для gtr это 0x08008000.
4. сняв дамп после сбоя питания я выяснил что при перепрошивке с sd, bootloader почему то прописывал марлин с 0x0800A000. а управление передавал на 0x08008000. не совсем уверен в этом выводе но выглядит все именно так. загадочная китайская душа программиста.
5. вылечил следующим образом. нашел в дампе где заканчивается bootloader и обрезал файл.
обнулил flash. записал загрузчик с 0x080000000. и прописал марлин по 0x08008000. (можно записать с sd)

pecherskiy-v, 2020-04-13:
> осенило меня после просмотра твоего видео. и заливки твоего дампа (я понял что плата жива ну или по крайней мере com port).
> у меня gtr плата. в момент прошивки марлина пропало питание. и начались [...]

я не заметил в прошивке марлина инициализации проца, думаю как раз этим и занимается бутлоадер. с помощью st-link можно получить информацию о том что было инициализировано в момент его запуска. думаю нужно попробовать подключиться и пройти по шагам и посмотреть что делаем загрузчик или попробовать дисасамблером пройти по бутлоадеру ( но это нее самая приятная процедура )

Bob-the-Kuhn, 2020-04-13:
I don't quite know what is going on here but just in case it's of any use ...

Here are binary dumps of the bootloader section (0800 0000 - 0800 7FFF) and of all of flash with a Marlin 2.0.5.3 image in it ( 0800 0000 - 08040 3FFFF).
[BTT.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4472171/BTT.zip)

These files were created by the STM32 ST-LINK Utility using a ST-Link V2.1 dongle.

Yes, there are two levels of bootloader.  The ROM based bootloader starts the application at location 0800 0000.  In the our case a FLASH based bootloader has been programmed at 0800 0000 - 0800 552F by BTT.  It's the FLASH based bootloader that eventually starts the user application starting at 0800 8000.

---

Предоставлено Google Translate

Я не совсем знаю, что здесь происходит, но на всякий случай это пригодится ...

Вот двоичные дампы раздела загрузчика (0800 0000 - 0800 7FFF) и всего флэш-памяти с изображением Marlin 2.0.5.3 (0800 0000 - 08040 3FFFF).
[BTT.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4472171/BTT.zip)

Эти файлы были созданы утилитой STM32 ST-LINK с использованием ключа ST-Link V2.1.

Да, есть два уровня загрузчика. Загрузчик на основе ПЗУ запускает приложение в местоположении 0800 0000. В нашем случае загрузчик на основе FLASH был запрограммирован на 0800 0000 - 0800 552F с помощью BTT. Это загрузчик, основанный на FLASH, который в конечном итоге запускает пользовательское приложение, начиная с 0800 до 8000.

microhell, 2020-04-14:
> I don't quite know what is going on here but just in case it's of any use ...
 спасибо. но моя плата gtr v 1.0.  я искал bootloader от моей платы. выше я описал свою проблему с загрузчиком при пропадании питания в момент прошивки с sd. 
а pecherskiy-v  хочет прошивать marlin c помощью st-link, без bootloader от btt.

pecherskiy-v, 2020-04-14:
> I don't quite know what is going on here but just in case it's of any use ...
> спасибо. но моя плата gtr v 1.0.  я искал bootloader от моей платы. выше я описал свою проблему с загрузчиком при [...]

да все верно, я планирую прошивать его через ST-Link так как планирую производит удаленное обновление прошивки с возможностью свободно сменить прошивку например на smoothy или ещё что. и мне не удобно извлекать плату из устройства для прошивки, также наличие bootloader упрощает обновление марлина, но "закрывает" возможность работать с другими прошивками.
плату удобно использовать для целей не только 3Д печати и хотелось бы оставить возможность легко вернуть её к марлину.

closed 2024-02-12 by AnHardt