bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #27: Error on serial connection

Haschtl, 2019-08-12:
I just got my new SKR Pro 1.1 and now tried to flash the marlin firmware.

I cloned the BIGTREETECH-SKR-PRO-V1.1 marlin firmware and made some adjustments in Configuration.h, Configuration_adv.h and pins_BIGTREE_SKR_PRO_V1.1.h.

I have SDSUPPORT and EEPROM enabled. My baudrate is 250000.

When I try to connect it over serial USB with my PC (host mode), the serial device is recognized by my PC, **but** I cannot connect any client like Pronterface. Here is the serial response:

```
Connecting Machine @ USB,COM3,250000baud
 Machine connected
 echo:EEPROM version mismatch (EEPROM= Marlin=V67)
 echo:Hardcoded Default Settings Loaded
 No supported firmware detected. Closing port COM3
```

Haschtl, 2019-08-12:
This is the serial output of the arduino IDE serial monitor
```
start
echo: External Reset
Marlin bugfix-2.0.x

echo: Last Updated: 2019-07-07 | Author: HASCHTL
echo:Compiled: Aug 12 2019
echo: Free Memory: 100199  PlannerBufferBytes: 1792
echo:EEPROM version mismatch (EEPROM=? Marlin=V67)
echo:Hardcoded Default Settings Loaded
```

After this the arduino IDE gets stuck. It reacts again after I push the reset button or disconnect the USB cable

Haschtl, 2019-08-12:
I am now able to connect to the board over serial.
But I needed to disable SDSUPPORT and EEPROM_SETTINGS 

Seems that the initialization stucks at EEProm initialization. Did I miss anything?

golfromeo-fr, 2019-08-12:
somewhere it is says 115200 bauds not 250000

from memory
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1

Haschtl, 2019-08-13:
Thanks for your reply!
I can connect to the printer, when setting the serial ports the other way around:
#define SERIAL_PORT 6
#define SERIAL_PORT_2 -1

But I also need to disable 
//#SDSUPPORT
and
//#EEPROM_SETTINGS

Do you have EEProm or SD enabled?

golfromeo-fr, 2019-08-13:
> 
> 
> Thanks for your reply!
> I can connect to the printer, when setting the serial ports the other way around:
> #define SERIAL_PORT 6
> #define SERIAL_PORT_2 -1
> 
> But I also need to disable
> //#SDSUPPORT
> and
> //#EEPROM_SETTINGS
> 
> Do you have EEProm or SD enabled?

I have forgotten about something important: I am using MarlinFirmware/Marlin, no longer bigtreetech/BIGTREETECH-SKR-PRO-V1.1

from my config file
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1
(I have tried SERIAL_PORT_2  as 1 or 6)
 I have
#define SDSUPPORT
#define EEPROM_SETTINGS 
but I am not sure yet if it is eeprom emulation or real eeprom as I have a file eeprom.dat on the sdcard

Haschtl, 2019-08-13:
Ah! I didn't knew, that the SKR Pro 1.1 is already supported by the official MarlinFirmware repo!
I'm switching right now to the other repo. 

I think the default is the eeprom emulation on sdcard as you can see in pins_BIGTREE_SKR_PRO_V1.1.h:

```
// Use one of these or SDCard-based Emulation will be used
//#define SRAM_EEPROM_EMULATION   // Use BackSRAM-based EEPROM emulation
//#define FLASH_EEPROM_EMULATION  // Use Flash-based EEPROM emulation
```
I tried these options on the BIGTREETECH fork, but they didn't change anything - it was not working, but maybe it will work with the official repo

Haschtl, 2019-08-13:
I'm now using MarlinFirmware/Marlin and it is working somehow!

I think the problem was, that I had PRINTCOUNTER enabled, which caused the problems.

I have now SDSUPPORT and EEPROM_SETTINGS enabled, but after connecting over serial, marlin says:

`echo:SD init fail`

And if I try to save the config to the EEProm with M500, it says:

`echo:No EEPROM.`

Any ideas?

dant220, 2019-08-13:
Printcounter has a dependency on EEPROM according to:
https://github.com/MarlinFirmware/Marlin/pull/11689

I am sidelined right now because my board died and I am waiting for a replacement. But I also ordered an "AT24C256 I2C Interface EEPROM Memory Module". I can apparently use this as an EEPROM instead of the SDCARD.

AnHardt, 2019-08-13:
SDSUPPORT seems to work always with an external SDcard - never with the on-board.

Haschtl, 2019-08-13:
Thanks for your reply!

Do you mean, i cannot use the integrated sd-slot, which is used to Flash the stm32, for SDSUPPORT??

Haschtl, 2019-08-14:
My initial problem has been solved.

I now also enabled EEProm Emulation on SD card by uncommenting #define FLASH_EEPROM_EMULATION in pins_BIGTREE_SKR_PRO_V1.1.h, which didn't work before. (I'm now using MarlinFirmware/Marlin)

SDSUPPORT is still not possible

closed 2019-08-14 by Haschtl

dineshvyas, 2019-10-08:
FLASH EEPROM EMULATION WORKING. NO SDSUPPORT!

Baltazart, 2019-11-05:
Hello, this text is translated to English.
the following is a copy / paste.
Used on the Bigtree-tech SKR PRO V1.1 to deal with saving and remembering EEPROM settings. There was no instructions but I muddled through.

Leave WP jumper to the right, and remove all other jumpers.
Disable SRAM in **Board pins** and enable I2C EEPROM

//#define SRAM_EEPROM_EMULATION
#define I2C_EEPROM
That is all it takes.

[image]
>>> M500
SENDING:M500
echo:Settings Stored (602 bytes; crc 14044)
>>> M501
SENDING:M501
echo:EEPROM version mismatch (EEPROM=? Marlin=V68)
echo:Hardcoded Default Settings Loaded

it's just the beginning of my tests

LehrChristoph, 2020-01-11:
Hi  @Baltazart ,
I have the same board at home, do you have some documentation how to configure the I2C-EEPROM?

Baltazart, 2020-01-12:
> Hi @Baltazart ,
> I have the same board at home, do you have some documentation how to configure the I2C-EEPROM?

Hello, I configured as written on this link, https://github.com/sky8000/SKR-Pro-TMC5160-Ender-5-Config-Files-Marlin-2