bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #28: My SKR pro v1.1 with 2130 SPI steppers...

Anehall, 2019-08-12:
Hi!

Just got my Bigtreetech SKR Pro v1.1 with 2130 SPI steppers... my problem is that ive connected the 5 steppers as it show in the manual. 4 jumpers connected on the board for each axis and extruder (X, Y, Z, Z2 and E0)... But it wont work... Got the libs from here to, like it says but all 5 show this: 
testing x connection error: all high
testing y connection error: all high
testing z connection error: all high
testing z2 connection error: all high
testing e0 connection error: all high

What am I missing?

Anehall, 2019-08-12:
Forgot to mention that i baught everything from Biqu as a package... Tmc2130 v3.0 spi...

Potociny444, 2019-08-13:
Need #define TMC_USE_SW_SPI in Configuration_adv.h, because TMC is not connected to default SPI (default PB15/PB14/PB13, used for SD card connected to EXP2, declared in ..\Marlin\buildroot\share\PlatformIO\variants\BIGTREE_GENERIC_STM32F407_5X\variant.h,
but TMC uses PC12/PC11/PC10 - declared in pins_BIGTREE_SKR_PRO_V1.1.h)

Also do not use auto power control for motor power, because [https://github.com/MarlinFirmware/Marlin/issues/14558](url)

Haschtl, 2019-08-19:
I fixed this by enabling the 12/24V power supply. With only 5V the TMC (in my case 5160) SPI communication did not work for me

bidmangler, 2019-11-06:
@Potociny444 has nailed it for me (thanks). I uncommented the #define mentioned in his post and all started working. 
@bigtreetech - you could easily add this advice to your manual and save people time

GLGShadow, 2020-05-03:
So, I'm having the same issue with my motors not moving, and I also have 2130s but don't understand the explanation/fix. Help?