bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #29: SKR PRO v1.1 w/TMC2130 = no stall guard

chasinreno, 2019-08-13:


chasinreno, 2019-08-13:
I have my corexy printer xyz moving in the correct directions and have setup the board for TMC2130 SPI according to the manual.  I also have Marlin 2.0-bugfix setup for sensor less  homing but stall guard doesn't work.
Has anyone gotten this work?  Am I missing a library?
configuration.h & configuration_adv.h attached.

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3498786/Configuration.zip)

Potociny444, 2019-08-14:
Try increasing the sensitivity by reducing the STALL_SENSITIVITY value. In my setup is working value '1'

Evovita, 2020-01-18:
Experiencing same issue as @chasinreno. I also tried setting the STALL_SENSITIVITY value to 1 without success. Running Marlin 2.0.1, with TMC2130 V3.0 SPI drivers on SKR Pro V1.1 board.

oldman4U, 2020-06-08:
Hello chasinreno.

Seems this ticket has not been used for a while, so please close it.

Thank you

chasinreno, 2020-06-09:
Thanks for the help. I eventually went back to sensors.  I got it working at a sensitivity setting of 6 but without notice it just quit working again.

closed 2020-06-09 by chasinreno

oldman4U, 2020-06-09:
Hi.

Maybe it would be worth to give it a try using a newer version of Marlin. I mean it is almost a year ago and Marlin has changed a lot since then.

But the most important thing is that you can use your printer and that you enjoy your hobby.

Thank you for closing the ticket

;-)