bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #30: What this warning ?

budi67, 2019-08-15:
Hello all

I juste received my SKR Pro and when i compile Marlin 2.0, it compile with success but i have a lot af this warning.
Have you an idea ?

A big thanks

PS : sorry for my english

[image]

AnHardt, 2019-08-15:
Try

```diff
+ #ifndef BOARD_NAME
- #define BOARD_NAME "BIGTREE SKR Pro 1.1"
+  #define BOARD_NAME "BIGTREE SKR Pro 1.1"
+#endif
```
in 'pins_BIGTREE_SKR_PRO_V1.1.h'.

budi67, 2019-08-15:
A big thanks AnHardt, it's work fine ;)

omahena, 2019-08-20:
I do wish BTT would maintain the board support for Marlin 2.0.

oldman4U, 2020-06-08:
Hello budi67.

Could you please close this ticket. Seems it is not needed anymore.

Thank you