bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #31: 3D Touch stows, when homing

christianger, 2019-08-15:
Hello guys,

i connected my 3DTouch to my SKR PRO as shown in the manual.
I have also connected a Z-endstop.

When i go into the config menu and deploy the 3DTouch manually it deploys normally and M119 says the probe is not triggered.
Manually stowing work also as i expect it and M119 returns triggered the whole time. I think this is also not correct. It should send triggered just as a short pulse.

But when i press homing, the printer homes XY first and then stows the 3DTouch and moves the Z-axis downwards.
I tried everything, but i cant get the 3DTouch to deploy while homing.

christianger, 2019-08-15:
M43 S
Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PROBE_PIN: 11
. Z_MIN_PROBE_ENDSTOP_INVERTING: false
. Check for BLTOUCH
= BLTouch Classic 1.2, 1.3, Smart 1.0, 2.0, 2.2, 3.0, 3.1 detected.
** Please trigger probe within 30 sec **
. Pulse width (+/- 4ms): 4
FAIL: Noise detected - please re-run test
ok

budi67, 2019-08-16:
Hello

How do you active M43 ?
I have try to active #define PINS_DEBUGGING, but i have this error !!
Thanks

[image]

golfromeo-fr, 2019-08-16:
> 
> 
> Hello
> 
> How do you active M43 ?
> I have try to active #define PINS_DEBUGGING, but i have this error !!
> Thanks

try to use MarlinFirmware/Marlin instead
at least the PINS_DEBUGGING option compiles

budi67, 2019-08-16:
I have already try to compile with the Marlin Firmware but i always have error with TMC26x

golfromeo-fr, 2019-08-16:
> 
> 
> I have already try to compile with the Marlin Firmware but i always have error with TMC26x
uninstall platformio and close vscode. 
save your files Marlin\Configuration.h Marlin\Configuration_adv.h
remove c:\Users\YourName\.vscode and c:\Users\YourName\.platformio folders
restart with a fresh copy of Marlin with a clean platformio.ini 
open vscode and install platformio 
put in platformio.ini the line 
"default_envs  = BIGTREE_SKR_PRO"

budi67, 2019-08-16:
Thanks golfromero-fr 
Soryy for this question but how do i close vscode ?

golfromeo-fr, 2019-08-16:
> 
> 
> Thanks golfromero-fr
> Soryy for this question but how do i close vscode ?

I mean : closing the application 
File > Exit

budi67, 2019-08-16:
I have download a new Marlin/firmware and it compile fine (i don't know while)
My bfptouch work fine, G29 too

But as the SKR Pro as no Eeprom, where to save the Mesh ?

Servo probe test
. using index:  0, deploy angle: 200, stow angle:   70
. Probe Z_MIN_PIN: 106
. Z_MIN_ENDSTOP_INVERTING: false
. Deploy & stow 4 times
= Mechanical Switch detected
  DEPLOYED state: LOW (logic 0)  STOWED (triggered) state: HIGH (logic 1)

budi67, 2019-08-16:
My G29

      0      1      2      3      4      5      6
 0 +0.449 +0.033 -0.341 -0.583 -0.652 -0.643 -0.602
 1 +0.488 +0.157 -0.123 -0.294 -0.368 -0.371 -0.372
 2 +0.509 +0.227 +0.002 -0.155 -0.225 -0.240 -0.241
 3 +0.547 +0.328 +0.173 +0.027 -0.044 -0.100 -0.169
 4 +0.549 +0.369 +0.244 +0.101 +0.021 -0.042 -0.097
 5 +0.557 +0.419 +0.312 +0.182 +0.107 +0.020 -0.090
 6 +0.482 +0.353 +0.246 +0.131 +0.062 -0.029 -0.120

golfromeo-fr, 2019-08-16:
@christianger 
issue solved on my side: the connector on the bltouch (clone) was half disconnected, the signal was connected but no longer connected the z- pin

budi67, 2019-08-16:
@christianger 
yes whith your help and another I’m almost at the end

basilfx, 2019-08-30:
@christianger I have the same issue, and this is what they told me on the Marlin support channel (Discord):

"You're trying to do conflicting things. If you're homing with the BLTouch you want Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to be set and you want to connect the BLTouch to the z-min pin. If you're homing with Stallguard (Z_MIN_PIN) you don't want the BLTouch deploying when homing Z - only when using it to create a leveling mesh."

So, it's not used during homing.

christianger, 2019-08-31:
i thought it would make no difference, whether i use the Z-Min Pin with Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN option
or the PG8-Pin.

But using the Z-Min Pins solved my problem.

closed 2019-08-31 by christianger