bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #32: Skr Pro v1.1 from GitHub not working

Formation3D, 2019-08-16:
Hello, I am having some issues trying to get the bigtreetech SKR Pro V 1.1 master firmware from GitHub 2 work properly on Visual Studio code. I'm not sure why I'm having such a major issue with this. I have deleted all files and reinstalled all the programs a couple of times. I have looked at different stuff on the forms to try to figure it out and I'm having issues still. For whatever reason I get different errors at different times and my summary ends up being the big tree SKR pro has failed in the environment. Attached are the photos from the terminal and the problems that arise when I try to build Marlin. I'm in need of assistance if anybody has time that they would be able to help me out with the situation.
[image]
[image]

Evg33, 2019-08-16:
https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.0.x

[image]

Potociny444, 2019-08-16:
Try run VSCode as administrator

Evg33, 2019-08-16:
Use short path to sources.
Move source to c:\Marl\ as example.

golfromeo-fr, 2019-08-16:
you can try this
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/31#issuecomment-521911559

NickMilton, 2019-08-17:
I used the Marlin supplied Marlin 2.0, (not the BBT supplied one).

Of course I had to set up the configuration files myself but it all part of the learnign expereince.

[Configs.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3511466/Configs.zip)

Works fine

Formation3D, 2019-08-17:
Ok, so I have done everything so far that was given to me as advice. Python and Git are both working properly. This is where I am at. I am only working with Marlin 2.0 bug fix. I was trying to use the Marlin by GitHub but was getting to many errors with it. I figure I could build in Marlin everything I need. So, I tried to set Marlin up for my board, the SKR Pro v1.1... once I configured the basics I hit build. I get a couple of errors. The main one is the environment, its saying that BIGTREE_SKR_PRO is a bad env, and that I am to use LPC1768. Now, the funny thing is when I changed the env to the LPC1768 it built the program and it was successful. But this is not my board, and I am confused as to why changing one thing like that it will complete. In closed is some photos of the completed build and the failed build with my board in. 
[image]
[image]
[image]
[image]

NickMilton, 2019-08-17:
The  LPC1768 is for another board/chipset. You must define
#define MOTHERBOARD BOARD_BIGTREE_SKR_PRO_V1_1

Formation3D, 2019-08-17:
NickMilton, funny you say that, because in both configurations MOTHERBOARD BOARD_BIGTREE_SKR_PRO_V1_1 is defined as my board in cofiguration.h. All I did was change the environment. That's why I said I dont understand why changing the environment to LCP1768 it works and if I have the env set to BIGTREE_SKR_PRO it dont work.

NickMilton, 2019-08-18:
Ahh...
I defined environment as 
default_envs = BIGTREE_SKR_PRO

Noting that platformio.ini has...
#
# BIGTREE_SKR_PRO (STM32F407ZGT6 ARM Cortex-M4)
#
[env:BIGTREE_SKR_PRO]
platform = ststm32@5.4.3
framework = arduino
board = BigTree_SKR_Pro
extra_scripts = pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
build_flags = ${common.build_flags}
  -DUSBCON -DUSBD_USE_CDC -DUSBD_VID=0x0483 -DUSB_PRODUCT=\"STM32F407ZG\"
  -DTARGET_STM32F4 -DSTM32F407_5ZX -DVECT_TAB_OFFSET=0x8000 -DHAVE_HWSERIAL6
lib_deps = ${common.lib_deps}
lib_ignore = Adafruit NeoPixel, c1921b4, TMC26XStepper, SailfishLCD, SailfishRGB_LED, SlowSoftI2CMaster
src_filter = ${common.default_src_filter} +<src/HAL/HAL_STM32>
monitor_speed = 250000

Formation3D, 2019-08-18:
Got it, thanks for all your help. Now to try an configure the rest in Marlin 2.0

closed 2019-08-18 by Formation3D