bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #33: ESP-01S WIFI Module ESP8266 For SKR PRO - No setup guides or documentation?

NickMilton, 2019-08-17:
Problem - lack of documentation so cannot use the WSP-01S WiFi module. The SKR PRO itself is a great board and working properly with Marlin 2.0.

Testing to date - Successfully installed the ESP-01S onto the SKR PRO and loaded the BBT supplied esp3d.bin via the microSD card. I also set up Configuraiton.h for the SKR PRO with  
#define SERIAL_PORT_2 6 // (1 for BIQU TFT or 6 for WiFi adapter?)

Problem - The WiFi module blue LED comes on but that's it - what do we do next? How do we access the WiFi and how do we use it!

PS I  am aware of:
https://github.com/luc-github/ESP3D
and
https://github.com/luc-github/ESP3D-WEBUI

but have not gone done that track yet!
kind regards

NickMilton, 2019-08-21:
Bigtreetech - can you please give us some guidance or recommendaitons on this? Loaded with the BTT supplied binary, the WSP-01S does not appear as an access point so there in nothing to wirelessly connect to....

Appreciated

NickMilton, 2019-08-23:
I slogged through this problem over several nights. Now I have an SKR Pro serving webpages to my local network. Way better than a notebook PC and USB cable...

MASSIVE THUMBS UP AND GREAT THANKS to Chris Riley - without who I would have given up!

https://www.youtube.com/watch?v=pJGBRriNc9I

Attached are some notes I made - in part from following Chris' video and in part from the https://github.com/luc-github/ESP3D/wiki/Install-Instructions

Go welll

[Nick guide R000.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3534588/Nick.guide.R000.pdf)

closed 2019-08-31 by NickMilton

rlvanhuis, 2019-09-14:
https://github.com/NickMilton
Many thanks for the guide.  After installing and connecting over WiFi, it appears I am only able to get about 1 command per minute to the printer.  I have verified the com rate at 115200 for both the ESP and the main board.  Have I missed anything?

NickMilton, 2019-09-14:
Hmmm

Some commands of course need to complete (eg G29 - a 25point bed level process) before the printer willl act on a subsequent command (eg change hot end temperature). It could be something else too...

I have 2 printers with the BIQU SQK Pro v1.1board. One of them had similar 'slow' WiFi problems - which seemed to 'go away' after plugging the WiFi module in and out a few times. May have been poor electrical contact??? Never found cause.

rlvanhuis, 2019-09-15:
I will try.  I have all but given up on getting the ESP to work on this
board.  Pretty certain I am enhancing the definition of insane.

Thanks for the feedback.

NickMilton, 2019-09-15:
Hmmm.....

Maybe try reloading the ESP firmware and file system. Maybe try a different
ESP module in case yours is faulty.

In general I would also check how clean your power supply is. Ripple can do
weird thinks. Run off a battery to test ripple free power.

Just a thought

Snakefreak, 2019-10-08:
With esp3d loaded on the esp-01 only red light.   after plugging in.  We have to set enable pin high or something else?

NickMilton, 2019-10-08:
Don't really have enough information from your description of the problem to help...

Snakefreak, 2019-10-09:
Hello Nick my esp01 modul is prepared runs fine on usb serial converter.       when i plug the modul in the 8pin heade,r only the red led lights up.  the ch_pd (enable) pin is not on a high level. (should be high for opperation)  its PG1/FSMC_A11 on the stm32

Snakefreak, 2019-10-13:
Found the error my module is a esp-01   not a esp-01s  so i have to manual pull the pin high trough a 12k resistor  or  make the boot up trough marlin aviable.

jbaribeault, 2019-10-16:
What's the bootup option through marlin look like?

Snakefreak, 2019-10-16:
PIN pg1 high  think someone can code it in the skr pro 1.1 PIN Definition.   Should be compatible with both esp01 and esp01s .

kamahat, 2019-12-02:
I had an issue with esptoll 2.6.2 , I was not able to push it via de Ardunio IDE. 
So i've done it manually from the directory where the esptool.py is installed. 
Using the infromation from the error windows in the IDE (port, flash start, pathc to bin file)
`cd ~/.arduino15/packages/esp8266/hardware/esp8266/2.6.2/tools/esptool
python ./esptool.py --before default_reset --after hard_reset --port /dev/ttyUSB0 write_flash 0xBB000 /tmp/arduino_build_373729/esp3d.spiffs.bin
`

xkongox, 2020-04-29:
Hi, I have SKR 1.4 mainboard with Marlin 2.0.5 + BT touch and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
I followed instructions with changing options in HardwareSerial.h.
I have tried install ESP-01S. It works in programator I can login via WIFI. But when install to the board, sth goes wrong. Sympthoms.

1. Normally during booting LCD buzzer is loud and annoying. With WIFI it is rather silent, so I suspect it is some power problem or short circuit.
2 .Menu is jumping from Info screen to main screen. I hear clicking and it is loop. I can't stop it.
3. ESP-01S within 10-15 seconds is extremely hot.
When I remove module, all works fine. When I change HardwareSerial definition to default, the problem is still excists.

Maybe somone can say what is relation between LCD and WIFI on SKR 1.4? I work on 12V power.

mattster98, 2020-06-09:
> PIN pg1 high think someone can code it in the skr pro 1.1 PIN Definition. Should be compatible with both esp01 and esp01s .

Looks like the fix is making into the base Marlin repo.  Going to try and merge it into this one and see if it works. :)

https://github.com/MarlinFirmware/Marlin/pull/17679

rudranshsinghmahra, 2024-10-12:
> Found the error my module is a esp-01 not a esp-01s so i have to manual pull the pin high trough a 12k resistor or make the boot up trough marlin aviable.

I have the same issue I have SKR 1.4 board and ESP01 wifi module, module turn on but no wifi light. Between which two pins I have to solder the 12K resistor? If you could share the image it would be great

Snakefreak, 2024-10-12:
> I have the same issue I have SKR 1.4 board and ESP01 wifi module, module turn on but no wifi light. Between which two pins I have to solder the 12K resistor? If you could share the image it would be [...]

PG01 from mcu goes to  WiFi en in esp01 Modul (pin3). So solder IT from Pin 3 to vcc 3.3v pin 7 in esp01 Modul

rudranshsinghmahra, 2024-10-12:
> PG01 from mcu goes to WiFi en in esp01 Modul (pin3). So solder IT from Pin 3 to vcc 3.3v pin 7 in esp01 Modul

Thanks for your reply. I am too weak at electronics. I am sharing the image of the connection I could understand from your reply. Please let me know if it is correct.

[image]

Snakefreak, 2024-10-12:
Yes itstge right place

rudranshsinghmahra, 2024-10-12:
> Yes itstge right place

Thanks, will let you know if it works or not.

rudranshsinghmahra, 2024-10-14:
> Yes itstge right place

Thanks, It's working perfectly. I need a little help from you on my other issue.
The issue I am facing while using the SKR 1.4 board is that when I connect an **optical end-stop** to the X, Y, or Z end-stop port it does not trigger. The red LED on the optical end-stop remains constantly illuminated and the end-stop itself is not functioning as expected it never triggers and always shows open while using M119 command in pronterface.

I have tested the same setup with a **microswitch end-stop, and it works without any issues**. I have double-checked the wiring for the optical end-stop, and everything seems to be correct. I am using this optical endstop.
[image]

Could you please provide guidance on how to troubleshoot this issue or suggest any steps to resolve it?