bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #34: Z Homeing not working

bmarl, 2019-08-18:
This happens with both the original Marlin 2.0 branch as well as this.
On sending G28 X1, X2 and Y  home fine, Z moves up, then down and triggers the endstop and just stops. Only after a reset Z move again and only on homing or if i set the Z position to something negative.
After exchanging Y/Z motor and endstop for a test i can rule out electrical problems.

bmarl, 2019-08-19:
Just for clarification, i don't have BLTouch or similar activated.

bmarl, 2019-08-19:
The issue seems to be the board itself. The same Marlin version works fine on the printer with the SKR 1.3

oldman4U, 2020-06-08:
Hi bmarl.

Do we still need this ticket? If not, it would be great if you could close it.

Thank you

closed 2020-06-15 by bmarl