bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #35: Printing from SD-Card

Haschtl, 2019-08-19:
As 
> SDSUPPORT seems to work always with an external SDcard - never with the on-board.
> 
> _Originally posted by @AnHardt in https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/27#issuecomment-520928652_

How is it possible to print from SD-Card with the SKR Pro 1.1? 
The internal SD-slot does not work (till now?)
I also have an SD-expansion board with SPI-communication, but i'm using the TMC5160, which only support SPI mode, and as the manual says:

> The SPI expansion port shares its I/O pins with the SPI pins of the stepper motor drivers. It can only be used if you are not using a stepper motor driver in SPI mode.

So in the end it is not possible to use an SD-card with the SKR-Pro?

AnHardt, 2019-08-19:
Please verify in the [schematics](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/Schematic/SKR-PRO-V1.1%EF%BC%88SCH%EF%BC%89.pdf).
External SD-card (on display panel) schould connect to EXP2-port.
This is not the same SPI-bus as on the dedicated SPI-connector.
The dedicated SPI-port has the same bus as the stepper sockets, but not the same as EXP2 socket.

omahena, 2019-08-20:
OK, great, so it is at least possible in HW. Any clues on how to configure and use either the SD card in the BTT TFT35 3.1 or the on board one? The pins file seems very lacking....

Evg33, 2019-08-21:
@omahena You can see pins on scheme
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/Schematic/SKR-PRO-V1.1%EF%BC%88SCH%EF%BC%89.pdf

  SD_DETECT   PB11
  SCK        PA5
  MISO         PA6
  MOSI         PB5
  SS               PA4

Haschtl, 2019-08-23:
Thanks!
This solved my issue.

closed 2019-08-23 by Haschtl

omahena, 2019-08-23:
@Evg33 Thanks for the pinout and the SD_DETECT hint. Since I am not sure we have a clear understanding, what I was talking about is the Marlin **pins** file:
`Marlin\src\pins\stm32\pins_BIGTREE_SKR_PRO_V1.1.h` 
file in particular.

Let's say I would like to enable Marlin to use the on board SD Card. How does one do that?

Since the `pins_BIGTREE_SKR_PRO_V1.1.h `file has just the very basics in there and if I look at `pins_BIGTREE_SKR_MINI_V1.1.h` for inspiration... The following block seems appropriate:

```
  #define SDCARD_CONNECTION ONBOARD
  #define ENABLE_SPI1
  #define SD_DETECT_PIN    PB11
  #define SCK_PIN          PA5
  #define MISO_PIN         PA6
  #define MOSI_PIN         PB5
  #define SS_PIN           PA4
```

The ENABLE_SPI1 is probably wrong or even without meaning. I see no mention of it in the Marlin source files or any other file in the entire Marlin repository. This brings doubts into how legit the pins_BIGTREE_SKR_MINI_V1.1.h file is as well... But being new to the Marlin scene, please correct me if I am wrong.

Next if we look at the SDCARD_CONNECTION usage... It is also just mentioned in mostly config and verification headers:
```
C:\Marlin>findstr /s "SDCARD_CONNECTION" *.h | find /v "examples" | find /v "src\pins"
config\default\Configuration_adv.h:  #if HAS_SDCARD_CONNECTION
config\default\Configuration_adv.h:    //#define SDCARD_CONNECTION LCD
Marlin\Configuration_adv.h:  #ifdef HAS_SDCARD_CONNECTION
Marlin\Configuration_adv.h:    #define SDCARD_CONNECTION ONBOARD
Marlin\Configuration_adv.h:  #endif // HAS_SDCARD_CONNECTION
Marlin\src\core\drivers.h:#define SD_CONNECTION_IS(V) (_SDCARD_ID(SDCARD_CONNECTION) == _SDCARD_ID(V))
Marlin\src\inc\Conditionals_LCD.h:#define HAS_SDCARD_CONNECTION EITHER(TARGET_LPC1768, ADAFRUIT_GRAND_CENTRAL_M4)
Marlin\src\inc\SanityCheck.h:  #error "LPC_SD_(LCD|ONBOARD|CUSTOM_CABLE) are now SDCARD_CONNECTION. Please update your Configuration_adv.h."
```

And no cpp files:
```
C:\Users\alesom\source\SKR Pro 1.1\firmware\Marlin>findstr /s "SDCARD_CONNECTION" *.cpp
```

Furthermore it looks like the condition for the HAS_SDCARD_CONNECTION is limited to LPC1768 and ADAFRUIT_GRAND_CENTRAL_M4:
```
#define HAS_SDCARD_CONNECTION EITHER(TARGET_LPC1768, ADAFRUIT_GRAND_CENTRAL_M4)
```
Hmm... But... There is a `Marlin\Marlin\src\sd\cardreader.cpp` file that gets compiled if `SDSUPPORT` is defined, which I enabled in my `configuration.h`.

Unfortunately the `CardReader::initsd()` function aborts with an error. After a few hours of log based instrumentation and carrying the SD card between my computer and the board... I can see the `sd2card.init(SPI_SPEED, SDSS)` fails to initialise the SPI connection to the SD card and returns an `SD_CARD_ERROR_CMD0`.

```
// Set pin modes
  extDigitalWrite(chipSelectPin_, HIGH);  // For some CPUs pinMode can write the wrong data so init desired data value first
  pinMode(chipSelectPin_, OUTPUT);        // Solution for #8746 by @benlye
  spiBegin();

  // Set SCK rate for initialization commands
  spiRate_ = SPI_SD_INIT_RATE;
  spiInit(spiRate_);

  // Must supply min of 74 clock cycles with CS high.
  for (uint8_t i = 0; i < 10; i++) spiSend(0xFF);

  // Initialization can cause the watchdog to timeout, so reinit it here
  #if ENABLED(USE_WATCHDOG)
    watchdog_reset();
  #endif

  // Command to go idle in SPI mode
  while ((status_ = cardCommand(CMD0, 0)) != R1_IDLE_STATE) {
    if (ELAPSED(millis(), init_timeout)) {
      error(SD_CARD_ERROR_CMD0);
      goto FAIL;
    }
  }
```

`SD init fail. CPU Pins: SS: P13 SCLK: P14 MISO: P15 MOSI: P46 Code: 0x01`

Any ideas?

@Haschtl How did you make it work?

Evg33, 2019-08-23:
Maybe try to change the speed in Configuration.h ?

```
/**
 * SD CARD: SPI SPEED
 *
 * Enable one of the following items for a slower SPI transfer speed.
 * This may be required to resolve "volume init" errors.
 */
//#define SPI_SPEED SPI_HALF_SPEED
//#define SPI_SPEED SPI_QUARTER_SPEED
//#define SPI_SPEED SPI_EIGHTH_SPEED

```

@Haschtl How did you make it work?

Haschtl, 2019-08-23:
I had an old SDCard expansion from my old RAMPS lying around, which i connected to the EXP2-port, as @AnHardt clearified.

I still would like to use the onboard sd-slot, but for now it's ok

omahena, 2019-08-24:
@Haschtl Ah, of course. The EXP2 port SD card alternative. I didn't think of that being too focused on the on-board SD card.

I also read what @AnHardt wrote. If I understand him correctly there are three HW SPI buses on the system. SPI1 is dedicated for motor control (as well as the SPI dedicated port on the board that is just exposing the motor control SPI bus for external wiring or alternative use if SPI is not used to control motors). 

Then SPI2 is routed to the EXP2 port to allow connections for (an LCDs with) a SD card or just an external SD card like you are using. 

And SPI3 seems to be dedicated to the on-board SD card... That fails to initialise. I have a feeling the SPI3 is in conflict with some CPU feature that was turned on first and that is preventing successful initialisation of the SPI3 bus.

omahena, 2019-08-24:
@AnHardt Considering what we learned in issue #36 ... I tried cleaning up the SPI section.

I think this is correct now:

```
//*** SPI ***

#ifdef HAL_SPI_MODULE_ENABLED
const PinMap PinMap_SPI_MOSI[] = {
  //{PA_7,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  {PB_5,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)}, // TF_MOSI
  //{PB_5,  SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  {PB_15, SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},   // EXT2_MOSI
  //{PC_3,  SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {PC_12, SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},   // MOTO_MOSI
  {NC,    NP,    0}
};
#endif

#ifdef HAL_SPI_MODULE_ENABLED
const PinMap PinMap_SPI_MISO[] = {
  {PA_6,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)}, // TF_MISO
  //{PB_4,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  //{PB_4,  SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  {PB_14, SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)}, // EXP2_MISO
  //{PC_2,  SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {PC_11, SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)}, // MOTO_MISO
  {NC,    NP,    0}
};
#endif

#ifdef HAL_SPI_MODULE_ENABLED
const PinMap PinMap_SPI_SCLK[] = {
  {PA_5,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},  // TF_SCLK
  //{PB_3,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  //{PB_3,  SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  //{PB_10, SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {PB_13, SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},  // EXP2_SCLK
  {PC_10, SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},  // MOTO_SCLK
  {NC,    NP,    0}
};
#endif

#ifdef HAL_SPI_MODULE_ENABLED
const PinMap PinMap_SPI_SSEL[] = {
  {PA_4,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},   // TF_SS?
  //{PA_4,  SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  //{PA_15, SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  {PA_15, SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},   // MOTO_SS
  //{PB_9,  SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {PB_12, SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)}, // EXP2_SS
  {NC,    NP,    0}
};
#endif
```

Unfortunately still no joy on the SD card SPI. It looks like SD card is actually on SPI1 and not SPI3 as I thought since I could only resolve the pins from the schematic in this way.

Everything else appears to still be working (at least motor control). Any more ideas?

@Evg33 I have tried half speed already but not any of the slower speeds. Will do that when I get the next timeslice.

omahena, 2019-09-11:
@Evg33 Tried Full, Half, Quarter and Eigth. No joy.

I realised I forgot a lot of the other pin conflicts outside the SPI section, so I cleaned those out as well.

Yesterday a completely new pin definition was pushed to this repository in the `variant.h`. Here are just a few of the changes:

[image]

The pin number changes include the pins for the SD card... So until they figure out the pinout of the CPU they are using I am no longer investing my time into this.

piotrekm7, 2019-10-11:
@omahena 
Any progress with it ? I'm struglling too.

omahena, 2019-10-13:
@piotrekm7 Sorry, Piotr, I gave up on the issue. I finished cleaning up the `buildroot\share\PlatformIO\variants\BIGTREE_GENERIC_STM32F407_5X\PeripheralPins.c` and removed all overlapping pins and kept only the stuff that the SKR Pro 1.1 uses... And there was no improvement. Just a huge waste of time.

I also still have "entangled" HEAT0 and HEAT1 mosfets - both get driven by the HEAT0 logic. This also kind of puts a damper on my dual extruder plan.

I did just receive two SD card expansion boards and thought I'd give them a try some time this week.

So from my perspective, we should just give realistic feedback when rating BTT products and perhaps then BTT will listen to their customers and deliver what they promised.

jbaribeault, 2019-10-17:
Any luck with the external SD expansions? I'm trying to get one working on what amounts to SPI3 since I'm using UART for the steppers (2209) and not SPI, I've assigned:
```
// Custom SDCard on SPI3
#define SCK_PIN          PC10
#define MISO_PIN         PC11
#define MOSI_PIN         PC12
#define SS_PIN           PA15
```
But no joy....

AnHardt, 2019-10-17:
@jbaribeault 
That's likely the same problem as when trying to redirect the SD to the on-board-slot. If the one is solved also the other.

piotrekm7, 2019-10-17:
I received this reply from customer service:
"Hello，friend.
Sorry, the onboard SD card printing function and the SD card printing function of the display cannot be in the same firmware. If you want to print on an onboard SD card, you will need to modify the firmware.We have no such revision manual yet, sorry."

I have to admit I'm disappointed of this purchase.

jbaribeault, 2019-10-17:
That implies that the onboard function can work....

skypuppy, 2019-10-21:
I tried to avoid all that development with no documentation headache by using octoprint and letting the RPi handle the files.

piotrekm7, 2019-10-21:
@skypuppy I decided for the same solution. I recommend.

jbaribeault, 2019-11-07:
So far that's my solution as well, but I'd love to be able to use an SDCARD - whether internal or external for long printjobs. The Serial tends to drop randomly sometimes ( I think I've got a loose/badly soldered  USB port on the board)

YaMoef, 2019-12-26:
Hello as I bought this board I'm also expirementing with it to get the onboard sd working. So the only thing I got working is that the lcd of my printer shows media insterted when I put my sd in and media removed when I remove the sd card. What I did is put the SD_DETECT_PIN to PB11 and comment the SD_DETECT_INVERTED. But I'm think at this point maybe the pins are not properly configured in the files of marlin since I saw many people use other pins like I did and I lookt into the schematic. What I've got is:
SPI1 (onboard TF)

	MOSI	PB5
	SS	PA4
	MISO	PA6
	CLK	PA5

	TF_SW	PB11

SPI2 (external connector)

	MOSI	PB15
	SS	PB12
	MISO	PB14
	CLK	PB13
	
	EXP2_SW	PF12

SPI3 (TMC SPI)
	
	MOSI	PC12
	SS	-> CS	PA15
	MISO	PC11
	CLK	PC10

Can anyone confirm is this is correct? Also is it possible right now to use an external sd slot? because if you maybe change the spi pins from the external sd to the onboard sd you maybe could get it working. Also I found in the marlin folder some files which look very interesting to me: I've found some spi pins in Marlin/src/HAL but I'm not sure which on the SKR Pro uses. Is it also maybe possible to define the spi pins in the pins file in some way like @omahena did? I think this could also make it a bit easier.

YaMoef, 2019-12-26:
Also I forgot to mention:
-I use the Marling bugfix 2.0 
-when I instert the SD card it still shows no media
-I put the SD connection to onboard

Evg33, 2019-12-27:
https://github.com/MarlinFirmware/Marlin/pull/16260

YaMoef, 2019-12-27:
Thank you this solves a lot! I didn't think about that it would be this hard to fix the problem.