bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #36: M42 I P30 S1

tna11, 2019-08-20:
By entering M42 I P30 S1, both Heater2 and Fan0 come on, M42 I P30 S0 switches both off. Only Fan0 should come on. 

Entering M42 I P19 S1, Heater 2 only flashes on and off. 

Any idea what causes this? All the other heaters and fans are OK using the M42 command

omahena, 2019-08-21:
Hmm... On my SKR Pro 1.1 the 
`M42 I P30 S1`
will only turn on the FAN0 (Part cooling fan in my case) to the minimal PWM level - the fan actually stands still as there is not enough power to turn it. Setting it to 255 will turn it on full blast:
`M42 I P30 S255`

I have also verified that no other fans or heaters come on. And LED indicators turn on as expected for all pins:
```
P19	HEAT2
P20	HEAT0
P30	FAN0
P71	HEAT1
P54	FAN1
P55	FAN2
```
So if what you describe happens on your board it is definitely not normal behaviour. I don't think it is firmware related since I believe that pin functions can only be defined in software in terms of what they do... And it is not possible to configure pins to bind two of them together. But... Since you are using firmware that is interpreting gcode... it is in theory possible that the firmware will not allow you to turn on the fan without the heater. Although assuming you are running an extruder on Heater2 then one would expect the Fan2 to come on and also that this would only happen when you turn on Heater2 GPIO pin P19.

Sigh... Actually... Just as I have written all that, I did a board reset and I was retesting the P30 pin it started to behave the way you describe. And now it works correctly again after a second restart and some more gcode...

OK.. I may have discovered a pattern. After a fresh reset the 
```
M42 I P30 S1
M42 I P30 S255
M42 I P30 S0
```
works as expected. Then if you do a `M42 I P19 S1` by accident... P19 gets "bound" to P30. So when you are controlling the P30 it will also turn on the Heater2 and provide 12V on the heater pins. It looks like we won't be able to use that third extruder after all... Then again we are also out of fan connectors for cooling it. :P

Looking at the Marlin code:
```
void GcodeSuite::M42() {
  if (!parser.seenval('S')) return;
  const byte pin_status = parser.value_byte();

  const int pin_index = PARSED_PIN_INDEX('P', GET_PIN_MAP_INDEX(LED_PIN));
  if (pin_index < 0) return;

  const pin_t pin = GET_PIN_MAP_PIN(pin_index);

  if (!parser.boolval('I') && pin_is_protected(pin)) return protected_pin_err();

  pinMode(pin, OUTPUT);
  extDigitalWrite(pin, pin_status);
  analogWrite(pin, pin_status);

  #if FAN_COUNT > 0
    switch (pin) {
      #if HAS_FAN0
        case FAN0_PIN: thermalManager.fan_speed[0] = pin_status; break;
      #endif
      #if HAS_FAN1
        case FAN1_PIN: thermalManager.fan_speed[1] = pin_status; break;
      #endif
      #if HAS_FAN2
        case FAN2_PIN: thermalManager.fan_speed[2] = pin_status; break;
      #endif
    }
  #endif
}
```
I can see that you are bypassing the pin protection with the undocumented I(gnore) parameter? We can also see that thermalManager is concerned about the fan speed. So thermalManager is likely binding the fan with the heater trying to protect you from burning down your house in a clumsy and incorrect way... 

Controlling the cooling in an incorrect way might be risky and `M42 I` is looking... fishy. M106 might offer a more mainstream approach that Marlin handles in a better way?

That is as deep as I went. I can't tell precisely why that heater 2 behaves the way it does. Perhaps someone with more experience with Marlin and STM32 platform can answer this question better.

tna11, 2019-08-22:
Hello, thanks for taking a look at this.

Will look into it more, but must be a bug somewhere. I changed the #define HEATER_2_PIN to another pin, so in theory marlin isn't controlling P19(PB0) but the same thing is happening.

I'm not using the board on a 3d printer, so no worries about overheating and causing a fire

omahena, 2019-08-22:
My pleasure. I am happy to hear you are not using the board with a 3d printer and are taking proper safety precautions.

Cool idea, to feed a different pin to Marlin's thermalManager. What is interesting is that the HEATER_2_PIN doesn't seem to be mentioned much in the source files anywhere. It is initialised in Temperature::init. Other than that it is not referenced much and mostly in the context of BARICUDA.

I am sure I missed something because HEATER_0_PIN is also hardly mentioned outside the #define scope and heating obviously works on 3D printers. 😋 It would require a lot more time to figure this stuff out. And not having a debugger is also not helping.

Best of luck. I hope you figure it out.

tna11, 2019-08-23:
Think I've sorted the problem, both pins were tied to timer 3 channel 3 output, commenting out line 192 in PeripheralPins.c ( pc_8 tim3) has got it to work. PC_8 is then using timer 8 channel 3

AnHardt, 2019-08-23:
Excellent hint!
Looks as if also in the SPI sections a bit of cleanup is needed .

omahena, 2019-08-24:
@tna11 Ha! Wow. Good catch. How did you find this conflict?

@AnHardt Could this be the problem with SPI3 and the SD card? You seem to have experience with the STM32 platform and configuration. Do the SPI arrays dedicate/configure (multi-purpose) pins to a specific purpose?

So for example:
```
const PinMap PinMap_SPI_SSEL[] = {
  {PA_4,  SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  {PA_4,  SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  {PA_15, SPI1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI1)},
  {PA_15, SPI3, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_SPI3)},
  {PB_9,  SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {PB_12, SPI2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},
  {NC,    NP,    0}
};
```
Says that PA4 can be either SPI1 or SPI3 SS pin? And defining both leaves it up to "chance" what you get in the end (probably first match)?

closed 2019-09-02 by tna11

bmxorcist, 2020-01-27:
Can anyone tell me how to determine what pin number to enter for the letter/number pins on an STM32 based board (SKR Mini E3)?  Trying to activate the HEATBED(PC9), HE0(PC8), and Fan0(PA8) pins on an SKR Mini for a custom application. Entering "M42 PC9 Sxx" just echos "M42 P0 Sxx". Any other combination of numbers that I've tried doesn't work either. Looking in fastio.h led me to try "P29" for PC9... but no luck.