bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #37: Using USB CDC, TFT35 and ESP-01S at the same time

omahena, 2019-08-21:
How can we configure Marlin to allow a "triple head" configuration?

So that it is possible to use OctoPrint to control the board over the USB serial connection, the TFT35 when standing next to the printer and ESP-01S when we prefer it to OctoPrint?

Evg33, 2019-08-21:
Marlin supports only 2 interfaces.

omahena, 2019-08-21:
That seems evident from there being just two `SERIAL_PORT` and `SERIAL_PORT_2` defines in configuration.h...

@Evg33 Do you think... A third one might be easy to hack in? For STM32 platforms I couldn't find the "Serial.cpp" file implementation that specific platforms have.

@bigtreetech I wish this was mentioned on the buy pages for the LCD and WiFi module pages or even better - documentation. Maybe a compatibility matrix?

Does anyone need an ESP-01S board in the EU?

dant220, 2019-08-21:
It is something I would be interested in too.

omahena, 2019-09-18:
@dant220 I think I found a way... It includes sending more money to BTT... So... :/ But I think it is possible to get their TFT35 2.0 or 3.0 and it has a WiFi connector on the LCD board. Please verify all the details before you buy! So you connect Marlin to USB serial and Display serial and then the display will daisy chain the WiFi serial...

closed 2020-10-12 by omahena