bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #38: Remote firmware update

omahena, 2019-08-21:
How can the firmware be updated remotely over the USB connection?

That is without having to remove the SD card, take it over to the computer, write the firmware on it and take it back to the board, plug it in and restart the board?

golfromeo-fr, 2019-08-25:
afaik, it is not planned for now. Later?

Evg33, 2019-09-18:
* M997 - Perform in-application firmware update
If you can put the file on the card, then you can use 997.

Evg33, 2019-09-18:
I also think you can connect the st-link to a anoter usb of raspberry/comp.

omahena, 2019-09-24:
I am sorry I wasn't clear in my initial issue report. I am talking about a remote firmware update that is possible to do for the average user. For the SKR Pro 1.1 it would likely require two steps.
1. Uploading the file to the SKR Pro 1.1
2. Flashing the firmware into the right spot in flash memory.

Step 1 is already the problem here. Marlin cannot in any way use the SD card so uploading the file is not possible.

I agree it would require just a reset after the file is placed on the SD card. Did you test if the M997 actually resets your board? Marlin documentation doesn't mention the M997 directly so that usually means it is a little H4x0r.
http://marlinfw.org/meta/gcode/

I do have an ST-Link in the mail and when it arrives I hope I will be able to remote flash a firmware image to the board. But I think we both agree that the manufacturer didn't envision the end users to flash new firmware using ST-Link.

So to be clear, I am sure there is a way to hack the image in remotely. We could for example try to connect the SD card SPI to a RasPi that pretends it's an SD card. That might work. And a reset could be done with a relay controlling the power delivery to the board that is also controlled from the RasPi.

It's just... The SKR Pro 1.1 is the highest priced board from BTT and the board is lacking basic functionality that was promised at the time of sale. A Creality Melzi board will let you use an SD card and offer prints protected from power failure as well as remote flashable firmware if you install a bootloader. One pays money for a high end upgrade expecting more... Not less.

Evg33, 2019-09-25:
M997 - https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/firmware/Marlin-2.0.x-SKR-Pro/Marlin/src/gcode/gcode.h#L270
But this does not work with this board. I was wrong.

omahena, 2019-09-26:
I happened to see some changes in Marlin recently that were related to M997 and BTT. Got all excited and then realised that they were for the SKR v1.3. I guess the message is that real professionals don't need a board reset.

We could try for a "in soviet Russia" type joke: "With the SKR Pro board the operator doesn't reset the board, the board resets the operator." Meh.

I suppose the M997 wouldn't be too hard to implement. I am sure the functions are there in HAL.

darknode, 2019-09-28:
M997 has been [added](https://github.com/MarlinFirmware/Marlin/pull/15394), but still we need  USBComposit support to have a full solution :(
I've found [implementation](https://github.com/grafalex82/GPSLogger/tree/master/Libs/USB) for stm32f4 with stm32+cmsis, but I'm not sure how to apply it for Marlin.

sl1pkn07, 2019-09-28:
Hi

other solution (witch i use in my rearm) is install a Toshiba Flashair (SD+WiFi module), put the firmware in it via WebDAV and then reset the machine for update the firmware

robustini, 2019-12-22:
No solution?

closed 2020-10-12 by omahena