bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #39: TMC5160 not moving

Haschtl, 2019-08-24:
I have an issue with my TMC5160.

The motors are not moving, if i command a movement or homing. They feed some current to the motors if i enable them with M17 (doesnt feel very strong), but they will not move. 

I have configured Rsense to 0.075, microstepping to 256 and the current to 1700mA in configuration_adv.h

SENSORLESS_HOMING is not active, but SQUARE_WAVE_STEPPING and TMC_USE_SW_SPI is.

The is the TMC-debug output:
```
Send: M122
Recv: 		X	Y
Recv: Enabled		false	false
Recv: Set current	1000	1000
Recv: RMS current	990	990
Recv: MAX current	1396	1396
Recv: Run current	19/31	19/31
Recv: Hold current	9/31	9/31
Recv: Global scaler	133/256	133/256
Recv: CS actual	14/31	14/31
Recv: PWM scale	13	13
Recv: stealthChop	true	true
Recv: msteps		0	0
Recv: tstep		max	max
Recv: pwm
Recv: threshold	0	0
Recv: [mm/s]		-	-
Recv: OT prewarn	false	false
Recv: OT prewarn has
Recv: been triggered	false	false
Recv: off time	15	15
Recv: blank time	54	54
Recv: hysteresis
Recv: -end		12	12
Recv: -start		8	8
Recv: Stallguard thrs	0	0
Recv: DRVSTATUS	X	Y
Recv: stallguard
Recv: sg_result	0	0
Recv: fsactive
Recv: stst		*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 		X	0x80:0E:40:00
Recv: 		Y	0x80:0E:40:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: ok
[...]
Send: M17
Recv: ok
[...]
Send: M122
Recv: 		X	Y
Recv: Enabled		true	true
Recv: Set current	1000	1000
Recv: RMS current	990	990
Recv: MAX current	1396	1396
Recv: Run current	19/31	19/31
Recv: Hold current	9/31	9/31
Recv: Global scaler	133/256	133/256
Recv: CS actual	13/31	13/31
Recv: PWM scale	15532281	15532281
Recv: stealthChop	true	true
Recv: msteps		0	0
Recv: tstep		max	max
Recv: pwm
Recv: threshold	0	0
Recv: [mm/s]		-	-
Recv: OT prewarn	false	false
Recv: OT prewarn has
Recv: been triggered	false	false
Recv: off time	15	15
Recv: blank time	54	54
Recv: hysteresis
Recv: -end		12	12
Recv: -start		8	8
Recv: Stallguard thrs	0	0
Recv: DRVSTATUS	X	Y
Recv: stallguard
Recv: sg_result	0	0
Recv: fsactive
Recv: stst		*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 		X	0x80:0D:40:00
Recv: 		Y	0x80:0D:40:00
Recv: 
```

omahena, 2019-08-24:
@Haschtl I use TMC5160s. I want to say without issues, but it was hard to find the right current settings.

I did not use SQUARE_WAVE_STEPPING so no idea if that will influence it. 1700mA sounds like a lot of current (unless you have about 3.5A stepper motors). 

SENSORLESS_HOMING doesn't work for me at all. Just moves the printer head a little in X and Y axis and then stops. If I repeat homing it will reach home position at which point the skipped steps grinding starts.

Which stepper driver modules are you using? BTT V1.1?

Tell us more about your stepper motors (current rating, etc).

Haschtl, 2019-08-24:
My stepper motors are rated at 2.4A. Stepper type: 17HM5424 
They are fine, I used them before with DRV8266.
My SKR Board is running at 24V. Therefore i have `#define CHOPPER_TIMING CHOPPER_DEFAULT_24V`.

I did not desolder the stallguard pins, i have only disabled sensorless_homing and am using normal endstops, which are working with M119

My stepper driver modules are from Watterott. I am only using them on X and Y axis till now. The other drivers are DRV8266.
Did you try to increase the X_STALL_SENSITIVITY ? This could fix your sensorless homing issues.

omahena, 2019-08-25:
If you used the same steppers with DRV8266, this means that the connectors, wires and motors are likely working and connected properly.

You M122 output actually shows 1000mA configuration on both steppers and at this point more importantly that SPI communication is enabled and working.

I am currently using:
```
  #define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

  #if AXIS_IS_TMC(X)
    #define X_CURRENT      400  // 760mA still very hot // (mA) RMS current. Multiply by 1.414 for peak current.
    #define X_MICROSTEPS   256  // 0..256
    #define X_RSENSE     0.075
  #endif

#if AXIS_IS_TMC(Y)
    #define Y_CURRENT      400
    #define Y_MICROSTEPS   256
    #define Y_RSENSE     0.075
  #endif

  #if AXIS_IS_TMC(Z)
    #define Z_CURRENT      400 // 800
    #define Z_MICROSTEPS   256 // 16
    #define Z_RSENSE     0.075
  #endif

  #if AXIS_IS_TMC(E0)
    #define E0_CURRENT     500 // With 500 slightly warm to touch //800 reasonable  // 900
    #define E0_MICROSTEPS  256
    #define E0_RSENSE    0.075
  #endif

  #define TMC_USE_SW_SPI

  #define STEALTHCHOP_XY
  #define STEALTHCHOP_Z
  #define STEALTHCHOP_E

  #define CHOPPER_TIMING CHOPPER_DEFAULT_12V
  #define MONITOR_DRIVER_STATUS

  #define HYBRID_THRESHOLD

  //#define SENSORLESS_HOMING // StallGuard capable drivers only

  //#define SQUARE_WAVE_STEPPING
  #define TMC_DEBUG
```

I find myself wondering if perhaps you need to have a Z axis enabled since... Marlin usually drives a 3D printer... Since it is easier to disconnect and test rather than vice versa, I tested X, Y mode for you.

It is definitely outside of normal operation. For me M122 displays:

```
Recv: Driver registers:
Recv: 		X	0x00:00:00:00	 Bad response!
Recv: 		Y	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... Error: All LOW
```
Despite that, motors work... A little. Kind of... Well - erratic. It might make an X move immediately and then wait for 15s before allowing an Y move.

I have just empirical evidence to support this... But would you humour me by connecting a Z and E motor as well. I tried X, Y, Z motors only and things also didn't work much better until I connected the E motor as well.

In my case, I had to have all four of them before things got normal again:
#define X_DRIVER_TYPE  TMC5160
#define Y_DRIVER_TYPE  TMC5160
#define Z_DRIVER_TYPE  TMC5160
#define E0_DRIVER_TYPE TMC5160

I find it interesting you got a successful SPI connection reported by M122 at all.

Haschtl, 2019-08-25:
I have now connected all other TMC5160, now no axis is working (before the Z-Axis with the DRV8266 worked fine)

I also decreased the current to 500mA and have the same configuration as you now... only CHOPPER_TIMING  is CHOPPER_DEFAULT_24V in my case.
I think I will need to test the drivers with some simple examples first.

Here is the M122 output

```
Send: M122
Recv: 		X	Y	Z	E
Recv: Enabled		false	false	false	false
Recv: Set current	500	500	500	500
Recv: RMS current	490	490	490	490
Recv: MAX current	691	691	691	691
Recv: Run current	9/31	9/31	9/31	9/31
Recv: Hold current	4/31	4/31	4/31	4/31
Recv: Global scaler	133/256	133/256	133/256	133/256
Recv: CS actual	31/31	4/31	4/31	4/31
Recv: PWM scale	4294967295	4	4	4
Recv: stealthChop	true	true	true	true
Recv: msteps		0	256	256	256
Recv: tstep		4294967295	max	max	max
Recv: pwm
Recv: threshold	0	396	2111	241
Recv: [mm/s]		-	100	3	30
Recv: OT prewarn	true	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time	15	4	4	4
Recv: blank time	54	24	24	24
Recv: hysteresis
Recv: -end		12	2	2	2
Recv: -start		8	1	1	1
Recv: Stallguard thrs	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: stallguard		*
Recv: sg_result	1023	0	0	0
Recv: fsactive		*
Recv: stst		*	*	*	*
Recv: olb		*
Recv: ola		*
Recv: s2gb		*
Recv: s2ga		*
Recv: otpw		*
Recv: ot		*
Recv: Driver registers:
Recv: 		X	0xFF:FF:FF:FF	 Bad response!
Recv: 		Y	0x80:04:40:00
Recv: 		Z	0x80:04:40:00
Recv: 		E	0x80:04:40:00
Recv: 
Recv: 
Recv: Testing X connection... Error: All HIGH
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
Recv: ok
```

omahena, 2019-08-25:
Thanks for trying what I suggested. I mean... it all looks right. Just no motion you say... Short of offering you my firmware build (or vice versa) as a  ultimate firmware test... I am not sure what to suggest.

We do use different modules though and although I think BTT made an exact copy of Watertott modules (and there was even an angry comment somewhere about atribution)... There could be a difference somewhere.

There... was some talk about certain stepper modules and having to add or remove a certain pin or add a connection... Which version of the Wattertott modules do you have?

https://learn.watterott.com/silentstepstick/pinconfig/tmc5160/

Haschtl, 2019-08-26:
I found the solution for my problem!

I have PSU_CONTROL active. I already figured out, that the SPI communication is dead, if the 24V supply is off. If I enable the PSU, the SPI communication seems correct (M122) but I think the drivers are not initialized correctly, if the power supply was off. 
Here is an related issue: https://github.com/MarlinFirmware/Marlin/issues/15055
Btw: I'm using MarlinFirmware/Marlin and not this fork.

I also have a hardware button, which cuts the power for the motors (a safety switch). If i toggle it, the motors will not move.

Now that the motors are moving, I have another problem.
@omahena: You have configured your steppers to run with 256 microstepping. 
I encountered some issues, when I had this configuration. The stm32 crashed and rebooted after a fast and long Z-movement. I think this is related to the high step/second rate in this movement. The Z-axis has the biggest step/mm configuration (2000 at 16 microstepping and 32000 at 256 microstepping). If the velocity is high, e.g. 30 mm/s, the steps/second will be 960000, so 960 kHz. 
With this list  https://reprap.org/wiki/Step_rates, i think the maximum frequency will be around 96-120kHz. But this is still not enough. Could this cause the firmware crash?

This problem could be solved, if the marlin firmware was capable of using the TMC5160's "Motion Controller" via SPI. This would reduce the workload of the controller board a lot.

omahena, 2019-08-26:
I am glad you figured it out. PSU_CONTROL would mess with your drivers, yes. The issue with driver initialization seems a likely problem in retrospect. I am also now completely on Marlin and only use this fork to double check stuff if something isn't working correctly. Essentially, just to check if BTT had something working but then the future Marlin development broke it.

@Haschtl Thanks for the heads up on the 256 microstepping. So far I didn't have any issues wit it. I did read about interpolation issues on the older version of TMC drivers due to motion extrapolation. 

You could be right on the high step rate but I do not know enough to comment on the subject with certainty. I think there are differences between modes of interfacing with the stepper driver. I think that if you use the old STEP/DIR interface the issue you describe is likely to occur since you need to provide one step pulse for any type of move. So I guess that if you want to do a full step with say 1/16 microstepping you need to provide 16 pulses. I thought that in SPI/UART mode things are different. You can probably send a packet to "do 16 microsteps" and the stepper driver IC does that for you. Things would get less convenient for SPI/UART driving when you would want to do non-linear precise motion using full precision. So I'd be surprised if motion controller of the TMC5160 is not already in use for SPI control.

Since I didn't have any issues until now... I didn't put too much thought into microstepping other than more sounds better. Already finding the optimal driving current took waaay too much of my time. :/ But maybe the issues are actually related.

closed 2019-09-18 by Haschtl