bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #40: 3d Model Inconsistencies

blindMoe, 2019-08-24:
It looks like the two 3D models you provide are different from one another.  If I align the two on top of each other items are not the same size and are slightly offset.  Any chance you can provide a .obj or .f3d file instead of just an STL? I want to design a case and would prefer to use models that are accurate but converting the SKR-Complete stl to BREP in fusion has too many faces and the simplified version is not true to scale

hipejoe, 2019-09-23:
In the user manual: "SKR PRO V1.1 user manual v1.1.pdf" there are dimensions that should help. see page 3.

rubienr, 2023-04-18:
Is it possible to export the current v1.2 and place `SKR-Pro-1.2-Complete-board.stl` in [BIGTREETECH-SKR-PRO-V1.1/SKR-PRO-V1.2/models](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/SKR-PRO-V1.2/)  (same as [SKR-Pro-1.1-Complete-board.stl](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/models/SKR-Pro-1.1-Complete-board.stl))?

rubienr, 2023-04-21:
> Is it possible to export the current v1.2 and [...]

:man_facepalming: The model can be found in [models](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/SKR-PRO-V1.2/manual) in [SKR-PRO-V1.2-step.rar](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/SKR-PRO-V1.2/manual/SKR-PRO-V1.2-step.rar)