bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #41: SKR PRO PWM Pins?

Rics3D, 2019-08-25:
Hi, 

can someone tell me the PWM Pins for the Partcollingfan?
FAN0 and FAN1 is used for hotendcooling.
So i choose FAN2 for the Partcooling, but they just start working @80% and this is full blast...
Ist there a list with all PWM Pins?

Thank you

Haschtl, 2019-08-26:
You can have a Look AT the datasheet of the stm32
https://www.mouser.de/datasheet/2/389/stm32f405rg-956214.pdf

Rics3D, 2019-08-26:
Okay, but where are the pins located on the board?

Haschtl, 2019-08-27:
Have a Look here: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/manual/SKR-PRO-V1.1-Pin.pdf

dineshvyas, 2019-09-19:
pine_STM32F4.h
look at these lines.
#ifndef FAN_PIN
  #define FAN_PIN          PC6
#endif
#define FAN1_PIN           PC7
#define FAN2_PIN           PC8

defalkner, 2019-09-24:
> pine_STM32F4.h
> look at these lines.
> #ifndef FAN_PIN
> #define FAN_PIN PC6
> #endif
> #define FAN1_PIN PC7
> #define FAN2_PIN PC8

This is a huge help. I know the hardware side but the software, programming and setting locations for the pins in the software I have no idea where the locations are.  The pine_STM32F4.h was the information I needed otherwise Id be hunting around the pins.h all day. Thanks!

Formation3D, 2019-09-24:
What I ended up doing was making the part cooling fan0 the part cooling fan, fan1 for the hotend fan and fan2 for 3 fans cooling the stepper drivers...type in fan in the search bar in configuration adv.h.... scroll to controller fan uncomment define use controller fan and uncomment define controller fan pin and put PE6 in its place, this is where you control the fan to cool stepper drivers.  Scroll down to define E0 auto fan pin and put PE5. You can set the temp you want for the extruder fan to come on at. Leave it at 255. Scroll down to define fanmux0 pin and type in PC8, this is your part cooling fan and can be controlled by you. If you go in Marlin to src, pins, stm32, pins bigtreetech skr pro v1.1 it will give you the pin numbers you need.

jkdesignz, 2019-09-27:
I think what he is saying is that the fan only comes on when it is set to 80%. 
what you need to do is go into Configuration_Adv.h  and set the PWM 
here  line 304 -305 or around that line,
#define FAN_MIN_PWM 0
#define FAN_MAX_PWM 255 

also set the kick start, it is just above line 304
uncomment =  #define FAN_KICKSTART_TIME 1000 
set to 1000

that should fix the problem. i know it fixed mine. now i can use it all the way down to 5%. but i have ball bearing fans. hope that helps

martins6413, 2019-10-06:
Hallo,
die CNC Lüfter Pins (SKR Pro V1.1) sind Frequenz gesteuert und sind in niedriger Drehzahl sehr laut und bleiben manchmal auch stehen.Das Slicerprogramm kann diesen Lüfter nicht ordentlich regeln und somit ist ein Druck mit PETG oder ABS fast nicht möglich.
Gibt es hierfür einen Lösungsvorschlag ?

martins6413, 2019-10-06:
Hi,
The PWM pins in the SKR Pro V1.1 are frequency controlled. The parts fan (24V) sometimes stops at low RPM and is very noisy. The slicer program also does not get the fan properly controlled and thus PETG or ABS printing is very difficult as the fan rotates either too fast or not at all.
Is there a solution for this ?

AnHardt, 2019-10-06:
```
// When first starting the main fan, run it at full speed for the
// given number of milliseconds.  This gets the fan spinning reliably
// before setting a PWM value. (Does not work with software PWM for fan on Sanguinololu)
//#define FAN_KICKSTART_TIME 100

/**
 * PWM Fan Scaling
 *
 * Define the min/max speeds for PWM fans (as set with M106).
 *
 * With these options the M106 0-255 value range is scaled to a subset
 * to ensure that the fan has enough power to spin, or to run lower
 * current fans with higher current. (e.g., 5V/12V fans with 12V/24V)
 * Value 0 always turns off the fan.
 *
 * Define one or both of these to override the default 0-255 range.
 */
//#define FAN_MIN_PWM 50
//#define FAN_MAX_PWM 128
```

martins6413, 2019-10-06:
Thanks for the tip, I've read over. Exchange the fan probably broken.

Lyr3x, 2019-12-19:
I am trying to find free PWM pins for RGBW Leds. All pins i tested (Checked the timer pins in the datasheet). However only `PC9`is working correctly. All other Pins are just fully on or off.

Any ideas here?

AnHardt, 2020-01-04:
If not using SPI stepper drivers on Y and Z the Y_CS (PB8 on TIM4_CH3 or TIM10_CH1) and Z_CS (PB9 on TIM4_CH4 or TIM11_CH1) could be usable and are relative accessible. That will need some additions in the PWM/Timer section of 'Marlin\buildroot\share\PlatformIO\variants\BIGTREE_SKR_PRO_1v1\PeripheralPins.c'

```diff
const PinMap PinMap_PWM[] = {
  {PB_1,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 4, 0)}, // TIM3_CH4    HEATER0
+ //(PB_8, TIM4,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 3, 0)}, // TIM4_CH3 already used for HEATER_1 id hardware PWMd
+ (PB_8, TIM10,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM10, 1, 0)},
+ //(PB_9, TIM4,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 4, 0)}, // TIM4_CH4 chanel 4 is free but the timer is already in use for heaters. So the PWM-freqency is already fixed if one heater is on hardware PWM
+ (PB_9, TIM11,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM11, 1, 0)},
  {PD_14, TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 3, 0)}, // TIM4_CH3    HEATER1
  {PB_0,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 3, 0)}, // TIM3_CH3    HEATER2
  {PD_12, TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 1, 0)}, // TIM4_CH1    BED
  {PC_8,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 3, 0)}, // TIM8_CH3    FAN0
  {PE_5,  TIM9,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM9, 1, 0)}, // TIM9_CH1    FAN1
  {PE_6,  TIM9,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM9, 2, 0)}, // TIM9_CH2    FAN2
  {PC_9,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 4, 0)}, // TIM8_CH4    EXTENSION1-4
```

AnHardt, 2020-01-04:
If you can swap the probe away from PA1 and PA2 (for exaple to any of the EXTENSION2 pins) you can use them following the same scheme.

```
PA1/TIM5_CH2/TIM2_CH2
PA2/TIM9_CH1/TIM2_CH3
```
Just check if the timers are used somewhere else.

AnHardt, 2020-01-04:
If someone of the Marlin developers is around:

Usually we use only one servo - for the probe.
How about a single servo library using hardware PWM on pin PA1 with timer 5 channel 2 for the BTT SKR PRO V1.1?
That would have exceptional low jitter and would not use more timers than the normal library.

AlexanderLL95, 2020-01-16:
> If not using SPI stepper drivers on Y and Z the Y_CS (PB8 on TIM4_CH3 or TIM10_CH1) and Z_CS (PB9 on TIM4_CH4 or TIM11_CH1) could be usable and are relative accessible. That will need some additions in the PWM/Timer section of 'Marlin\buildroot\share\PlatformIO\variants\BIGTREE_SKR_PRO_1v1\PeripheralPins.c'
> 
> ```diff
> const PinMap PinMap_PWM[] = {
>   {PB_1,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 4, 0)}, // TIM3_CH4    HEATER0
> + //(PB_8, TIM4,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 3, 0)}, // TIM4_CH3 already used for HEATER_1 id hardware PWMd
> + (PB_8, TIM10,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM10, 1, 0)}, // Y_CS
> + //(PB_9, TIM4,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 4, 0)}, // TIM4_CH4 chanel 4 is free but the timer is already in use for heaters. So the PWM-freqency is already fixed if one heater is on hardware PWM
> + (PB_9, TIM11,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM11, 1, 0)}, // Z_CS
>   {PD_14, TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 3, 0)}, // TIM4_CH3    HEATER1
>   {PB_0,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 3, 0)}, // TIM3_CH3    HEATER2
>   {PD_12, TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 1, 0)}, // TIM4_CH1    BED
>   {PC_8,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 3, 0)}, // TIM8_CH3    FAN0
>   {PE_5,  TIM9,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM9, 1, 0)}, // TIM9_CH1    FAN1
>   {PE_6,  TIM9,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM9, 2, 0)}, // TIM9_CH2    FAN2
>   {PC_9,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 4, 0)}, // TIM8_CH4    EXTENSION1-4
> ```
Thanks for the tip. I was searching a couple of months for this solution to implement a RGB led strip!

Now my skr pro board has a bltouch and RGB led strip onboard with tmc2209. Pins PC9, PB8 and PB9 are user for the RGB led strip.

dineshvyas, 2020-02-22:
// These pins have been defined for something else on the board but they MIGHT be
  // used by the user as PWM pins if they aren't used for their primary purpose.
  {PC_6,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 1, 0)}, // TIM3_CH1    ESP8266 connector. Available if 8266 isn't used
  {PC_7,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 2, 0)}, // TIM3_CH2    ESP8266 connector. Available if 8266 isn't used
how about these pins?
I am not using ESP8266.

DroneMang, 2020-05-09:
I have an SKR PRO, and I am having the issue that the part cooling fan which I have set as below from pins (stock):

#define FAN_PIN                             PC8   // Fan0 

will jump from a low speed to basically full blast when changing from S246 to S247.  As in there is not a gradual increase in fan speed.  Below S246 there is some adjustment to the speed that I can hear.  There may be some flight increase in speed from S247 to S255.

I am running TMC5160's with sensorless homing, I can reply with other enabled features if needed.

Thanks!
DroneMang

defalkner, 2020-05-09:
I believe PWM needs to be enabled

DroneMang, 2020-05-09:
Do you mean:

FAN_SOFT_PWM

defalkner, 2020-05-09:
Yes, I have the SKR pro 1.1  and this was a bit of a nightmare for me to
figure out but it can be corrected. This is from my marlin code

*configuration.h*
#define FAN_SOFT_PWM
#define SOFT_PWM_SCALE 0

configuration_adv.h
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN PE6           // Set a custom pin for the
controller fan
  #define CONTROLLERFAN_SECS 30             // Duration in seconds for the
fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255           // 255 == full speed
  //#define CONTROLLERFAN_SPEED_Z_ONLY 127  // Reduce noise on machines
that keep Z enabled

#define FAN_MIN_PWM 0
#define FAN_MAX_PWM 255

i dont know if any of this will help, just adjust as needed

DroneMang, 2020-05-09:
Perfect!  I just set 

#define CONTROLLER_FAN_PIN FAN2_PIN

but FAN2_PIN is PE6 in pins so same thing.

So do you still use M106 S255 down to 0 as usual?

aliman1, 2022-02-25:
> > If not using SPI stepper drivers on Y and Z the Y_CS (PB8 on TIM4_CH3 or TIM10_CH1) and Z_CS (PB9 on TIM4_CH4 or TIM11_CH1) could be usable and are relative accessible. That will need some additions in the PWM/Timer section of 'Marlin\buildroot\share\PlatformIO\variants\BIGTREE_SKR_PRO_1v1\PeripheralPins.c'
> > ```diff
> > const PinMap PinMap_PWM[] = {
> >   {PB_1,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 4, 0)}, // TIM3_CH4    HEATER0
> > + //(PB_8, TIM4,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 3, 0)}, // TIM4_CH3 already used for HEATER_1 id hardware PWMd
> > + (PB_8, TIM10,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM10, 1, 0)}, // Y_CS
> > + //(PB_9, TIM4,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 4, 0)}, // TIM4_CH4 chanel 4 is free but the timer is already in use for heaters. So the PWM-freqency is already fixed if one heater is on hardware PWM
> > + (PB_9, TIM11,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM11, 1, 0)}, // Z_CS
> >   {PD_14, TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 3, 0)}, // TIM4_CH3    HEATER1
> >   {PB_0,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 3, 0)}, // TIM3_CH3    HEATER2
> >   {PD_12, TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 1, 0)}, // TIM4_CH1    BED
> >   {PC_8,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 3, 0)}, // TIM8_CH3    FAN0
> >   {PE_5,  TIM9,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM9, 1, 0)}, // TIM9_CH1    FAN1
> >   {PE_6,  TIM9,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM9, 2, 0)}, // TIM9_CH2    FAN2
> >   {PC_9,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 4, 0)}, // TIM8_CH4    EXTENSION1-4
> > ```
> 
> Thanks for the tip. I was searching a couple of months for this solution to implement a RGB led strip!
> 
> Now my skr pro board has a bltouch and RGB led strip onboard with tmc2209. Pins PC9, PB8 and PB9 are used for the RGB led strip.

how exactly do you modify the PeripheralPins.c file .