bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #44: Extension-1 ADC inputs not functioning

Paul47, 2019-09-01:
I'm using the SKR PRO v1.1/Marlin 2 bugfix, for a 4 extruder printer. For the 4th temp sensor I have tested the ADC pins on Ext-1: Pins PF7, PF8, PF10, PF9 (all identified as ADC capable) using a tested thermistor circuit design from RAMPS as input. Defining TEMP_3_PIN to any of these pins I get only zero temp reported.  (I've also tried re-defining TEMP_1,2, and 3). 

The other 3 extruders and steppers are operating correctly. Other settings: extruders=4, Sensor types=5, single hotend disabled. I have used this same Marlin setup on my RE-ARM controller and can map 4 termistors successfully, but am running out of pins and power for motors and heaters.

omahena, 2019-09-11:
It is a bit of a long shot. But I have custom temp sensors running on PF5 and PF6. They are intended for hotend temp sensors so... I was only mildly surprised to see them working.

But if we look at the `buildroot\share\PlatformIO\variants\BIGTREE_GENERIC_STM32F407_5X\PeripheralPins.c` the PF_5  pin in only defined as part of the PinMap_ADC. On the other hand PF_6 is also defined as a member of PinMap_PWM so perhaps I was lucky to see values from PF_6.

But since PF_7 and PF_8 are defined as both members of PinMap_ADC as well as PinMap_PWM PinMaps... So... Maybe... Just maybe if you comment out lines 211 and 212, your temp sensors will start to work:

```
{PF_7,  TIM11,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM11, 1, 0)}, // TIM11_CH1
{PF_8,  TIM13,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF9_TIM13, 1, 0)}, // TIM13_CH1
```

If you look at issue [36](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/36#issuecomment-524234963) there have been some similar type of problems with pins solved the same way.

@Paul47 I am really curious how your 4 extruder printer will work with this board. I would expect a lot of problems. Keep an eye on your HEATER1 and HEATER2 channels as they may turn on as a side effect of some other pin change. Have a look at this [video](https://www.youtube.com/watch?v=pVOuqxbJl3w) of how my board behaves with a single extruder. Definitely do not leave the board unattended and have a fire extinguisher handy.

Paul47, 2019-09-11:
Thanks @omahena I will test each of these recommendations. My SKR PRO is still breadboarded until I get all the electronics working. I use a stepper expander for the additional motor. I use external MOSFET boards on ALL my heaters to reduce current draw through the Controller. I found this also greatly speeds the heat up cycles. These are so cheap now days I recommend it to everyone. I have a smoke+fire detector in the room and a fire extinguisher at the door.

The 4 steppers and extruders work, but I've assigned temp sensor 4 to 3 for now. This could lead to problems as you mentioned. For Marlin 2.0 I modified the various override settings for the heater/sensor/stepper #defines  so I can change them in all configuration.h.  This way I can re-configure from 1 to 4 extruders and their pin assignments quickly for testing.

evandene, 2019-10-07:
Hi,
Question from a total dummy. I would like to use my PT100 amplifier board(s) with PT100 sensor(s) were the out put go's to 5V
Are the pins from extension 1 & 2 5V tolerant? Can I hooked-up the PT100 f.e. at PC4 and an other one at PC5?
I hooked-up PS_ON at PC9 and that worked perfect

AnHardt, 2019-10-07:
There are **NO** 5V tolerating ADC inputs on the ST processors.

evandene, 2019-10-07:
Thanks for the info

evandene, 2019-10-07:
I will use PF8 and PF10 as signal pin
I will use GND and 3.3V form the free limit switch connectors
Now i still need to find out the PID settings

Just for your info:
According the PIN schematics from Bigtree-tech. The I2C rail of the SKR PRO v1.1 should be powered by 3.3V but be careful, it is 4.95V. the same story for the UART rail.

Paul47, 2019-10-08:
After trying the suggestions from omahena on these ADC pins, I have still not been able to get them to function. evandene, if you have success, please report back with your configuration. I would like to find out if it's my board or in my Marlin setup.

closed 2019-10-08 by Paul47

evandene, 2019-10-08:
> After trying the suggestions from omahena on these ADC pins, I have still not been able to get them to function. evandene, if you have success, please report back with your configuration. I would like to find out if it's my board or in my Marlin setup.

It works,
T0 is responding on my E3D amplifier with PT100 sensor
I use PF8 from EXTENSION1 as signal pin
I use GND and 3.3V from a free limit switch connector
Do not forget to change pin settings in 
[image]

[image]

evandene, 2019-10-08:
For more trustful measuring, I will buy me an Adafruit PT100 amplifier with 3.3V output signal
https://www.adafruit.com/product/3328
[image]

ktotomskru, 2019-10-25:
> For more trustful measuring, I will buy me an Adafruit PT100 amplifier with 3.3V output signal
> https://www.adafruit.com/product/3328
> [image]

This one is supposed to be connected via SPI bus. I use it with SKR pro without any issues.

evandene, 2019-10-25:
Great, thanks.
I will implement this in my new design