bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #46: Pins still wrong? BIGTREE_GENERIC_STM32F407_5X /variant.h

carl1961, 2019-09-02:
#define PB4   45 //0:NJTRST  1:SPI3_MISO / TIM3_CH1 / SPI1_MISO
 #define PB5   **45** //1:TIM3_CH2 / SPI1_MOSI / SPI3_MOSI
[image]

omahena, 2019-09-11:
@carl1961 Awesome find. However the variant.h was updated yesterday... And if you compare the files it is pretty much a completely different pinout definition. I resorted the first part to match pin names and see differences in values.

I spent a lot of time trying to get the on-board SD card working. And PB5 is the TF_MOSI in the SPI3 pin set.

[image]

I can explain the error you found: coding error. It happens. To all of us. But the differences between original and current pin numbers are just deeply disturbing. For a moment I thought that the file is perhaps not used in the build... But it sure is. I now deeply regret having invested any time into this firmware.

The `pins_BIGTREE_SKR_PRO_V1.1.h` is still a wasteland devoid of all promised features like SD card support, printing protected from power failure, WiFi module, filament runout sensors...

The `buildroot\share\PlatformIO\variants\BIGTREE_GENERIC_STM32F407_5X\PeripheralPins.c` is still a complete mess of overlapping pin definitions that will lead to problems like the MOSFET for HEATER2 to turn on when you are controlling the FAN0. Imagine you actually had something connected to HEATER2. Firmware has no clue the heater is turned on, because it was never turned on by the thermal management code of Marlin. 

Same thing goes for HEATER1 actually. [Here](https://www.youtube.com/watch?v=pVOuqxbJl3w&lc) you can see it blinking in unison with HEATER0. 

([image])

I only have one hotend configured in Marlin... Yet power is directed to HEATER1.

Considering that any and all reports here are simply ignored. No response is ever offered. No acknowledgement of error or help with user error from BTT side.

It looks like the community will simply have to resort to the only language they understand - honest user reviews.

LinoBarreca, 2019-11-14:
I fixed them. go check now on official marlin