bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #47: Limit switch and TMC2209 driver

theWituch, 2019-09-09:
Hi,
I have SKR PRO board with TMC2209 v1.1 stepper drivers and when i connect driver into board, suitable limit switch seems stop working. Without driver in slot, led shows switch state (led is on when switch is opened, and off when closed). With driver in slot, nothing happened.

I checked schematics of board and TMC driver and i think TMCs GND pin is connected to limit switch ouput signal. This signal is connected to uC too, so pin is always pulled down and will never work.

Im right? Is any solution for this issue?

Meddleglitch, 2019-09-20:
Aha! I've been having this issue too, and I hadn't considered that my 2209s were the problem, but now that you mention it, I did happen to notice that the limit switch for my E2 driver (which doesn't exist / isn't installed) works fine, while the other five won't work even if I use an external pullup (oh jeez, I sure hope I didn't break anything! >_<)

..........wait.......... hang on a minute... Are WE the problem, here!?
[image]
Is THIS why they tell us to castrate our poor 2209 drivers? o_o 
[image]

omahena, 2019-09-24:
Hmm. I have the TMC5160 drivers. I had to add the  stallGuard pin to actually be able to use the sensorless homing feature. However I only get into trouble when I use a stallGuard feature AND try to use the endstop switches as well. The endstops worked fine if I disabled sensorless homing even with the diag pin of the steppers connected to the board.

But... TMC2209 might be a little different and I suppose they wouldn't dedicate a part of the manual to tell you to cut that pin off.

Btw... You don't have to cut the pin. You can also just heat it up to melt the solder and push it through the board so it doesn't make contact with the socket below it. When you need it again... push it through again to the bottom side. Not sure how many cycles of this modification your connector pin can take though.

Personally I find the stallGuard based sensorless homing overrated. Or maybe it just doesn't work well on my system. With a lot of tuning of the sensitivity parameter I still get grinding. Especially for example when homing an already homed hotend. I only keep it enabled because it was so hard to enable it and it refused to work for weeks.

Meddleglitch, 2019-10-06:
Yeah, I don't think they can work together unless you can somehow assign two different pins to the same endstop, because it seems like the TMC drivers connect that pin to ground, and whichever way I set up my endstops before I realized what was going on (or rather, before I remembered what I had read about weeks in advance and then forgotten to actually *do* when the time came) I couldn't get the mechanical ones to trigger/un-trigger with that extra ground wired in, even with a jumper cap directly from 3.3V/GND to the sense pin. I saw the value rising and falling when I checked it with a voltmeter, but only by about 0.3V in either direction.

I suppose if you *really* wanted to use both at the same time (assuming that it's not currently possible to condigure a whole extra set of endstop pins and wire the mechanical ones to them instead), you could have the standard normally-closed endstop (+pullup) signal pass through a PNP mosfet before draining to ground (either side of the switch ought to be fine, I think) with the normally-grounded TMC StallGuard pin (clipped/pushed through/not connected to the board) controlling the gate, so that when either it goes high or the switch gets pushed, the circuit breaks just like it would with only one or the other.

(I doubt the few nanoseconds of rise (no wait: *fall*) time would significantly affect the StallGuard's accuracy, but then that's the whole reason for wanting the endstops to keep doing their jobs in the first place.) 

I ca't say for certain whether that would actually work (the driver would certainly need to output enough voltage to deactivate the PNP mosfet, at least.), but to my somewhat-trained eye, it seems like that ought to isolate the two switches while keeping them on the same circuit, although I'm not entirely sure whether or not the StallGuard would mess with the repeatability of the mechanical endstops during a normal home operation... however, I do strongly suspect that the spring-action switches would trigger "long" before the motors would ever actually stall out. Like, entire *microseconds* beforehand... Ages, really.

I might just try this myself sometime, if the day ever comes where I actually have nothing better to do & don't have a dozen other neglected projects cluttering up my desk. <_<