bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #48: No USB serial

UnderXirox, 2019-09-11:
Hello
sorry for my english ^^
(That's when I draw the card on my card
There is Probe! NOT BlTouch and no com usb port)
 

sorry for my english ^^
(That's when I draw the card on my card
There is Probe! NOT BlTouch and no com usb port)

[image]
 
here is my concern I can not have a com usb port with the SKR-PRO (in fete ca depuit the beginning of the acha of the card)
with the maj Marlin nothing nothing changes

            #define SERIAL_PORT -1 // USB
/ **
         *Select a secondary serial port on the board to use for communication with the host.
                 * This allows the connection of wireless adapters (for instance) to non-default port pins.
                 * Serial port -1 is the USB emulated serial port, if available.
            *
        *: [- 1, 0, 1, 2, 3, 4, 5, 6, 7]
        * /

            #define SERIAL_PORT_2 6 // Wifi

for the esp any function with Proterface and the esp with the addres 192.168.0.1:8888 to paste in pronterface instead of the COM port

here are the files that I use

Thank you

dsl je n'arrive pas a upload my Configuration.zip dsl

omahena, 2019-09-11:
I am not sure I completely understand your post. But here is what I know...

The 
```
#define SERIAL_PORT -1
```
Will only give you a serial port to control the printer. SD card is not working with Marlin... at all. You can only use if for flashing the board by removing the SD card, plugging it into an SD card reader in your computer, writing on the firmware and plugging it back into your printer.

Defining the
```
#define SERIAL_PORT_2 6
```

Should connect your ESP-01s to Marlin's command interpreter, but since the firmware delivered on the ESP-01s is not working and the firmware in the repository `BIGTREETECH-SKR-PRO-1.1/firmware/esp-01s/` isn't working either, you will have to build it yourself from scratch. Try these [instructions](https://www.youtube.com/watch?v=pJGBRriNc9I). You can flash the firmware file to the WiFi module directly using the SD card. I haven't built the ESP-01s firmware myself, since I cannot use it in parallel with USB serial and LCD serial... Marlin only supports two COM ports.

UnderXirox, 2019-09-11:
[FR]
pour moi   ESP-01 fonction tres bien avec le port

    #define SERIAL_PORT_2 6

avec ip dans pronterface 192.168.0.1:8888  via  ESP-01

je ne veux pas avoir la sd avec l'USB juste 

le port com USB

    #define SERIAL_PORT_1 -1

ne fonction pas et ca depuis le debut (achat de la carte)

je veux juste pouvoir contrôler ma carte pas octoprint ou via le pc avec pronterface

Will only give you a serial port over the USB cable to control the printer

[EN]
for me ESP-01 function very well with the port

    #define SERIAL_PORT_2 6

with ip in pronterface 192.168.0.1:8888 yes via ESP-01

i do not want to have sd with usb

the USB com port

    #define SERIAL_PORT_1 -1
    //#define SERIAL_PORT_2 6

does not function and that since the beginning (purchase of the card)

i just want to be able to control my card via octoprint or via pc with pronterface

omahena, 2019-09-11:
I see what you mean. I am using OctoPrint over USB.

My exact define is as follows:
```
#define SERIAL_PORT -1   // Port for USB
```

Please note it is not `SERIAL_PORT_1` in case you have the _1 in there by mistake.

Your issue may also be in the platformio.ini file:
```
#
# Bigtreetech SKR Pro (STM32F407ZGT6 ARM Cortex-M4)
#
[env:BIGTREE_SKR_PRO]
platform = ststm32@5.4.3
framework = arduino
board = BigTree_SKR_Pro
extra_scripts = pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
build_flags = ${common.build_flags}
  -DUSBCON -DUSBD_USE_CDC -DUSBD_VID=0x0483 -DUSB_PRODUCT=\"STM32F407ZG\"
  -DTARGET_STM32F4 -DSTM32F407_5ZX -DVECT_TAB_OFFSET=0x8000 -DUSE_USB_HS_IN_FS
# -DHAVE_HWSERIAL6
lib_deps = ${common.lib_deps}
lib_ignore = Adafruit NeoPixel, SailfishLCD, SailfishRGB_LED, SlowSoftI2CMaster
src_filter = ${common.default_src_filter} +<src/HAL/HAL_STM32>
monitor_speed = 250000
```
I removed the HAVE_HWSERIAL6 and added USE_USB_HS_IN_FS to lower the chance of any potential GPIO pin conflicts.

Is the `-DUSBCON` missing in your config? It looks like it may be needed...

UnderXirox, 2019-09-11:
je viens de test idem avec votre solus   le probleme ne change pas 

toujours pas de carte detecte par les pc
  
juste que la carte n'est pas reconnu pare les pc deja a la base 
les pc  Windows 10 /7 et (Linux no virtual machine linux installer sur un pc)  ne  détecte meme pas comme  news  périphérique
pour linux que se soit avec lsusb et demsg ideme rien pas de périphérique connecter

UnderXirox, 2019-09-12:
oki 
measure the USB D+ and
D- to see if there is a short circuit

[image]

closed 2019-09-13 by UnderXirox

omahena, 2019-09-13:
Pas de problème! Bonne journée.

jbaribeault, 2019-10-14:
> `. Pardon my OCD but when you get a chance insulate your temperature sensor wires in T0 sensor connector. :D Shrink tubing maybe.`
> 
> yes is its test probe that does not connect normally ^^
> 
> yes there is a DC to D + and D- USB port HS
> view with the seller Thank you for your help
> 
> Pardon my English mad in FR ^^

Est-ce que le probleme etait un "short" entre D+ et D- sur le USB? A tu retournez la carte aux vendeur?

UnderXirox, 2019-10-14:
bonjour
pour le port  com entre la carte skr-pro et le pc
je passe par le wifi esp-01 avec esp3d dessus et avec Printrun via 192.168.0.1:8888

  `#define SERIAL_PORT_2 6`

 `A tu retournez la carte aux vendeur?`
oui

Hello
for the com port between the skr-pro card and the pc
I go through the wifi esp-01 with esp3d above Printrun via 192.168.0.1:8888

    `Do you return the card to the seller?
Yes

Pardon my English mad in FR ^^

HugeMissedSteak, 2019-11-08:
Your USB jumpers are wrong, you need to turn them 90 degrees.
[image]
As previously stated: currently you cannot access the SD-card through the pc, but you should see the board as COM device and be able to connect to it.

UnderXirox, 2020-01-26:
### Sorry for the response time

**yes** i noticed ... i changed place like on the doc and nothing nor changes
but I had the concern of the card received since
I go through ESP and it rolls except uploads gcode via esp I pass the lcd sd card (cr-10s)
I received a 2nd card and the USB function works well.

sorry for my english ...

abdulrehaman72, 2020-11-04:
Hi all, I have an issue while using BIGTREETECH SKR-PRO-1.1. After uploading the MARLIN 2.0 in to the board, I checked with each and every motor orientation using Pronterface.exe by connecting with USB (serial port's) . The motors were working fine. Now I'm trying to connect the board through pronterface, it's giving an issue like no serial port is connected. I checked thoroughly, no short-circuit in board, no wrong connections to the board,checked by connecting it with many laptops. Its showing the same issue like ' SERIAL  PORT NOT FOUND '.BL TOUCH is probing fine and input to the board is fine even TFT is glowing but it's not functioning. I tried resetting and rebooting the board still my issue was not resolved yet. My doubt is whether any internal issue with communication chip. I request you to give me a solution for this issue. Will be waiting for your support. Thank you.

ujfe71, 2021-12-17:
> Your USB jumpers are wrong, you need to turn them 90 degrees. [image] As previously stated: currently you cannot access the SD-card through the pc, but you should see the board as COM device and be able to connect to it.

Sorry but you are wrong.
The jumpers are ok, as mentioned in the manual of the board.
Now the user can acces the board over USB cable and Printerface for example. If the jumpers are in the direction you mentioned, there is no USB connaction...