bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #49: Unable to control fan1 or fan2 with M106/M107 commands

rlvanhuis, 2019-09-11:
Looking for the location of code lines to allow control of the parts cooling fans.  I was able to locate the extruder fan controls.  
Thanks

omahena, 2019-09-16:
You didn't provide any configuration details, so that is probably why nobody responded so far. It is not even clear if you are running a CNC machine or a 3D printer and how many fans you wish to drive. I am betting you have a 3D printer, please correct me if I am wrong. Also I will assume you are looking in the main Marlin repository (not sure what state the code is in here).

I use `Marlin\src\pins\stm32\pins_BIGTREE_SKR_PRO_V1.1.h`:
```
#define FAN_PIN            PC8  // Fan0
#define FAN1_PIN           PE5  // Fan1
#define FAN2_PIN           PE6  // Fan2
```

`Marlin\Configuration_adv.h`
```
#define E0_AUTO_FAN_PIN PE5
...
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed
```

The code you are looking for is likely in `Marlin\src\module\temperature.cpp`. What felt really annoying to me and significantly lowers the maintainability of the Marlin code (yet makes it more "human readable" if that is a factor important to you) was the marco magic that is used to handle PIN defines. 

In other words you may have specified `FAN1_PIN` somewhere in the code and you can tell that if you change the value it has an influence on the resulting firmware. But... if you search for `FAN1_PIN` in the code it is not used ANYWHERE. It was driving me nuts. Then I noticed that macros for pin defines were declared like PIN_EXISTS(FAN1)  and ANY_PIN(FAN1, FAN2) ... that are to be used without the _PIN postfix to make the code... prettier and save 4 bytes while wasting countless human hours... :D

From some of your other comments on the other issues, I think you simply connected the fans to fan0, fan1 and fan2 connectors and wish to control them using the M106 and M107 commands. Your wish seems reasonable but there are some reasons why this won't and shouldn't work.

Because of thermal management of Marlin not all fans may be controllable by G code. For example I can control my fan0 (part fan) and fan2 (not connected on my board - was surprised that worked). However Fan1 (hotend fan) will not respond to manual control but it will turn on during a print and function correctly.

M42 command may also be of interest to you and the undocumented I flag. See: `Marlin\src\gcode\control\M42.cpp`. But quite honestly I'd avoid using it on hotend fans for safety reasons.

Also keep in mind the `buildroot\share\PlatformIO\variants\BIGTREE_GENERIC_STM32F407_5X\PeripheralPins.c` seems to be all messed up and full of conflicts.

See: issue #36 

P.s.: You may not wish to drive a fan with a heater channel. It looks like it works... But... To me it feels like using a bulldozer to remove a leaf from the road. You may wish to also save the heater channels for say a case heater. PWM pins on the STM32F4 CPU will apparently have enough power to drive an LED but for a DC fan you'll need to step up the current a bit. See: [here](https://circuitdigest.com/microcontroller-projects/stm32f103c8t6-pwm-tutorial-fan-speed-control).

rlvanhuis, 2019-09-16:
I am trying to keep up with a very steep learning curve.  My apologies for
the lack of info.
System Information
3D Printer
SKR PRO V1.1
BIQU 5160 Drivers V1.2
Anet A8 Plus (X,Y, ZZ, E).
Single Hot End
Hot End Cooling Fan
Parts Cooling Fan
Heated Bed.
24V System Power Supply. (common to each supply point x3)
Logic power supplied by the board
ESP-01S, not working
TFT35 V3.0 not working
USB comms to PC, working normally.
System prints normally

Bed Heat connected to Hot Bed +/-
Hot End Heater to Heat0 +/-
Hot End Cooling Fan to Fan0 +/-
Parts Cooling Fan to Fan1 +/-

You have given me much to look into.
First, I noticed you used Fan1 for your Hot End Cooler.  I could not find
any reference to it in the literature.  Just thought I would follow
convention Heat0/Fan0, Heat1/Fan1.

I really appreciate the input.

Thank you

Randy

omahena, 2019-09-17:
No problem, thanks for the info. It is indeed a steep learning curve. I was aiming for a drop-in board replacement for my CR-10 mini mobo... You know... 2 days max. LOL! And now two months later I'm still tweaking the system... More of a DIY...

Our systems are very similar. The CR-10 mini is a (X,Y,Z,E) system and 12V on the bed heater. For now I am powering the board with the Printer PSU but RaspberryPi USB might be powering it later. I can't decide on how to include power delivery control into the system (RasPi driven relay vs. SKR driven). I don't think this will influence things too much unless the source doesn't provide enough power. A USB power meter might be useful here. I think the stepper drivers are then also powered by the mobo power source, but the motor power comes through the separate power in (obviously).

I got the ESP-01S integrated well enough to know it is seen by Marlin and that the mobo will flash the firmware to the WiFi board. However since I can't have 3 serial ports (WiFi, PC and TFT35) I gave up on the WiFi interface in favour of OctoPrint on a RasPi and TFT35 LCD control interface. Yesterday I realised that had I bought a newer version of the TFT35 like you have, I could plug the WiFi directly into the TFT35 and this way daisy chain the serial ports in effect having only 2 ports registered in marlin and the WiFi serial in the TFT35 LCD. I would recommend you try that... You will need to recompile the TFT firmware (BTT repos here on github) and the ESP-01S firmware as well I think. I couldn't make either publicly accessible firmwares work. The one delivered on the WiFI module offered an AP but failed to respond to HTTP and the second one didn't offer an AP and didn't connect to my WiFi as expected. I hope it doesn't require an old 2.4GHz AP. I wouldn't be surprised...

Your wiring looks fine. My OCD was also bending me towards convention Heat0/Fan0, Heat1/Fan1... But... Too many plugs, too few holes to satisfy the OCD so I just left them the way they were by default... I think. Technically I wouldn't mind a case fan as well, but for that I'd need a case... So... :D And no idea what I'll do if I'll ever switch to dual extruder. But your fan situation should be covered as well. If the fans or heater MOSFETs behave strangely it is probably due to conflicting PIN configurations of the platform.

We could exchange config files, but this thread is probably not the place.

My pleasure.

Ales.

rlvanhuis, 2019-09-17:
@omahena, attached is a complete file set.  It is my 31st iteration.  So far the best running.  Still have small issues.  I have most of it working now.  I am beginning to understand more of the board construction (regretfully).  I have been on this journey for about 2 months.  Not sure yet, but it appears the pin for Fan1 may not be connected.  Was able to get fan2 to work as my hotend fan with fan0 as the parts cooling fan.  I will assess if board/driver cooling is needed, if so I will parallel with the hotend fan.
[RLV31Marlin-bugfix-2.0.x.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3623002/RLV31Marlin-bugfix-2.0.x.zip)

Good Luck

Randy

omahena, 2019-09-18:
@rlvanhuis You seem to have the Marlin code from 2019-09-04, mine is a week newer, so there are some other differences. 

I included both configs and `pins_BIGTREE_SKR_PRO_V1.1.h` with some "pin conflicts removed". It works for me (tm). Make a backup of your file and try using mine and in platformio.ini also add `-DUSE_USB_HS_IN_FS` in your build_flags for BIGTREE_SKR_PRO. I also don't have the `-DHAVE_HWSERIAL6` define in there and it might add pin conflicts. Maybe try removing it just for fun.

I checked and PE_5 (aka PE5 elsewhere) is only used for the PWM pinmap. I **think** this means that it cannot be used for anything else. But... HAVE_HWSERIAL6 might use it for serial. I am not sure where and how this define is used and I can't find a mention in the source ode. It shouldn't have an influence on the PE5 pin... But my trust levels in this platform are pretty low atm. It's one of the differences we have between our repos. If you will move your WiFi to your TFT35 3.0... You don't need the Serial6 enabled anyway.

I seriously doubt the PE5 pin has an electrical issue... You could check the trace (on a powered off board).

Cooling doesn't seem to be needed on the electronics. Just my stepper motors get very hot on higher currents... So I added temp probes to E an Y axis steppers.

As a side note. I noticed your threshold settings for Hybrid mode are very high. For me this caused skipped steps and layer shifts. Trinamic support recommends stealthChop2 to be used up to 1 RPM. So my value of 50 is still to high. Ah, but I just noticed you also have HYBRID_THRESHOLD disabled and use stealthChop2 all the way. Any... print issues? I get perfect prints in prints restricted to spreadCycle. Boosting the current doesn't really help with skipped steps in SC2 on my system. I think I just need to finetune the hybrid threshold.

I hope this helps.

[config.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3625108/config.zip)

rlvanhuis, 2019-09-22:
For anyone looking.
I was able to get the expected response from my system using the following:
Fan0 for my parts cooling fan, controlled by program and TFT make no changes
Fan2 for my hotend cooling fan controlled by program in Auto (set the auto pin to PE6)
I have not tested yet, but just a SWAG...Fan1 is for the case if no changes are made.

closed 2019-09-22 by rlvanhuis