bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #51: [BUG] Part-Cooling fan always on

Haschtl, 2019-09-25:
Since the latest update the part-cooling fan is on after the board is started up.

My fan-configuration is as followed:

Configuration:
#define USE_CONTROLLER_FAN

Configuration_adv:
#define E0_AUTO_FAN_PIN PD3
#define E1_AUTO_FAN_PIN PD3
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define E5_AUTO_FAN_PIN -1
#define CHAMBER_AUTO_FAN_PIN PE6

pinfile:
#define FAN_PIN            PC8  // Fan0
#define FAN1_PIN           PE6 //PE5  // Fan1
#define FAN2_PIN           -1 //PE6  // Fan2
#define CONTROLLER_FAN_PIN PE5        // Set a custom pin for the controller fan

So fan1, chamber fan, controller fan and E0/E1 Auto fans are working. Only Fan0 is always on. 

I don't know if it's an electrical issue or software. I have an ultimaker 2 printhead with 2x 30mm 12V fans in series (because i'm running at 24V) and one fan died some time ago... I hope that didn't destroy the FAN0 mosfet

Haschtl, 2019-09-26:
Has anybody encountered similar issues with multiple fans like me?

JJHMARR2020, 2019-09-26:
//#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  //#define CONTROLLER_FAN_PIN -1           // Set a custom pin for the controller fan

What is the pin for this setting in Configuration_adv? Whatever pin it is set to will automatically turn on because it thinks it will cool stepper controllers or mosfets.

Haschtl, 2019-09-26:
I have uncommented the CONTROLLER_FAN_PIN in the Configuration_adv.
I will change the fan0 with fan1 to check if its an hardware issue or software.

Haschtl, 2019-09-27:
I now replaced the damaged 12V fan and the second one with two 24V fans parallel.
The green fan led on the SKR Pro lights up immediately after i connect the SKR to the 5V-supply.

I think the mosfet got damaged. Could this happen due to the broken fan? The fan has a short circuit so the current increases, which kills the mosfet? It's the [AO3400](https://www.digikey.de/product-detail/de/alpha-omega-semiconductor-inc/AO3400/AO3400-ND/7560344), am I right?

Haschtl, 2019-09-30:
I have found the problem. I didn't connect the two 12V fans in series but parrallel. The current must have been too high sometime and killed the mosfet for the FAN0. I'm now using an old Ramps RRD fan expansion board to control two fans with these pins: PD2, PE0

closed 2019-09-30 by Haschtl