bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #52: Endstops not working with TMC5160

MtPegasus, 2019-09-26:
Hello,

I have set up the board with TMC5160 on X/Y/Z-axis.
I use mechanical switches like before on my old board.
When the endstops switch the LED on the board goes on but the axis continues to move.

I tried different things like changing the logic, changing the pin and the port.
It does not work
Sensorless homing is off!

MtPegasus, 2019-09-26:
additional info: the diag pin is not connected

JJHMARR2020, 2019-09-26:
Does the Z axis acknowledge the endstops?

MtPegasus, 2019-09-26:
no she does not

JJHMARR2020, 2019-09-26:
Are the drivers running in SPI?

JJHMARR2020, 2019-09-26:
In the section where motor drivers are defined you should try changing from TMC5160 to TMC5160_Standalone; this disables SPI for the motors. This setting is found starting on line 650 in Configuration.h
There is a possibility it might fix the issue, but it doesn't explain why it was happening. I originally had a similar issue with my TMC2130's. I run all in SPI with Stallgaurd 2 on X and Y with two physical endstops for the dual Z axis. This board requires inverted logic for endstops.

MtPegasus, 2019-09-26:
Yes they are running in SPI. Im compiling now with TMC5160_Standalone.

MtPegasus, 2019-09-26:
ok, now the axes are not moving

JJHMARR2020, 2019-09-26:
I would connect to the board with Pronterface and send the command M119. 
Share your results to see the states of your endstops

emilhse, 2019-09-30:
> Hello,
>
> I have set up the board with TMC5160 on X/Y/Z-axis.
> I use mechanical switches like before on my old board.
> When the endstops switch the LED on the board goes on but the axis continues to [...]

can you please share how did you setup tmc5160 ?
some how i can't get it working please help

emilhse, 2019-09-30:
@MtPegasus can you please share your config files to set up tmc5160 with skr pro?

evandene, 2019-10-12:
Could it be you are using the row of connectors closest to de stepper drivers?
X Y Z is the other row

bdxgrs, 2019-10-30:
Bonjour, j'ai une erreur de compilation sur le firmware fourni  sur le site bigtreeteck pro v1.1 avec atom. Pourriez-vous partager firmware marlin  fonctionnel pour bigtreetech pro v1.1 SVP. ? J'ai des stepper 5160 et bl touch. Mais ce que je cherche, c'est déjà un firmware complet que je puisse copiler sans erreur pour bigtreeteck pro et qui ne plante pas de base parce qu'il ne reconnait la carte, ou qu'il manque le dossier build ..etc... merci d'avance

emilhse, 2019-11-29:
> Hello,
>
> I have set up the board with TMC5160 on X/Y/Z-axis.
> I use mechanical switches like before on my old board.
> When the endstops switch the LED on the board goes on but the axis continues to [...]

Please check if you have inverted endstops in Marlin settings because most likely board sees them as they are triggered ,that can cause the problem

dineshvyas, 2019-12-07:
can you post photoghtaph of your board with jumpers? this might give some clue.

emilhse, 2019-12-07:
Ok I will do that tomorrow

evandene, 2019-12-07:
> can you post photoghtaph of your board with jumpers? this might give some clue.
No need for jumpers when in SPI mode = standard. Works out of the box including end stops

MrLAtt, 2019-12-15:
I am having the same issue, but I am using sensorless homing with TMC2209 drivers.

the sensorless homing on the X and Y axis works perfectly, but the endstop on the Z Axis doesn't work at all. 

I'm sure I'm plugged into the right port, and that my endstop is triggering the correct way ( I've tried all the ports, and have switched between the endstop pulling low, and high)

When I send an M119 it says that all three are open, regardless of actual endstop status

can anyone help?

HugeMissedSteak, 2019-12-15:
If you use sensorless homing compatible drivers, then usually the stallguard pin needs to be disconnected (cut off) so that the mechanical switches will work. Alternatively you can change the firmware to use the endstops of a different input that is not shared with the stallguard (i.e. from an unsed axis)

[image]

MrLAtt, 2019-12-15:
Perfect, that (and then reversing the triggering of the endstop) solved the problem!

however, now i have a new problem, my extruder motor isnt moving. i've tried swapping driver boards, extruder motors, and switching the pins of E0 and E1 and then using the E1 'slot' instead, and nothing has solved it. 
As far as i can tell, i have ruled out the board, the driver and the motor. What could it be?

And its not the cold extrusion prevention, I am preheating the nozzel ;)

MrLAtt, 2019-12-15:
(the motor doesn't even seem to be powered up, since it doesn't have any holding strength either)

MrLAtt, 2019-12-16:
also, I've just noticed that when i have the extruder setup in the 'E1 Slot' the motor still wont move, but it will hold position, whereas it wont hold position in the 'E0 Slot'.

not sure if that helps in any way though

evandene, 2020-05-22:
It's now a few months ago and I wonder if you have managed to run you SKR PRO V1.1 board with the TMC5160 drivers

> Perfect, that (and then reversing the triggering of the endstop) solved the problem!
> 
> however, now i have a new problem, my extruder motor isnt moving. It worked for about 5 Minutes, then just stopped.
> 
> i've tried swapping driver boards, extruder motors, and switching the pins of E0 and E1 and then using the E1 'slot' instead, and nothing has solved it.
> 
> I also tried switching to the BigTreeTech pre-setup firmware, and it still didn't work.
> 
> As far as i can tell, i have ruled out the board, the driver and the motor. What could it be?
> 
> And its not the cold extrusion prevention, I am preheating the nozzel ;)

It's now a few months ago and I wonder if you have managed to run you SKR PRO V1.1 board with the TMC5160 drivers, did you? Could you please share your solution(s)?
Thanks

oldman4U, 2020-06-09:
Seems this ticket is not in use any more.

@thinkyhead (Could you please close this ticket due to inactivity)

Thank you

thinkyhead, 2020-06-14:
Only @trendelkamp or @bigtreetech can close this issue.

closed 2020-10-10 by MtPegasus