bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #54: TMC2208 and Z-homing

Livenmach, 2019-09-28:
I have SKR_Pro_V1.1 and TMC2208v3 drivers in uart mode using the dual z connection. I've gotten all 3 axis moving with commands in Pronterface properly. when I home the X,Y axis they seem to work ok however the Z-axis homes 1 motor at a time. 1 moves about 5mm then the other moves about 5mm then back to the other until it hits the limit switch. its also really loud. after that it moves fine and sounds great.??

HaythamB, 2019-10-05:
Hey. I have mine set up with individual Z drivers (one on the Z and the other on E1) and it works perfectly. Have you tried this setup? Or are you using a dual Z with a single driver?

Livenmach, 2019-10-14:
this board has the dual Z connectors with a single driver built into it. it works fine with the single driver and a separate splitter board.

oldman4U, 2020-06-08:
Hi Livenmach.

Would you please be so kind to close this ticket.

Thank you

closed 2020-06-09 by Livenmach