bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #55: Tmc5160 Cannot connect to Skr pro

emilhse, 2019-09-30:
tried everything can not somehow manage to make it work
please help

extesy, 2019-09-30:
This is the same issue as in https://github.com/MarlinFirmware/Marlin/issues/15415

piotrekm7, 2019-10-01:
They're working for me. You can check my configs [here](https://github.com/piotrekm7/Marlin/tree/dev-2.1.x/Marlin), maybe you'll notice some differences.

extesy, 2019-10-04:
@piotrekm7 I see that you are using `dev-2.1.x` branch which is a few weeks old and is not maintained anymore. Can you try reapplying your configs on top of the latest `bugfix-2.0.x` branch?

ogsv, 2019-10-08:
I'm also having this issue. Power is connected to both POWER and MOTOR_POW, and VCC jumper is set to IN, not USB, if that matters... Running the Marlin build from this repo, not from the MarlinFirmware repo.
M122 gives:
`Driver registers:
		X	0xFF:FF:FF:FF	 Bad response!
		Y	0xFF:FF:FF:FF	 Bad response!
		E	0xFF:FF:FF:FF	 Bad response!
Testing X connection... Error: All HIGH
Testing Y connection... Error: All HIGH
Testing E connection... Error: All HIGH`

piotrekm7, 2019-10-11:
@extesy I used tmc 5160 with 2.0 before. About power supply, I have 28V on MOTOR_POW. Should work with 12V, but who knows, maybe that's an issue.

piotrekm7, 2019-10-11:
You can even flash my firmware and check if motors are working.
Note that it's corexy kinematics, will be fine for test though.
I have usb connection disabled, but can recompile if you wish.
[firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3719483/firmware.zip)

SlavaAn, 2019-11-24:
In my case it (firmware.zip) work! Big thanks piotrekm7. But USB-PC link don't work.

emilhse, 2019-11-29:
thank you very much i have sorted the problem

closed 2019-11-29 by emilhse

piotrekm7, 2019-12-03:
> In my case it (firmware.zip) work! Big thanks piotrekm7, I try to make my code 2 week without result. But USB-PC link don't work.

I have setup for lcd and esp8266. If you want to use USB you will have to change serial configuration and rebuild the project. 
You can clone https://github.com/piotrekm7/Marlin and in Configuration.h change:
**#define SERIAL_PORT 6**
to:
**#define SERIAL_PORT -1**