bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #56: Third serial port possible?

Haschtl, 2019-10-04:
Hello, 
I am using the Skr pro with a raspberry running octoprint as the main host. I'm using the ESP3D as a host, too. 
And i also have a touchscreen connected to the raspberry. Until now I'm using the Octoprint API to send Gcode via the touchscreen to the Skr pro. But this API has very limited possibilities, as i cannot read the serial output. 

So i can only rely on some octoprint-flags like "printing" oder "operational". But I cannot get the position or any serial output from marlin. 

Is it possible to use the UART-connector on the skr pro for a third serial interface? Or is there another way using spi or i2c?

extesy, 2019-10-04:
@Haschtl Marlin doesn't currently support 3 serial ports. It doesn't have anything to do with a board, it's a firmware limitation. Feel free to create a feature request for Marlin team.

omahena, 2019-10-05:
@Haschtl I am not sure I follow how you use the OctoPrint API via the touchscreen. Also not sure what touchscreen you are using. But considering the current limitation of Marlin to only support two serial ports at the time, I recently saw the option to plug the ESP3D WiFi module into the TFT35 V2.0 and V3.0 LCDs. This should let you use OctoPrint using the USB Serial interface, the TFT35 probably allows direct touch control and I assume the ESP3D WiFi module also functions when plugged into the TFT35 board.

**PLEASE NOTE:** I currently do not believe a single claim put forward by BigTreeTech since in my experience a lot of these claims are just hot air. So if it not verified and confirmed by me or a trusted 3rd party - I just don't believe them.

I have a TFT35 V1.2 that they "bundled" into my AliExpress purchase... So I cannot verify the TFT35 V2.0 and V3.0 work. From what I hear from a trusted 3rd party, the TFT35 V3.0 firmware experience is just as problematic as the SKR Pro 1.1 experience. USB key insertion will freeze the device and ESP3D WiFi apparently doesn't work. I bring up the potential solution because I do not have any info on the V2.0 device and well... Perhaps we missed something when testing the V3.0 version.

My recommendation is that if you intend to try this, that you buy your BTT stuff somewhere where you can send them back if they don't work as advertised. Please double check everything before you spend money.

If anyone can provide additional test results... I am all ears.

Haschtl, 2019-10-08:
Btw. My opinion to the SKR Pro 1.1 is very good... its a good boad with a lot of additional gpio's. 
I have connected 3 fans + two fans with external mosfets, 4 temperature sensors, two hotends, one light, a BLTouch sensor, an SPI Oled, an encoder, speaker, sd-extention, esp3d, a laser, two filament sensors and all is working properly now (only the esp3d hotspot is not avilable when connected to the skr). 
I have also most of the marlin-features activated. For e.g. the power-supply control, automatic controller and hotend fans.

The only things I dont like is: 
- flashing with sd-card (over usb would be much easier in my setup) 
- the lack of support and incomplete  pins-file

But with my huge setup i need to adjust the pins manually anyway, as a predefined pin-configuration would not fit my needs probably.

extesy, 2019-10-08:
@Haschtl Since you have so many external inputs enabled and working, do you mind sharing your full config files and any Marlin modification you've done to make it all work?

Haschtl, 2019-10-09:
Just have a look at my [MarlinFirmware-Fork](https://github.com/Haschtl/Marlin/tree/bugfix-2.0.x)

I have edited the following files:
**General configuration:**
- Marlin/Configuration.h
- Marlin/Configuration_adv.h
- Marlin/src/pins/stm32/pins_BIGTREE_SKR_PRO_V1.1.h

**Custom modifications**
I have calibrated all of my temperature sensors by it's own. Therefore I have replaced some thermistor-Lookup-tables

- Marlin/src/module/thermistor/thermistor_11.h
- Marlin/src/module/thermistor/thermistor_12.h
- Marlin/src/module/thermistor/thermistor_13.h
- Marlin/src/module/thermistor/thermistor_20.h

** PID - Fan - scaling **
I added an additional PID-value to compensate temperature drops, if the part fan is enabled/disabled.
Therefore I made changes in the following files:

- Marlin/src/module/temperature.cpp
- Marlin/src/module/temperature.h
- Marlin/src/lcd/menu/menu_advanced.cpp
- Marlin/src/module/configuration_store.cpp
- Marlin/src/gcode/config/M301.cpp
- Marlin/src/inc/SanityCheck.h

** Laser modifications **
I have also some changes made in Marlin/src/HAL/HAL_STM32/fastio.h. I had problems to define the SPINDLE_LASER_PWM_PIN. I don't know If this is still needed.
```
//#define PWM_PIN(P)              digitalPinHasPWM(P)
#define PWM_PIN(P)              true
```

** Added a custom controller **
I have a selfmade controller with an encoder and an I2C-OLED Display. 

Therefore I added `#define U8GLIB_SSD1306_CONTROLLER`, which enables the following in Marlin/src/inc/Conditionals_LCD.h:
```
#define U8GLIB_SSD1306
#define IS_ULTIPANEL
```

** Other stuff you will not need **
- Marlin/_Bootscreen.h
- Marlin/src/lcd/language/language_de.h
- Marlin/src/lcd/language/language_en.h

I have also added the ESP3D Firmware to my repository. I can flash the firmware with the ArduinoIDE and it is also running, while connected to the PC, but when I connect it to the SKR, the hotspot doesn't show up.

Short pin overview:
```
#define X_MIN_PIN          PB10
#define X_MAX_PIN          -1 //PE15 - is used for filament sensor
#define Y_MIN_PIN          -1 //PE12
#define Y_MAX_PIN          PE12 //PE10  - is used for filament sensor
#define Z_MIN_PIN          PG5 //PG8
#define Z_MAX_PIN          PG8 //PG5

#define Z_MIN_PROBE_PIN  PA2

//
// Steppers
//
#define X_STEP_PIN         PE9
#define X_DIR_PIN          PF1
#define X_ENABLE_PIN       PF2
#ifndef X_CS_PIN
  #define X_CS_PIN         PA15
#endif

#define Y_STEP_PIN         PE11
#define Y_DIR_PIN          PE8
#define Y_ENABLE_PIN       PD7
 #ifndef Y_CS_PIN
  #define Y_CS_PIN         PB8
#endif

#define Z_STEP_PIN         PE13
#define Z_DIR_PIN          PC2
#define Z_ENABLE_PIN       PC0
#ifndef Z_CS_PIN
  #define Z_CS_PIN         PB9
#endif

#define E0_STEP_PIN        PE14
#define E0_DIR_PIN         PA0
#define E0_ENABLE_PIN      PC3
#ifndef E0_CS_PIN
  #define E0_CS_PIN        PB3
#endif

#define E1_STEP_PIN        PD15
#define E1_DIR_PIN         PE7
#define E1_ENABLE_PIN      PA3
#ifndef E1_CS_PIN
  #define E1_CS_PIN        PG15
#endif

//
// Software SPI pins for TMC2130 stepper drivers
//
#if ENABLED(TMC_USE_SW_SPI)
  #ifndef TMC_SW_MOSI
    #define TMC_SW_MOSI    PC12
  #endif
  #ifndef TMC_SW_MISO
    #define TMC_SW_MISO    PC11
  #endif
  #ifndef TMC_SW_SCK
    #define TMC_SW_SCK     PC10
  #endif
#endif

#define TEMP_0_PIN         PF9
#define TEMP_1_PIN         PF5  // T2 <-> E1
#define TEMP_BED_PIN       PF3  // T0 <-> Bed
#define TEMP_CHAMBER_PIN   PF6

#define HEATER_0_PIN       PB1  // Heater0
#define HEATER_1_PIN       PD14 // Heater1
#define HEATER_2_PIN       PB0  // Heater1
#define HEATER_BED_PIN     PD12 // Hotbed
#define FAN_PIN            PD2 //RRD-Expansion1     //PC8  //Fan0
#define FAN1_PIN           PE6 //                     PE5  // Fan1
#define FAN2_PIN           PE0  //RRD-Expansion2 //PE6  // Fan2

#define SDSS               PB12
#define BEEPER_PIN       PG4
#define BTN_ENC          PA8

#define PS_ON_PIN          PG13
#define FIL_RUNOUT_PIN     PE15
#define FIL_RUNOUT2_PIN     PE10

#define SPINDLE_LASER_ENA_PIN     PG11   // Pin should have a pullup/pulldown!
#define SPINDLE_LASER_PWM_PIN     PC9   // MUST BE HARDWARE PWM ?
#define SPINDLE_DIR_PIN           PC1

#define CASE_LIGHT_PIN  PF8   // MUST BE HARDWARE PWM

#define CONTROLLER_FAN_PIN PE5        // Set a custom pin for the controller fan

#define BTN_EN1        PG10
#define BTN_EN2        PF11  // A2 - rotary encoder B
#define BTN_ENC        PA8  // A3 - rotary encoder push switch

#define CALIBRATION_PIN PC5 // Override in pins.h or set to -1 to use your Z endstop

#define HEATER_CHAMBER_PIN       PD11

#define E0_AUTO_FAN_PIN PD3
#define E1_AUTO_FAN_PIN PD3
#define CHAMBER_AUTO_FAN_PIN PE6
```

Some pins in my configuration are switched with no practical sense. Like the heater, temp or fan pins. These should be the default-pins more or less to be less disheveling

closed 2019-10-09 by Haschtl

Haschtl, 2019-10-13:
https://youtu.be/9TnhRrMQPQw

I made a video demonstrating what is possible with a touch interface using the octoprint serial log. 
I have all settings included in the gui including a visualization of the bed-level mesh. 
The octoprint serial log costs a lot of ressources, so having an extra serial connection would be better. But for now its pretty good

extesy, 2019-10-13:
@Haschtl Very impressive! What are you using for that UI?

Haschtl, 2019-10-13:
I have a raspberry pi 4 running Octoprint
I'm using PyQt for the GUI together with the OctoprintAPI to send gcode to the controller and the serial-log to parse the output. 
Ah... and I have included another piece of software from me ([RTOC](https://github.com/Haschtl/RealTimeOpenControl)) to collect data and create the graph for the temperatures, etc. 

The GUI itself is a dirty piece of software... There is a lot hardcoded stuff especially for my printer-setup (dual extruder, chamber, TMC's, ...) and also the stuff connected to the raspberry pi (power-consumption, switches for the enclosure-doors, camera, two temperature sensors, smoke sensor).