bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #57: SKR Pro V1.1 issue with labeling of hotbed output

Reick, 2019-10-08:
I have connected the output of the hotbed to a SSR and the SSR dont work, because minus and plus are twisted. There are a labeling issue on the documents and pcb. See the picture: https://imgur.com/KfVhJQn
https://imgur.com/y4b2Drl
https://imgur.com/NMEPPR2

Meddleglitch, 2019-10-09:
Yep. Mine is exactly like this, too.
Glad somebody else pointed it out.

viper93458, 2019-10-12:
I determined the same and never thought to post it. :(

UnderXirox, 2019-10-14:
Hello 

Mine is exactly like this too

dingbatca, 2019-10-24:
Bumping this.  Also confirming this issue.

Haschtl, 2019-10-25:
Can confirm this too. 
But I did simply switch the connectors because I thought I connected it wrong. Didn't think of that issue

dch1921, 2019-11-09:
Originally posted about this on July 20 in the Bigtreetech SKR Pro user group.

DroneMang, 2019-12-06:
Yep, just figured this out myself!

sneakym8, 2019-12-08:
Must be all like that. Mine is also.

turnali, 2020-01-23:
I m having a problem with my heatbed. My skr pro card is attached to a ssr and to a silicon heatbed. The problem is, as soon as i turn on the card, the silicon bed starts to heat up. The connection between bed and the ssr is correct. The ssr is working properly because i used another ssr and still have same problem. The ssr is connected to the hotbed input on the card and the bedpower input is to the power supply. I even tried to get the bed power directly from the power supply but still have the same problem. I used a normal 12v heatbed on the card to check if the problem is on the card but it heats normally so the problem is the connection i think. Do you have any suggestions to my problem?

PaschyS, 2020-01-23:
> I m having a problem with my heatbed. My skr pro card is attached to a ssr and to a silicon heatbed. The problem is, as soon as i turn on the card, the silicon bed starts to heat up. The connection [...]

What SSR do you use? I had similar problems with an AC Silicon heating bed in combination with the original SSR in my printer. What firmware configuration do you made?

turnali, 2020-01-23:
i use ''fotek ssr 40 da''. my friend made the configuration for me. can you send a screenshot of your config. please? maybe its because of the config.

turnali, 2020-01-23:
how did you solve your problem?

turnali, 2020-01-23:
> What SSR do you use? I had similar problems with an AC Silicon heating bed in combination with the original SSR in my printer. What firmware configuration do you made?

i use ''fotek ssr 40 da''. my friend made the configuration for me. can you send a screenshot of your config. please? maybe its because of the config.
how did you solve your problem?

PaschyS, 2020-01-23:
Do you use a 12v DC heatbed? You have the wrong SSR when you use the DA this means DC input for control and AC for output. DD means DC control input DC output.  You must use a 40Amper or better more Amps. And do you need a heat sink.

turnali, 2020-01-23:
> Do you use a 12v DC heatbed? You have the wrong SSR when you use the DA this means DC input for control and AC for output. DD means DC control input DC output. You must use a 40Amper or better more Amps. And do you need a heat sink.

my heatbed is a 5mm aluminum plate with a silicon heatbed under it. on top, its a magnetic pei spring steel plate from Twotrees. My power supply is a Meanwell LRS-350-12 12VDC 29.0A AC/DC. I have the same setup in two different printers but they have different control boards. This is the first time i am having this problem. i dont need a heatsink

PaschyS, 2020-01-23:
When you use a 12v Silicon Heatbed then is the Fotek DA the wrong SSR. I had the same problem with a 230V Heatbed, then I changed the SSR to a AC Version instead of a DC version and it works great.

turnali, 2020-01-23:
> When you use a 12v Silicon Heatbed then is the Fotek DA the wrong SSR. I had the same problem with a 110-230V AC Heatbed, then I changed the SSR to a AC Version instead of a DC version and it works great. And Importend when you use a 12v/24v Heatbed do you need a heatsink for the ssr, it will reach high Temps with no cooling.
> [image]

i understand  but the ssr i use is the same one in two different printers with the same silicon beds. they have no problems. even my friend who made fixed the firmware for me uses the same ssr and silicon bed in his machines as well and he owns 3d printer service store.  with this one, when i turn on the machine, the bed starts to heat up and continue to heat up. it doesnt stop if set it at a certain degree from the screen or even by gcode. before it wasnt getting any power because the wiring connection was wrong until i read this post. so i switch the +- wire positions. its getting power now but it heats up by itself. ssr wires goes to ''hotbed'' output and ''bedpower'' output goes to the power supply. so the connection is right. My friend couldnt solve this problem either because he never worked on a SKR Pro board before. the board output connections are different than other boards

PaschyS, 2020-01-23:
I understand you and your problem. That I don't get you wrong now.  Are you using a 12v heating bed or a 110 / 230v? And which one, have you a link? 
You can messure that input at the ssr. When the board don't switch on the ssr and the Heatbed works, than im right. When not the board switch on the ssr. My ssr have a led where can I look if the ssr works or not.

turnali, 2020-01-23:
the silicon bed is 220v 250W. my ssr has a led as well. it turns on as soon as i turn on the printer and start heating up the bed. in my other printers, the led turns on only when the bed is heating and when its not heat, the led turns off

> I understand you and your problem. That I don't get you wrong now. Are you using a 12v heating bed or a 110 / 230v? And which one, have you a link?
> You can messure that input at the ssr. When the [...]

the silicon bed is 220v 250W. my ssr has a led as well. it turns on as soon as i turn on the printer and start heating up the bed. in my other printers, the led turns on only when the bed is heating and when its not heat, the led turns off

PaschyS, 2020-01-23:
Can you measure up the bed output from the board without ssr? 
At the Board are leds for the outputs, when the output is green , then is the mosfet on. 
[image]

The only thing what I use in the configs is PID for the bed with bang bang.

AnHardt, 2020-01-23:
@turnali 
You did find this issue. Did you read it? Did you understand the consequences?

Did you try to connect the SSR with the opposite polarity?
Did you power the bed-mosfet on its own connector to the power supply?

turnali, 2020-01-23:
> @turnali
> You did find this issue. Did you read it? Did you understand the consequences?
> 
> Did you try to connect the SSR with the opposite polarity?
> Did you power the bed-mosfet on its own connector to the power supply?

ofcourse i connected the ssr with opposite polarity. if you read my conversation with ''PaschyS '', i couldnt make the heatbed and ssr work until i found this post.
And yes to your other question as well. its also in my conversation with ''PaschyS''. you can read it

turnali, 2020-01-23:
> Can you measure up the bed output from the board without ssr?
> At the Board are leds for the outputs, when the output is green , then is the mosfet on.
> [image]
> [image]
> 
> The only thing what I use in the configs is PID for the bed with bang bang.

i will measure up the bed output without the ssr tomorrow and let you know the result. But as i mentioned before, i attached a 12v heatbed without the ssr to the skr pro board from another printer  just to see if the board and bed connections are working properly. And it was working fine with no issues. i think mosfet led was green but i will again tomorrow

turnali, 2020-01-29:
I was able to stop the heating by itself problem but this time the bed doesnt heat at all :))).  Everything works fine on the board; output connections, current, leds....  I connected the no 1 output of the SSR and other bed wire together to L or N on the power supply and it stopped the heating but...
Before, the connection was like the diagram below.
Still couldnt fixed this stupid problem

[image]

GadgetAngel, 2020-01-30:
@turnali Here is how I would hook up your situation: see image below.  What I see are two problems. You do not have Neutral hooked up to the SSR AC side.  Also You do not need to hook to the BED POWER terminals.  The whole point of using an SSR is to power the AC Silicon Heatbed from the wall AC. The connection to the SKR Pro is just to control the SSR switch side. The SSR switch side will allow AC Voltage and current to flow into the AC Silicon Heated Bed.  Therefore, DO NOT use the Bed Power terminals on the SKR PRO (those are meant to hook up to the Bed directly without an AC SSR).  You need to power the MCU (SKR PRO) so that the board can control the SSR, so add the 12VDC power to the POWER termials on the SKR PRO.  Now plug up to the AC and the MCU will control the SSR and heat up the BED only when the SKR Pro turns on the SSR.
[SSR Hookup -needs Neuutral hooked to SSR and 12vdc power hooked to BOARD power.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4133697/SSR.Hookup.-needs.Neuutral.hooked.to.SSR.and.12vdc.power.hooked.to.BOARD.power.pdf)

GadgetAngel, 2020-01-30:
Here is how I would hook up your situation: see image below. What I see are two problems. You do not have Ground hooked up to the SSR AC side. Also You do not need to hook to the BED POWER terminals. The whole point of using an SSR is to power the AC Silicon Heatbed from the wall AC. The connection to the SKR Pro is just to control the SSR switch side. The SSR switch side will allow AC Voltage and current to flow into the AC Silicon Heated Bed. Therefore, DO NOT use the Bed Power terminals on the SKR PRO (those are meant to hook up to the Bed directly without an AC SSR). You need to power the MCU (SKR PRO) so that the board can control the SSR, so add the 12VDC power to the POWER termials on the SKR PRO. Now plug up to the AC and the MCU will control the SSR and heat up the BED only when the SKR Pro turns on the SSR.
[SSR Hookup -needs Ground hooked to SSR and 12vdc power hooked to BOARD power.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4133987/SSR.Hookup.-needs.Ground.hooked.to.SSR.and.12vdc.power.hooked.to.BOARD.power.pdf)
[image]

GadgetAngel, 2020-01-30:
> I was able to stop the heating by itself problem but this time the bed doesnt heat at all :))). Everything works fine on the board; output connections, current, leds.... I connected the no 1 output [...]

@turnali Here is how I would hook up your situation: see image below. What I see are two problems. You do not have Ground hooked up to the SSR AC side. Also You do not need to hook to the BED POWER terminals. The whole point of using an SSR is to power the AC Silicon Heatbed from the wall AC. The connection to the SKR Pro is just to control the SSR switch side. The SSR switch side will allow AC Voltage and current to flow into the AC Silicon Heated Bed. Therefore, DO NOT use the Bed Power terminals on the SKR PRO (those are meant to hook up to the Bed directly without an AC SSR). You need to power the MCU (SKR PRO) so that the board can control the SSR, so add the 12VDC power to the POWER termials on the SKR PRO. Now plug up to the AC and the MCU will control the SSR and heat up the BED only when the SKR Pro turns on the SSR.
[SSR Hookup -needs Ground hooked to SSR and 12vdc power hooked to BOARD power.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4134000/SSR.Hookup.-needs.Ground.hooked.to.SSR.and.12vdc.power.hooked.to.BOARD.power.pdf)
[image]

turnali, 2020-01-30:
> @turnali Here is how I would hook up your situation: see image below. What I see are two problems. You do not have Ground hooked up to the SSR AC side. Also You do not need to hook to the BED POWER [...]

The green ground wire output connection you draw from the ssr is a mounting screw hole. Its the same as the one below. Its not a output hole. There are only 4 outputs on the ssr. Numbers 1,2,3 and 4. Number 1 goes from the ssr to the power supply L or N. Number 2 comes from the silicon heat bed and the other part of that wire goes to the power supply as well to L or N. Numbers 3 and 4 are Positive and Negative outputs that goes to the control boards Hotbed outputs.

turnali, 2020-01-30:
But power connections you mentioned, i will try them

turnali, 2020-01-30:
> @turnali Here is how I would hook up your situation: see image below. What I see are two problems. You do not have Ground hooked up to the SSR AC side. Also You do not need to hook to the BED POWER [...]

The green ground wire output connection you draw from the ssr is a mounting screw hole. Its the same as the one below. Its not a output hole. There are only 4 outputs on the ssr. Numbers 1,2,3 and 4. Number 1 goes from the ssr to the power supply L or N. Number 2 comes from the silicon heat bed and the other part of that wire goes to the power supply as well to L or N. Numbers 3 and 4 are Positive and Negative outputs that goes to the control boards Hotbed outputs.
The board is already connected to the power supply. I just didnt draw that. I also try only use Hotbed output and left the Bed Power empty but this time the heat bed didnt get power. Its a 220v silicone heat bed, i have to use Main Power connections( L and N ) to power it.

GadgetAngel, 2020-01-31:
The back part of the SSR should be a grounding plate. So if the connection on the other side is a Hole use that one. Just because the connection on the side of the AC is an open hole does not mean you should not connect up the ground. Connecting up ground is a safety precaution. If you do not connect ground you do it at your own risk. The SSR will still work with or without ground hooked up. Good luck! Did you try disconnecting the power to the input Power bed and connect up the power to the Board input power only? Did that work?

turnali, 2020-01-31:
> The back part of the SSR should be a grounding plate. So if the connection on the other side is a Hole use that one. Just because the connection on the side of the AC is an open hole does not mean [...]

i tried that but the board didnt heat up the bed

dch1921, 2020-02-01:
You still need to hook up the bed power or you will not power the SSR control. Review the schematic and it will make sense

GadgetAngel, 2020-02-01:
> You still need to hook up the bed power or you will not power the SSR control. Review the schematic and it will make sense

You mean you still have to connect up the input Bed power??? So all the power connection have to be made?? I will look at the schematic tomorrow to see what you are saying

GadgetAngel, 2020-02-01:
@turnali Do you have a 12VDC bed to hook up to the SKR pro just to see if the board can heat a 12V bed. Maybe you have a bad SKR pro board or a bad  SSR

turnali, 2020-02-01:
> @turnali Do you have a 12VDC bed to hook up to the SKR pro just to see if the board can heat a 12V bed?? Maybe you have a bad SKR pro board or a bad SSR.

The board works with a 12V bed i checked that. I also tried the ssr and the bed with a SKR v1.3 board and they work fine as well. Tried another SSR on Pro board but same problem again. 
I think its the Pro boards unique bed powering inputs with the SSR is causing the problem. Seperation of the Bed Power and Hot Bed connections on the board might be it. its not working with the SSR.
I even thought it might be a faulty On/Off switch so i cancelled it and connected the Main power(220v) directly to the power supply but didnt work as well.
I sent a message to BigTree but because of the Chinese New Year holiday they hvent answered it yet.

GadgetAngel, 2020-02-01:
> The board works with a 12V bed i checked that. I also tried the ssr and the bed with a SKR v1.3 board and they work fine as well. Tried another SSR on Pro board but same problem again.
> I think its [...]

Did you send a message to BIGTREETECH support via their Facebook page? Please post their response back here because I am planning on powering my bed with AC power and SSR too. I am still testing out my SKR Pro and I am still waiting for parts to test out the SSR configuration. I am testing out the Driver hookup on my board now. I hope BIGTREETECH helps.

turnali, 2020-02-01:
> Did you send a message to BIGTREETECH support via their Facebook page? Please post their response back here because I am planning on powering my bed with AC power and SSR too. I am still testing out [...]

I contact BigTree via their Aliexpress store. They respond faster from there. 
If they give me a solution for my problem, I will definitely post it here

GadgetAngel, 2020-02-01:
@turnali Did you wire the CLK pin to GND pin on your TMC5160 driver boards?? Here is a .pdf that shows the driver configuration for the SKR Pro board https://doc-0k-c4-docs.googleusercontent.com/docs/securesc/seim4417k8n16rvc6rl28ib17otc3iek/3ohsos0r5999avqtuin34rmgbvo6bq7t/1580551200000/06483777329211841429/05169398407596378373/1CR7NcZsWs012gB1fWaCSrZSwWh_6bvzq?e=download&nonce=atcujea3apa26&user=05169398407596378373&authuser=0&hash=rsfcp8t4m3890olu52lalaj8va4kdtqd

turnali, 2020-02-01:
> @turnali Did you wire the CLK pin to GND pin on your TMC5160 driver boards?? Here is a .pdf that shows the driver configuration for the SKR Pro board [...]

I am using TMC2209 V3.0 drivers and your link doesnt open

GadgetAngel, 2020-02-01:
> I am using TMC2209 V3.0 drivers and your link doesnt open

Sorry about that, trying to help. Two people at once. Please ignore my last comment it only applies to TMC5160 drivers. Funny I tested the link and it works for me. If you want the driver configuration for SKR PRO board try this one: https://drive.google.com/file/d/1AK4_RM3cvM4ar4Fx1Kf7UHncVgKvZTg5/view?usp=drivesdk

turnali, 2020-02-01:
> Sorry about that, trying to help. Two people at once. Please ignore my last comment it only applies to TMC5160 drivers. Funny I tested the link and it works for me. If you want the driver [...]

Thanks a lot, but the other board(skr v1.3) that i tried the ssr and the silicone bed also have the same drivers(TMC2209 V3.0). So its not the drivers. 
I am just gonna wait an answer from BigTree for now.
Thanks for your help

DroneMang, 2020-02-01:
I'm not sure if this was answered above for you guys but there needs to be
12-volt or 24-volt power from your power supply going to each of the three
input power ports on the board. As in you need to have power going to the
motor power the board power and the bed power. Otherwise I think you're
wiring diagram looks fine and you already mentioned switching the plus and
minus going to the SSR. I apologize if this was already discussed above. I
have TMC 2209s working with sensorless homing on an FT-6 which is set up
for 24 volts and has 110 volt bed, I could post a config file here if you
would like? It's Marlin bug fix 2.0. Good luck!

turnali, 2020-02-01:
> I'm not sure if this was answered above for you guys but there needs to be 12-volt or 24-volt power from your power supply going to each of the three input power ports on the board. As in you need [...]

On the diagram i draw, i only draw my bed connections. All my other necessary connections are already connected just not drawn on the diagram.
I would appreciate if you post your connection config file just to compare it with mine. 
On you 110v bed, do you use a SSR? if yes, 
How did you make the connection to your board and power suplly?
Is there any difference compare to mine?

turnali, 2020-02-01:
> I'm not sure if this was answered above for you guys but there needs to be 12-volt or 24-volt power from your power supply going to each of the three input power ports on the board. As in you need [...]

On the diagram i only draw my bed connections. All my other necessary connections are already connected just not drawn.
I would appreciate if you post your connection config file just to compare it with mine.
On your 110v bed, do you use a SSR? if yes,
How did you make the connection to your board and power supply?
Is there any difference compare to mine?

GadgetAngel, 2020-02-02:
> > I'm not sure if this was answered above for you guys but there needs to be 12-volt or 24-volt power from your power supply going to each of the three input power ports on the board. As in you need to have power going to the motor power the board power and the bed power. Otherwise I think you're wiring diagram looks fine and you already mentioned switching the plus and minus going to the SSR. I apologize if this was already discussed above. I have TMC 2209s working with sensorless homing on an FT-6 which is set up for 24 volts and has 110 volt bed, I could post a config file here if you would like? It's Marlin bug fix 2.0. Good luck!
> > […](#)
> 
> On the diagram i only draw my bed connections. All my other necessary connections are already connected just not drawn.
> I would appreciate if you post your connection config file just to compare it with mine.
> On your 110v bed, do you use a SSR? if yes,
> How did you make the connection to your board and power supply?
> Is there any difference compare to mine?

I am right now in the process of checking out the SKR PRO V1.1 board. I am currently only checking the stepper drivers but I will be getting to setting up the SSR. I just want to make one change at a time, so if I run into a problem I will know where the issue is located. The diagram I sent was the way I am planning on making the connection.