bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #58: SKR Pro v1.1 with A4988

oldman4U, 2019-10-11:
Hi. Hope someone can help me. The board works fine, which mens i can upload new firmware and i can control fans and other hardware devices using PrintRun for Mac. But when i try to control the stepper motor, all i get is a vibrating motor which does not spin. Changing the Vref did not help. The same motor with the same cable works fine using a SKR E3 DIP. I use the fw from this github and changed only the stepper motor section to A4988. Any help would be appreciated. Georg

[image]
[image]
[image]
[image]
[image]
[image]

cobrettee, 2019-10-12:
I am having the same issues with my CoreXY build on A4988s and DRV8825s. Everything else about the board works but steppers are a huge problem right now - VREF changes not helping and if I put my ramps card back in the steppers are fine. If it wasn't for the fact that I already order 2209's I would have put the card back in the box and asked for refund. I did get a little further though after several hours testing - if I lower my steps and feed rate to certain values then they actually move albeit incorrect steps - e.g I say move 10mm it moves 100mm. Now you would think simple math would apply and I could adjust my steps and rate accordingly based on that but as soon as I change the values to the correct ones the steppers refuse to move. If I try values in between it makes jerk movements or beeps and sometimes even just goes into kill mode because the stepper sounds like it is trying to move and heats up. Very frustrating to say the least. Is it that maybe we should have a different jumper configuration that actually correctly enables microstepping for these 4988/8825s?

cobrettee, 2019-10-12:
That was it - you have to use the MS1,MS2 and MS3 pins according to your stepping. Works now.

oldman4U, 2019-10-12:
Hello cobrettee.

Thank you for your answer.

According to the data sheet, I have to shorten 1, 2 and 3 to get 16 micro steps.

[image]

Anyway, I also tried all the other combinations and nothing changed. So you believe switching to 2209 will solve the problem. I tried 2208s from a working SKR E3 DIP and had the same problem, but please let me know in case it makes a difference for you. Finally I believe the board is broken and now I do not know if I should order the same board again or something else. It would be great to know if there is a single user out there, using the SKR v1.1 PRO with A4988 successfully. If so it would be great to know the jumper setting and maybe to get a Firmware file, so I can upload it to my board.

Georg

cobrettee, 2019-10-12:
No I just happened to have ordered 2209s at the same time. There is no reason these 4988/8825s should not work. I can only tell you that I set my jumpers according to what was in the Marlin configuration_adv.h file and now my steppers move and more importantly correctly move. When I say 10mm it moves 10mm as per the calculations for belt driven(xy) and leadscrew(z) driven 1/32 steppings. Did you try the order if looking from right to left at the pins L,L,H?

RudyBenner, 2019-10-13:
Looking at your cable, I would say that you have moved 2 wires incorrectly. The windings as shown are in series which leads to low torque and ambiguous rotation.

RudyBenner, 2019-10-13:
An ohmmeter check at the dupont connector should show continuity between red/blue and green/black but should be open between red/green and black/blue.

oldman4U, 2019-10-13:
Hi corbrettee.

Sorry for the late answer, but weekend is family time, so my hobby has to wait;-)

So you mean you have the dip switches set like this and the adv. config file in Marlin like this and this makes your steppers rotate correctly?

[image]

[image]

Would you be so kind to send me your Marlin files and/or the FIRMWARE.CUR file from your SKR so I can make sure I use absolutely the same setup board and firmware wise. And do you use the cables like shown on my pictures with green and blue twisted or like shown on the picture from Rudy?

Georg

oldman4U, 2019-10-13:
Hi Rudy.

Also a big thank you to you. I started with a cable exactly like shown on your picture, but this did not work at all - also not on the SKR E3 DIP. Then I checked the pins on the motors to see which one belongs to which coil (and also checked the data sheet) and after changing the green and blue wire the motor and the cable works on the E3 DIP and at least it generates the vibes using the v1.1 Pro. But what I really learned the last few weeks is, that there is always a little surprise waiting for you around the next corner;-) So I will test it again and let you know.
Georg

RudyBenner, 2019-10-13:
Do the ohmmeter checks, and verify the pinouts on the board.

cobrettee, 2019-10-13:
Hi Georg...my DRV8825's are working - I had some time this morning so I also tested with working A4988s(from another printer) and I am seeing same behavior you see. No matter what settings I choose the A4988s won't turn the motors and it is not a motor wiring issue at least for me since I am using the 8825s with no issue. You can also do a simple led test to confirm you have the correct pairings.

oldman4U, 2019-10-13:
Hi Rudy.

Twisting the cable I get most of the time no sound and no vibration and sometimes a bit of one or the other;-(

I will do more multimeterwork as soon as I have the time to do so, but so far I trusted the documentations.... Maybe I should stop doing so.

Thank you again

oldman4U, 2019-10-13:
Hi cobrettee.

AH, you are using a different driver and this works, but not the A4988. Slowly it makes sense to me that the support does not answer any more. Thank you for the time you spent to verify my problem. May I ask you which 2209s you ordered? I would then order the same and we could work on the firmware together - if it is ok for you. The amount of time I spent to try to make it work is nothing compared to the costs of new drivers.

RudyBenner, 2019-10-13:
I may have confused the issue, if so, I apologize.

cobrettee, 2019-10-13:
Sure - they should be delivered next week [https://www.aliexpress.com/item/33028054557.html?spm=a2g0s.9042311.0.0.23de4c4dpthCwf](url)

When I have more time I may take another stab at the 4988's and will let you know if any success.

oldman4U, 2019-10-13:
;-)

Nothing to worry. Double check can never be wrong.

Georg

oldman4U, 2019-10-14:
Hi cobrettee.

I ordered the same, thank you.

Rudy.
I tested the connectors on the board and the connectors at the motor and the two middle wired has to be twisted. A is 1 and 4, B is 3 and 6. Thank you again for the input.

Georg

RudyBenner, 2019-10-14:
Can you close this?

oldman4U, 2019-10-15:
Hi Rudy.

Unfortunately not.

So far we can exclude almost 100% that the problem is not the motor or the cable to the motor and I would say after days testing and searching - the problem also does not sit in front of the computer.

On the other hand we have 2 independent systems with the same issue and the two manufacturers most likely responsible for the issue do not care.

I would like to keep this ticket opened until it is solved or at least to tell others to look for another solution.

In case there is someone out there using the SKR v1.1 PRO mainboard successfully with the A4988 motor driver it would be great to let me know. So far we haven't heard from a single user.

It would be great if I would be able to close this ticket soon and I will do so, once the issue is solved.

Thank you

Georg

RudyBenner, 2019-10-15:
**_* A4988 is assumed for unspecified drivers._**
 *
//#define X_DRIVER_TYPE  A4988
//#define Y_DRIVER_TYPE  A4988
//#define Z_DRIVER_TYPE  A4988
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define E0_DRIVER_TYPE A4988
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988

**_Leave them all undefined, A4988 is the default. It's worth a try._**

RudyBenner, 2019-10-15:
also E0 is the first extruder, you have E0, E1, E2.

DJKyomi, 2019-10-21:
Hello, 
Looking for a more modern board I found this picture, can it be due to that?
https://cdn.shopify.com/s/files/1/1619/4791/files/BIGTREETECH-SKR-PRO-V11_09.jpg?v=1561201378

oldman4U, 2019-10-22:
Hello DJKyomi.

Thank you for your input. Using the A4988 with the SKR v1.1 PRO the jumper to connect the RST and SLP has to be set plus the jumpers for MS1 - MS3 depending on the number of micro steps you want to use. You can see in the first and the last picture i have sent, that the jumper you mentioned is always set, but with different settings for the micro steps. So according to the description from BTT, this should be fine.

In the meantime i ordered a new SKR v1.1 PRO as well as 2209 drivers to see if i can get this baby work somehow. I also got some help from tech support, but nothing since one week. Its strange.

Regards

Georg

oldman4U, 2019-10-30:
Hi cobrettee.

Have you got your TMC2209 in the meantime?

I got mine today and after approximately 10 minutes, a basic test with a single stepper driver and motor has been made successfully. This is exactly how I would have been expecting it also with the A4988....

For me it seems to be more and more that the SKR Pro v1.1 is sold with A4988, but that this combination of board, driver an software simply does not work.

Please let me know

Thank you

Georg

cobrettee, 2019-10-31:
yes got mine a week ago and they have been working very well. I simply won't bother with the A4988s any longer but it should be fixed for others. I think they assume if you are going with this board you are going with better steppers as it is the "pro" version so won't be to eager to fix it.

oldman4U, 2019-10-31:
Hi.

Thank you for the update. For me it is fine as it is, now that i have a working solution. But how can we inform others to avoid this combination? I think it is enough that i wasted my time.

rflulling, 2019-11-29:
Hello all. I understand this is an old post. But it's been left open. I noticed some work being done By BTT on several new revisions of default firmware configurations, since the onboard SD card has been made to work for them. I am a user. I have direct experience with this setup. I have since moved to ST820, and plan to change to Trinamic later. 

So. 1st thing. A4988 works just fine with the SKR PRO 1.1 and Marlin 2.0.x

2nd. A) Jumpers are not the same pattern as with other Drivers and no assumptions can be made. I generated a informative PDF to help users with Jumper configurations for SKR PRO. But even this should not be taken as absolute. Please for sure compare notes with the ACTUAL manufacturer of your driver. Not just one that makes the same chip. It matters! B) The A4988 are not well suited to all motors. Check the math. Check the specifications for the driver and the Motor to be sure that the driver can handle it. It is possible to make a motor run that the driver doesn't like and this will cause the driver, or motor, or both to get very hot. A well configured combination might get warm, but not hot. Other drivers will get very hot no mater what. C) Check the specifications for the the driver and chosen motor make sure the wires match correctly. 

3rd. Once your jumpers are configured then you need to either manually do the math and calibrate the POT on the driver, one at a time or set each one slightly clock wise. That is if you note that there is a max left and max right. Set the POT to dead center between those points. This may even have been done for you by factory. Now turn it slightly to the right. This should be good for testing. 

4rth. Some defaults for A4988 from Configuration.h 
It is more than probably that one or more of these may need to be tweaked for your setup. 
Bear in mind too. I am not sampling End stop defines or config. But having that set wrong may literally deny you ability to move an axis, even if you have the right motor driver config. So check your work. You will need to configure your motion yourself. Jerk, acceleration, max speed can come later but do require tweaking. 
#define X_ENABLE_ON 0
#define Y_ENABLE_ON 0
#define Z_ENABLE_ON 0
#define E_ENABLE_ON 0 // For all extruders
#define INVERT_X_DIR false
#define INVERT_Y_DIR false
#define INVERT_Z_DIR true
#define INVERT_E0_DIR false
#define INVERT_E1_DIR false
#define INVERT_E2_DIR false
#define INVERT_E3_DIR false
#define INVERT_E4_DIR false
#define INVERT_E5_DIR false

5th. Configuration.adv defaults
#define AXIS_RELATIVE_MODES { false, false, false, false }
#define INVERT_X_STEP_PIN false
#define INVERT_Y_STEP_PIN false
#define INVERT_Z_STEP_PIN false
#define INVERT_E_STEP_PIN false

6th. I have never seen the options for setting microstep from the firmware do any good. Simpler to leave it disabled. So as to not complicate setup, introduce over ride or bugs. 

7) When you have a good compile and the firmware is saved to the SD card, and the printer is reset. 
You can try to move an axis. 

8) To set the POT you can configure on the fly with a small screw driver. But it is recommended to do this only with a ceramic tool. Because the risk of touching anything else is very high and this can damage the chip, or power regulators by shorting VSS to ground. 

9) If you try to move the motor and it's very noisy, or shudders. A) Check the pins on the cable and make sure they are configured correctly to the motor. B) If you can be 100% certain the wires and jumpers are right. (this is kind of a big deal, do your homework). Then you can consider adjusting the POT further. C) from the LCD send the motor in a direction where it has some room to travel and the firmware is willing to let you do this. Say 50 to 100mm. Now adjust the POT slightly until the motion is smooth. It is very unlikely that you will need to adjust much at all. If you have to rotate more the 1/4 turn its likely that some thing is very wrong in your hardware setup, likely the motor, the jumpers, cables, or even your chosen DEFAULT_AXIS_STEPS_PER_UNIT.

oldman4U, 2019-11-29:
Hello rflulling.

THANK YOU for your help and all the time you spent to answer this ticket.

Could you please send me a picture of the jumper setup of your SKR Pro v1.1 using an A4988 successfully as well as the FIRMWARE.CUR file of your machine, so i can do another test.

It is not that the motors need a bit of adjustment when using A4988, but they do not move at all or only piep a bit. 2 different motors with the same cables work fine with the original Ender-3 mainboard which has A4988 on board as far as i know, work fine with the SKR E3 DIP with 2208, 2209 and also work fine with the SKR Pro v1.1 using 2208 or 2209. In the meantime i am able to use 2209 based sensorless homing, using the same motor/cable combination. I believe they are fine. In general, everything works fine, as long as i do not combine the Pro v.1.1 with the A4988 and i am not the only one who is not able to get it work.

So in case you use this combination successfully, i am more than willing to test it again and to close the ticket, the same time it works. Otherwise i think it would be important to let the others know, that this combination does not work, so i would like to keep this ticket opened.

Kind regards

rflulling, 2019-11-29:
Attached are the jumper config I meant to attache to the last.

*I will have to double check on the Compiled firmware.* This one, although
I have a running record of every compile done. They are the Folders and in
the begging for the SKR PRO most were hybrid A4988 + ST820. My X Axis, and
Z if I recall were both A4988. That is because the ST820 have been
unpopular and they only sell them in 4 packs... So I had to order a total
of 8 for my 6 seats, and the second batch came later. I upgraded from an
MKS 1.4 to the SKR PRO, and the MKS was 5 slots of A4988.

*I am also not sure if it is legal for me to just distribute a compiled
firmware* as apposed to the "configuration" files. I have not read the
various legal documents well enough.

I will venture that two distinct advantages of the A4988 over the ST820.
Quieter, and generate less heat.

The ST820 driver do everything in backwards world. So they spend more time
on than off, and every setting in Marlin for this driver must be reversed
to a mirror of what the A4988 are set.

oldman4U, 2019-12-02:
Hello rflulling.

Unfortunately there is still nothing attached, at least i can not see it.

I spent several weeks of my time to try to get this to work, so i would need pictures and a firmware which makes the SKR Pro v1.1 work with the  A4988 drivers. Once i have both, i will test it again. Until then i will enjoy my SKR Pro v1.1 with the 2209's which are working very very nice without spending days and weeks to get them to work.

Thank you

Georg

rflulling, 2019-12-07:
[driver seat pins SKR PRO.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3935226/driver.seat.pins.SKR.PRO.pdf)

Not sure how its been over looked. I know it was attached. Sorry it did not get through. Hope this helps. Will try to get time to offer better this weekend. Sorry been very busy with job. 9-8 and fighting to get my dual extruder behaving the way I need.

oldman4U, 2019-12-07:
Hi.

No hurry, but the pdf is 6 empty pages.

rflulling, 2019-12-07:
Ack....

rflulling, 2019-12-07:
Ok, this one works... sorry for the trouble. Realy I am.

https://drive.google.com/file/d/1CR7NcZsWs012gB1fWaCSrZSwWh_6bvzq/view?usp=drivesdk

oldman4U, 2019-12-09:
Hello rflulling.

No reason to be sorry, you try to help and that's great, thank you.

The pdf is a great summary of all driver related options for the SKR Pro, but the information is not new to me. In addition to all the options you listed i also tried all settings without the jumper shortening STEP and DIR, i tried different cables, motors, even different Marlin settings and this over several weeks, spending all my spare time. So in case you have an SKR Pro v1.1 with A4988 successfully working on your side, please send me also the firmware. Otherwise there is no reason for me to believe that repeating a test which i have already done several times will result in something different. In the meantime i use this board without any problem with 2209's and i also ordered a new SKR Pro v1.1 which i will use for additional tests to make sure that it is not the current board which has a problem with this particular driver.

Until then i must expect that this combination simply does not work. Even more, because there is at least another user with the same issue and so far not a single user using exactly this combination (SKR Pro v1.1, A4988 driver with Marlin v2) successfully.

Kind regards

keyser187, 2019-12-16:
i have the exact same problem. has anyone got this working yet?

DJKyomi, 2019-12-16:
Ich hätte eine Idee, aber ich kann sie nicht testen.
Am Ramps 1.6 Plus muss man bei den A4988 und DRV8825 die Pins SLP und RST verlöten. vielleicht löst das auch Eurer Problem ?
scroll down on the page

https://www.biqu.equipment/products/biqu-3d-printer-parts-ramps-1-6-plus-upgrade-from

oldman4U, 2019-12-16:
Hi.

Unfortunately we have 3 users where this combination does not work and so far not a single user using the A4988 successfully on the SKR Pro v1.1.

On the SKR Pro v1.1 you have to set a jumper to connect SLP and RST and at least I did it exactly like this.

Thank you DJKyomi for trying to help.

[image]

keyser187, 2019-12-16:
im following the same guide and the pics you posted, as i say, i get no movement just a very hot driver. even tried it with just the one jumper as in the guide but nothing.  i cant afford new stepper drivers after buying the fricking board! so would really like to get this to work (and its my BD 2mora :$  ) im still waiting for a reply to my email from bigtree support. if i hear anything new ill let you know

rflulling, 2019-12-16:
I cannot upload a fully compiled firmware. Doing so puts me in a vendor
position where by the terms of the contract we all agree to, knowingly or
not, I become liable to you for updates. Thats not going to happen. I can
however offer copies of my config, so long as you know and understand that
it is NOT currently configured for your machine. I did reconfigure an older
mixed file. That was set up to run both A4988, and ST820. FYI they are
mostly the same thing but the ST820 operates 100% inverse the A4988 in
every possible way. Anyway I test compiled that build several times with
100% success. If not being advised to liability of vendor status, I would
have already offered that compile. I also know that when I grabbed the a
copy of Marlin from BTT for the SKR PRO and merged it with my MKS 1.4 the
only issues, were my own.

The Only thing I have not done, is to switch out the Drivers on my PCB for
a test. This I think I can do today, snow day... Probably a good time to
grab the current and OFFICIAL release version of Marlin 2.0. I think I will
generate two branches. One for the ST820 and one for the A4988. I will take
the time to test both flavors. I can and will offer a zip folder with the
Configuration used. Its high time I official mount the SKR PRO anyway. It's
been floating on a billing envelope to keep it from touching the Heat syncs
on the A4988 standing on the MKS 1.4 behind it...

I think perhaps you might take this complaint over to the actual
conversation on Marlin.

BigTree listens but for sure picks and chooses who to listen and reply too.
Getting a reply from BTT is very unlikely, unless they believe you have an
actual issue.
So maintaining this topic here is a bit like shouting in frustration to the
coats on a closet. Sometimes the other tenants in the building show back,
but that's about the best you can hope for.

oldman4U, 2019-12-16:
Helle rflulling.

Thank you for Your continuous support.

Seems you are the only one able to make this combination of drivers and
boards work. So please provide a firmware OR the config files and a picture
of your board showing the jumper setting and i will give it another try.
This issue is now the one with the most replies. It would be great to be
able to say - now it works and here is what you have to do to make it work
and to close it the same time.

rflulling, 2019-12-17:
I did not get the hardware switched for testing but the firmware is
branched and passed compile on both version. Ready for the hardware.
As stated, I cannot give the compiled Firmware for legal reasons. I can
offer the Configuration files.

I will zip both sets, for your viewing after I have validated the function
in the hardware test.

keyser187, 2019-12-18:
hi oldman
i got an email back from BTT with a BIN file attached for me to try. i tried it and nothing worked again so i plugged the drivers and motor back into my GT2560 (which was literally working before i carefully unplugged it all) but nothing works on that now either. As i have no electronics experience i have no idea whats gone wrong/broke so i give up. i only got into 3d printing to help with mental health but it makes it worse.
if you would like to try the BIN file for yourself to try i have attached it. but i have no idea what they changed as it is just the bin file and not the marlin folder.
good luck
[BTT-email-firmware.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3978509/BTT-email-firmware.zip)

oldman4U, 2019-12-18:
Hello keyser187.

Thank you for the bin. Do you also have a picture which shows the jumper configuration you have been using? I am also a 3D printer newbie and bought my first multimeter at the age of 50 6 months ago;-)

oldman4U, 2019-12-18:
I have a general question to the experts here. Which MARLIN settings are driver relevant when using a Stand Alone driver like the A4988 (compared to the 2209's in UART mode which i use at the moment). All driver related current and micro steps settings are defined directly at the driver or mainboard, as far as i understand. Right or wrong?

keyser187, 2019-12-18:
sry no, they sent no pic of the jumpers. ive been using the same as you H,H,H and L,L,H (from right to left) but as i say i just tried my gt2560 (which FYI has printed on the board that 1/16 is H,H,H) but nothing, no humming, no vibration the motors are dead so i have no clue what to try next. ive spent more time trying to get things to work than actually running the printer (i got my geeetech pro w about 2 years ago) it now looks like this and im walking away from 3d printing!
[image]
[image]

rflulling, 2019-12-18:
You must have power to all three sets of power input. They do not have to
be the same source. They must be connected and powered on.

Jumper in middle of board. The board will power up and accept new firmware
from onboard SD if the USB power is switched on. If the 5V is switched on,
then the board will only power up when the main power is connected.

oldman4U, 2019-12-18:
@keyser187 Yes, the jumpers are the same i have been using. My son has an Ender 3 Pro with SKR E3 DIP and TMC2208's working fine and now i build my own using the Pro v1.1 and 2209. Sometimes the way is the most funny part, so i play around with it and learn every time something new.;-) Enjoy and relax.

@rflulling. I have 24V connected to all inputs and the board works fine using 2209s. So in general my setup should work, but i guess you wrote this related to the pictures from keyser187.

keyser187, 2019-12-18:
> You must have power to all three sets of power input. They do not have to be the same source. They must be connected and powered on. Jumper in middle of board. The board will power up and accept new [...]

sorry im lost. i have connected motor power and power to my psu, i thought the other 2 were for the heat bed which am not using ATM. i have the jumper set to use the psu and not usb. firmware flashes fine. as i say the motors are not working with the gt2560 now either. (i had it printing the night before i started the swap)

oldman4U, 2019-12-18:
Not sure the other power lines have to be connected. I will check and let
you know.

oldman4U, 2019-12-18:
Hello keyser187.

I have tested the SKR Pro v1.1 with TMC2209 with only power connected to 1-4 (so Motor Power and Main Power) and it works.

keyser187, 2019-12-18:
Thanks for trying that out for me and confirming. With the fricking day I've had I thought sod it and brought some cheap tmc2208's, I believe they are standalone and not uart (just want to get it working first). Should be here 2mora so I'm sure I'll be having another bad day 2mora!!

rflulling, 2019-12-19:
If the A4988 are not working and we can walk you through proper hardware
setup and you have the proper basics set in the configuration. Then you
likely have something else wrong. Like a an ESD nuked controller. ESD can
kill a board not just by it being unprotected and you having touched it.
But also by spinning a motor by hand while connected by wire to the board.
There are other ways, I have seen the burned boards. Though I am less clear
on how that happens. Regardless. Switching hardware right now. Should have
an an answer Very soon. My expectation is to see the motors move as
expected based on the old configuration.

rflulling, 2019-12-19:
[image]

**Attached are the files I alter.** 

PlatformIO.ini, is altered for viability, and need, not function. Not changing the PlatformIO.ini with the one I have, will change nothing. Just be sure you select the right environment. Mine, the default is the SKR PRO so there is less risk of oops. 

Configuration.h and Configuration_adv.h are configured for my machine. This includes the feed rates for my motors and my drivers as well as a tweak to bring the system fan on when the drivers are used. EEPROM and SD cards are enabled. However SD card enable in in default so, Marlin will auto configure, meaning differences should not matter. EEPROM and the Fan tweak require the Pins file to be tweaked. Too.

For testing with my files. My instruction is to Download a clean ZIP directly from Marlin for the Release version of Marlin2.0. _Please do not use any other variant, including BTT version or Nightly build_. Using the attached files that are best suited for testing your configuration, Drag and drop the contents onto the your newly downloaded copy. Keep mind, that this needs to happen one level up from Marlin. You should see the Marlin folder in the folder you are pasting into. _You should not be inside Marlin._ Doing this right the contents will perfectly replace what they need to. 
[image]

Now open Visual Studio and open folder, select the new folder you just created and merged. Wait for it to do its cataloging. Then instruct go into configuration and check to be sure the LCD configuration is good for your build. Otherwise you will get a blank screen. Do not change anything else if possible for this test. Now, build the environment. When the file is moved to the SD and the SD is installed. After the lights stop flickering and the Controller has rebooted. 
**Now from the menu, select "Configuration" then "Restore failsafe." Not doing this will result in previously loaded possibly bad values being used for the test. You should repeat this when switching to your own build, or any other variant.** 

Bear in mind. All this should tell you is if you have something configured wrong or if you have a hardware issue. If you get Nada. Something is very wrong. It might be user error, it might be hardware failure. But you should get some kind of life otherwise.  

Both ST820 and A4988 have been tested with and verified working. 

[Marlin2(A4988).zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3981183/Marlin2.A4988.zip)
[Marlin2(ST820).zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/3981184/Marlin2.ST820.zip)

rflulling, 2019-12-19:
> Thanks for trying that out for me and confirming. With the fricking day I've had I thought sod it and brought some cheap tmc2208's, I believe they are standalone and not uart (just want to get it working first). Should be here 2mora so I'm sure I'll be having another bad day 2mora!!

Given the number of variables and new issues that TMC introduce I strongly suggest NOT playing with them until your A4988 are working. Plus A4988 are dirt cheep to blow and replace. Your TMC2208 are not as cheep. 

This is like strapping a jet engine to air plane missing a wing. We need to fix the wing first. Then we can talk about upgrades. Please focus on getting the controller behaving right before upgrading.

oldman4U, 2019-12-19:
Hi rflulling.

In case the board would be broken, i would expect other stepper drivers to fail also, but the 2209 are working fine. But you are right and this is the reason why i ordered a second board - just to be sure. Usually i work "electrically grounded" (hope you say it like this;-), but there have been exceptions due to the many tests i made in too little time. I checked the config files and besides the second Y and E steppers and the Babystepping which i have turned off, i can not see any obviously important differences. But i will follow your very detailed and also very helpful instructions above as soon as possible and let you know.

THANK YOU

Georg

oldman4U, 2019-12-19:
Hi again.

I understand everything but this:
**Now from the menu, select "Configuration" then "Restore failsafe." Not doing this will result in previously loaded possibly bad values being used for the test. You should repeat this when switching to your own build, or any other variant.**

Is this something i have to do in VSC or on the printer (its display)?

Thank you

keyser187, 2019-12-19:
i believe its from your display. its basically clearing everything from the memory etc and restoring "factory settings" which will be the firmware you just flashed, so you can be double sure its only using that.
 i maybe wrong tho

oldman4U, 2019-12-19:
OK. I am using a TFT28 and not an LCD. Not sure i have seen this in the menu there before but will check.

AnHardt, 2019-12-19:
The safe option to never forget to load the configured values during tests, is to disable the EEPROM (persistent storage) option at all.
Always compiling in new values lasts a bit, but it's much faster than testing not the values you believe in. On the long term this saves time if you tend to make errors with M50x, or that's what fails .

keyser187, 2019-12-19:
at last my 2208s came, plugged em in and working (well the x axis so far) so it looks like ALL 4 of my a4988 broke somehow. have to say tho, BTT support was pretty good. They replied to all emails within 24hrs (im uk, them china) polite and even sent me a modified firmware to try.
sry i couldnt help anyone with the same problems but im sure rflullings fix will work, they put a lot of work into that it seems.

[112 more replies]