bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #60: How to connect a 12v auto leveling sensor

mooner1022, 2019-10-16:
I have a auto leveling sensor which uses 6V~36V(rated voltage of 12V) and I have no idea how to connect this sensor on the SKR PRO. It worked with 5v on my old Anet board, but It doesn't turn on on this board's 5v power.

mooner1022, 2019-10-19:
Solved it by adding a npn transistor for switching

sneakym8, 2020-01-10:
I think you could also wire it to 12v on the motherboard to get full sensitivity and then wire the signal through a potential divider to the input.
or even if you are good with your maths , through a current limiting resister.

Mariu86, 2020-02-06:
Hello @mooner1022  can you please tell me if you used the limit switch pins or the Bltouch pins ?

mooner1022, 2020-02-07:
> Hello @mooner1022 can you please tell me if you used the limit switch pins or the Bltouch pins ?

I used limit swich pins and 12v output pin.

Mariu86, 2020-02-07:
But the pin from the limit switch it's 5v tolerant I think, how do you put 12v in there? I have an induction sensor and tmc2209, but I can not get the sensor to work. if I put a mechanic switch and i desolder the diag pin form tmc2209 the mechanical switch it's working, but the induction sensor it's not working. Sorry for my bad English.

mooner1022, 2020-02-07:
> But the pin from the limit switch it's 5v tolerant I think, how do you put 12v in there? I have an induction sensor and tmc2209, but I can not get the sensor to work. if I put a mechanic switch and [...]

Your english is good bro :)
I plugged my sensor to 12v out pin and a npn transistor(in my case, 2n3904 worked.) to the limit switch pin. When the sensor on 12v triggers, it turns on the transistor and the board recognizes the triggered sensor. If you have something more to ask, feel free to ask any time :D

Mariu86, 2020-02-07:
@mooner1022 sorry but I don't have so much experience with npn transistors , can you be more explicit how do you do it please! From what pin do you take the 12v? And do you connect the signal pin to the PG8 pin? What settings do yo do In Marlin ? My old board is a MKS gen L 1.0 and the sensor was connected to the z- limit switch and I checked the pin configuration and I have the same configuration like the skr pro the only diference is the voltage in the skr pro we have the 3.3v and the MKS has 5v. Today I was looking in the marlin pin configuration for skr pro and I see mention the PA2 pin for probing. 
// Z Probe must be this pins
#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN  PA2
#endif
I will try tomorrow to use the Bltouch pins 5v, ground and PA2 to see if it's working. Thanks

mooner1022, 2020-02-09:
@Mariu86 Sorry for the late reply, my alarm for github didn't work. I'll show you my board settings when I go home. I think I'll be able to show you in 2~3 hours.

mooner1022, 2020-02-09:
> @mooner1022 sorry but I don't have so much experience with npn transistors , can you be more explicit how do you do it please! From what pin do you take the 12v? And do you connect the signal pin to [...]

[image]

This is the  schematic of my circuit. (Sry for the bad visibility of the circuit. I'm not good at drawing things D:) I didn't change any firmware things, just attatched this circuit to the board and sensor. I don't know if I'm explaining it well, so ask me if you have something not understanding.

GadgetAngel, 2020-02-09:
@mooner1022 Here is a nice write up. Just change the resistor across the sense pin for 3.3 V logic instead of 5V logic. https://mertarauh.com/2017/01/18/dont-trust-the-internet-and-how-to-add-an-inductive-proximity-sensor-to-your-3d-printer-the-proper-and-easiest-way/

mooner1022, 2020-02-09:
> @mooner1022 Here is a nice write up. Just change the resistor across the sense pin for 3.3 V logic instead of 5V logic. https://mertarauh.com/2017/01/18/dont-trust-the-internet-and-how-to-add-an-inductive-proximity-sensor-to-your-3d-printer-the-proper-and-easiest-way/

Thx for telling a way simpler way!

Mariu86, 2020-02-09:
Thanks, for all the info I will measure the output signal of the sensor and I will see what's the voltage to see if it's to low . I have been trying to use the Bltouch pins but i don't have a complete success yet , the PA2 pin it's not trigger, I try different settings in the Marlin but maybe I do something wrong. Latter I tried to use the 5v pin and ground from the Bltouch pins and the signal pin from the z-pin I bend the diag pin form tmc2209 and the inductiv sensor it's working. I will try to see what it's the problem with Bltouch signal pin. Can you share the config.h and and adv.h files ?

GadgetAngel, 2020-02-10:
@Mariu86 i am still testing my SKR PRO board and writing a manual on how to do the Stepper Diver Configurations for the SKR PRO (now over 70 pages long). I plan on putting a BLtouch and or a TH3D inductive sensor onto my printer but I have not got that far yet. When I do, if someone else has not posted the files, I will post mine here

Mariu86, 2020-02-10:
I tested the inductiv sensor but I don't have any success on the PA2 pin ! I tried every setting in the marlin but I don know what's the problem, I'm not a expert but I don't understand whats the problem, if anybody has a config file using the PA2 pin for probing will be very useful for me ! Thanks

mooner1022, 2020-02-10:
> I tested the inductiv sensor but I don't have any success on the PA2 pin ! I tried every setting in the marlin but I don know what's the problem, I'm not a expert but I don't understand whats the problem, if anybody has a config file using the PA2 pin for probing will be very useful for me ! Thanks

[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/files/4180640/Marlin.zip)
Here's my configuration.h and adv.

Mariu86, 2020-02-13:
Thanks for all the help , I manage to use the PA2 pin for inductiv sensor but i must use the limit switch for z min, I do the homing with limit switch and UBL leveling with inductiv sensor .