bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #61: PC9 not PWM-capable - but it should

Haschtl, 2019-10-17:
The PC9 pin is a PWM-pin.

In buildroot\share\PlatformIO\variants\BIGTREE_GENERIC_STM32F407_5X\PeripheralPins.c I commented the I2C usage of PC9 and uncommented the PC9 definition for TIM8_CH4 (which isn't used by any other pin). But anyway the behavior of the pin is not as expected. 

I connected a mosfet+fan to PC9. If I enable the fan, it runs immediatly at fullspeed and if i turn the fan off, it remains running.

The STM32F4 is a good microcontroller with a lot IO. But why are a lot of PWM-pins not available onboard (PB11, PB5, PB4, PA11, PA10, PA9, PA7, PA6, PA5) ? The only PWM-pins on the two extension-blocks are PF8, PF9. PF7 shares its TIMER with the Z_CS_PIN and is not useable for PWM.

So in the end, if you want to connect an LED and a LASER to the two additional PWM-pins you cannot connect any other PWM-device. Am i right? (I'm using the E2_STEP_PIN PD13 as I do not have 3 extruders)

omahena, 2019-10-18:
The PeripheralPins.c has been the focus of my attempts to get the SD card working and to remove a weird "entanglement" between the HEAT0 (PB1) and HEAT1 (PD14) pins. On a single extruder system they are always blinking in unison - as soon as the HEAT0 starts to be driven by Marlin, both LEDs for HEAT0 and HEAT1 are blinking as one.

The changes you did agree with my understanding of the file. Also I share your sentiments about the board. The support for it sucks. Only the absolutely basic single extruder Marlin based 3D printer system seems to have been (sloppily) tested... And the EXT1 and EXT2 headers leave a lot to be desired. I think they didn't expose other GPIO pins on the board to not make it too crowded with pin headers. Also the board would probably benefit from a different version of the MCU with more pins to reduce some of the collisions of alternate PIN usage.

The behavior of the PC9 pin you describe is a little surprising to me as well. I see PC8 shares TIM8 (FAN0) with PC9, but from the MCU documentation I think that should be OK as the timer offers 4 independent channels...

I see some MOSFETs also have a max input frequency. Only 44kHz in this case:
[MOSFET module](https://www.aliexpress.com/item/32950922168.html)

Now what also surprised me during testing and cleanup of this file, is that after I removed all the conflicting definitions the PB1 - PD14 entanglement remained. Next to do a sanity check, I removed the PWM definition for one of the used pins (it might have been PB1 for the heater0) and Marlin still compiled and worked without and issues. At that point I decided that I have more important stuff to do and aborted the endeavor.

The PeripheralPins.c code definitely gets compiled, but I cannot say how, where or if it gets used. :/

Haschtl, 2019-10-18:
Strange behavor of heater0 and 1. 
Luckily I have two hotends connected and do not need to face this issue.

I must say, i really use a lot of the pins from this board and now everything is working. But the pins are indeed a bit confusing and there exists no documentation of how they are configured. 

I think Marlin doesnt check if the pins are pwm-pins. Only for the Laser-PWM pin. But also there are some precompiler problems in marlin for stm32 boards, thats why i needed to disable the PWM check manually to use the laser.

I also saw that this file gets compiled. Maybe i will clear the file completely and check if marlin can still be compiled. Then these definitions are somehow redundant in another place

Haschtl, 2019-11-26:
I will close this, as some time has passed and i dont know if this problem still exists as i'm using a different pin

closed 2019-11-26 by Haschtl