bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #63: SKR pro 1.1 no eeprom

SolorEclipse, 2019-10-26:
Connecting...
Printer is now online.
>>> m500
SENDING:M500
echo:No EEPROM.
>>> m501
SENDING:M501
echo:No EEPROM.
echo:Hardcoded Default Settings Loaded
>>> m502
SENDING:M502
echo:Hardcoded Default Settings Loaded

Error Message from Pronterface ver 1.6.0

SolorEclipse, 2019-10-26:
I am having trouble uploading my configuration.h file so I uploaded it to Google Drive
config.h         https://drive.google.com/file/d/1HnrSHybRc1RVkOWFWsBod5Pkb0eSbWQl/view?usp=sharing

config.adv      https://drive.google.com/file/d/189sBCxe_v6YWAR4f1h61JiIVta-dbCD8/view?usp=sharing

dineshvyas, 2019-10-27:
[image]

in pins_BIGTREE_SKR_PRO_V1_1.h uncomment line 34. compile and flash new firmware and you are good to go.

SolorEclipse, 2019-10-27:
Thank you my problem has been solved

closed 2019-10-27 by SolorEclipse