bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #65: PT100 temp sensor - spare pins?

jaygrovr, 2019-10-27:
Does anyone know if there are any spare pins for connecting a pt100 amplifier board for my extruder on the skr pro?  I am already using analog in pins for bltouch. I also have tmc2130's not sure if that matters.  I would also like to connect the SKR lcd to this controller.  I cant seem to find any on the pinout schematic.

thanks
jason

evandene, 2019-11-29:
I use E3D PT100 sensor and E3D amplifier.
Will change this later by a Adafruit amplifier because of the 5V input and the 3.3V output signal of the board.
T0 is responding on my E3D amplifier with PT100 sensor
I use PF8 from EXTENSION1 as signal pin.
I use GND and 3.3V from a free limit switch connector.

settings in configuration.h
#define TEMP_SENSOR_0 20
#define TEMP_SENSOR_1 20

oldman4U, 2020-10-10:
Hi. Could you please check this #163. It should solve your problem and answer your question.

Thank

Please help the community and close the ticket once you do not need it anymore. Thank you