bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #66: Current Marlin 2.0 Fails to Compile with TMCStepper 0.5.2 and STM32 5.7.0

DroneMang, 2019-10-28:
Hi,

I am trying to get the most current Marlin 2.0 working from the official Marlin 2.0 page.  The main issue seems to be that the newest TMCStepper 0.5.2 and STM32 5.7.0 do not work together and SoftwareSerial.cpp references need to be taken out because the newest STM32 5.7.0 already has SoftwareSerial built-in.  Some are using STM32 5.6.0 and TMCStepper 0.5.1 to compile but there are still errors. 

Below are some applicable pages:
https://github.com/MarlinFirmware/Marlin/pull/15655
https://github.com/MarlinFirmware/Marlin/issues/15538
https://github.com/MarlinFirmware/Marlin/issues/15496

From this page https://github.com/MarlinFirmware/Marlin/pull/15530, it seems that official Marlin 2.0 will not include the version of TMCStepper hosted here at https://github.com/bigtreetech/TMCStepper

Would it be possible for someone from Bigtreetech to try to work on these issues as well? Thank you.

bidmangler, 2019-11-06:
Seconded from me. I get the same errors when trying to compile the latest Marlin - and don't have the skills to fix the SoftwareSerial.cpp references.

Please could @bigtreetech please give us an update?

Haschtl, 2019-11-06:
I can now compile with the latest TMCStepper library due to this pullrequest: https://github.com/MarlinFirmware/Marlin/pull/15796

SoftwareSerial is taken from another repository. 
Check if this fixes your problem, too.

sjasonsmith, 2019-11-06:
The SoftwareSerial in STM32 5.7 is broken, at least for half-duplex stepper communication. The issues are fixed, but are not yet released.

You should probably continue to use STM32 5.6 for now, if you need SoftwareSerial. You can then use whatever worked for you before, such as the BTT version of TMCStepper

> I can now compile with the latest TMCStepper library due to this pullrequest: [MarlinFirmware/Marlin#15796](https://github.com/MarlinFirmware/Marlin/pull/15796)

That change was specific to the SKR Pro, which still uses STM32 5.6. The fork of SoftwareSerialM I created is not compatible with STM32 5.7. It is only intended as a very temporary solution while @LinoBarreca finishes the changes needed to mode the SKR Pro to STM32 5.7.

I intend to go back and try building and figure out solutions for my SKR E3 DIP/Mini boards, but I am still actively working on SKR Pro issues.

LinoBarreca, 2019-11-07:
@DroneMang if you have TMC2209 you can actually pull [this](https://github.com/MarlinFirmware/Marlin/pull/15655) which should be working without issues for you.
Updated to the latestestestest packages available to you or as Jason said stick to the BTT's version.

LinoBarreca, 2019-11-14:
This can be closed. Since it has been solved.

closed 2020-10-10 by DroneMang