bigtreetech/BIGTREETECH-SKR-PRO-V1.1 #67: Upload firmware via USB port?

Franchelli, 2019-10-29:
Is it possible to load the firmware on the card through the USB socket without using the SD card?
I tried to use Atom and Visual Studio Code but at the end of the compilation there is an error in the loading in the card.
Sorry for my bad English ...

bidmangler, 2019-11-04:
@Franchelli  - As a beginner, it took a while to understand what was in the thread correctly pointed to by @Evg33 - but the simple answer is "No" - it is not possible to upload the firmware through the USB socket yet. 

As an owner of a shiny new SKR PRO 1.1 that's been confusing me as well.

LinoBarreca, 2019-11-14:
it should be possible according on how the bootloader is configured.

jgwinner, 2019-12-11:
I took a  look at that thread and offered to help.

I'm working with a CR-10 converted to a "Crearibo", 4 TMC2209's, and 2 TMC2208's, working on dual Z motor (not just split), dual extruder. BLTouch. 

        == John ==

oldman4U, 2020-10-10:
Please help the community and close the ticket once you do not need it anymore. Thank you

jgwinner, 2020-10-11:
I could still use it. Swapping SD cards around is a PITA.

oldman4U, 2020-10-11:
Others swap sd cards for being able to print.

This hardware can not do it.

theopensourcerer, 2020-10-11:
There is a suitable bootloader. Not tried it personally but here you go: https://github.com/Arksine/STM32_HID_Bootloader

oldman4U, 2020-10-11:
This is great.

Now we need someone, this feature is important enough, to flash the bootloader and try it out.

👍🏻

i-make-robots, 2021-01-11:
I just started writing code in the hopes of using a skr pro 1.2.  commenting for the day when someone shows how to upload via usb.  until then this board is a no-go for me, and I wish I'd known before buying.

jgwinner, 2021-01-11:
I did the sd card reboot thing a LOT, it does work, but I'll agree it's a PITA.

Other than that, it's a fantastic board.  There are hackery ways to do it; I bought an STM debugger USB connector that'll do it but never got it hooked up. It worked for some people.

        == John ==

frizzle101101, 2021-01-12:
Anyone still looking for a solution to this can use the marlin binary file transfer protocol to upload the firmware.bin through usb to sd card then use m997 to reset the board, where it will then upload the firmware. Enable it in firmware with #BINARY_FILE_TRANSFER

Because the binary file protocol uses the command intefrace one needs to have access to a programatical way of uploading the file where there is a m28 b1 called then the file is written through the command interface.

The best solution i have found to utilize this functionality is the marlinmtb plugin for octoprint...if you dont use octoprint you will need to find another program that utilizes the BINARY_FILE_TRANFER

frizzle101101, 2021-01-12:
Skr pro sd card hosting is not possible in current marlin version 2.0.7.2 , there is an issue with the hardware data mode of the sd card slot that does not allow for the sd card hosting feature when sd card is unmounted from marlin, like when using skr1.3, i will edit this post when i find the git hub issue in marlin where they identified this issue but yea the solution i mentioned above is the best workaround

jgwinner, 2021-01-12:
@frizzle101101 What about the STM debugger add-on boards? Sorry, hope that wasn't vague, I haven't worked on Marlin in months, but I though that was one way to do it. If you don't know what I mean I'll dig up the info. Honestly, I couldn't quite figure out how to attach it, the docs were hard to find.

        == John ==

frizzle101101, 2021-01-12:
Im sure you can go about it that way too if you want, its a direct way to do it from your pc...youd need to locate the rts cts 5v and grnd prins but yea its an f4 so i dont think theres any reason why not, unless bootloaders are locked...but i dont know about that area I've been sd card swapping to load firmware

jgwinner, 2021-01-12:
Thanks @frizzle101101 - your idea of uploading straight to the card, then rebooting is excellent though, I may try that. I have Octoprint on a different printer than my SKR though. 

Do you know of any PC programs that will do it? I've looked at a few but like S3D for slicing, so I usually have that connected to the printer. I don't know if it'll do the Marin transfer. I assumed binary was binary.

        == John ==

frizzle101101, 2021-06-14:
I have not found a program to do this but there must be something out there that can use it...maybe pronterface https://github.com/kliment/Printrun project